MarlinFirmware/Marlin #4768: M226 - remove "sensitive_pins" check ?

Bob-the-Kuhn, 2016-09-07:
If it's reasonable I'd like to see the "sensitive_pins" check removed from the M226 code.

It's not obvious to me why a read only command is worried about pins the M42 command shouldn't be messing with.

I was trying to play with my end stops and had a devil of a time figuring out why the M226 command never waited for a change of state.  

   Bob

Roxy-3D, 2016-09-07:
In the UBL branch, there are some extra pin scan functions.  Specifically, M43 and M44.   Let's go very very slow on removing any pins from the "sensitive" list.   I've seen some really bad behavior when "sesitive pins" are accessed by both read or writes.   And if we are going to dig into this, I had to add some more "sensitive pins" to the list.  Does anybody know what those extra pins are that I had to avoid touching?

```cpp
/**
 * M43: Scan for a pin to help the user locate unused pins on their controller board
 *      Either an LED or volt meter can be placed on the desired pin.
 *      This command will scan the non-sensitive pins and put a voltage on the pin for
 *      a short duration.   
 *
 * S	Start Pin number.   If not given, will default to 0
 *
 * E	End Pin number.   If not given, will default to 127 
 *
 * W    Wait time (in miliseconds) between pulses.  If not given will default to 500
 *
 * R    Repeat pulses on each pin this number of times before continueing to next pin
 *
 */

//
// sensitive_pin() is used by both M43 and M44 to avoid messing with pins that should not be touched.
// It is cleaner to have it as a function call than as in-line logic.
//
static bool sensitive_pin(int p) {
int i;
    for (uint8_t i = 0; i < COUNT(sensitive_pins); i++)  {
	if (p == sensitive_pins[i] || p==68 || p==69 || p==70 || p==71 || p==72 || p==73 || p==74  //causes KILL on my printer
												)  {
	  return true;
      }
    }
    return false;	
}

inline void gcode_M43() {
int p, j, s=0, e=127, w=500, r=1; 

  if (code_seen('R')) 
    r = code_value_int();

  if (code_seen('S')) 
    s = code_value_int();

  if (code_seen('E')) 
    e = code_value_int();

  if (code_seen('W')) 
    w = code_value_int();

  for(p=s; p<=e; p++) {
      if ( sensitive_pin(p) ) {
        SERIAL_ECHOPAIR("Sensitive Pin: ", p);
        SERIAL_ECHO(" untouched.\n");
      } else {
    	SERIAL_ECHOPAIR("Pulsing Pin: ", p);
        pinMode(p, OUTPUT);
        for(j=0; j<r; j++) {
           digitalWrite(p, 0);
           idle();
           delay(w);
           digitalWrite(p, 1);
           idle();
           delay(w);
           digitalWrite(p, 0);
           idle();
           delay(w);
        }
      }
    SERIAL_ECHO("\n");
  } 
}

/**
 * M44: Input Scan    Scan for a pin that is connected to a sensor or switch
 *
 *      This command will scan the non-sensitive pins and read them with Pull Up Resistor
 *      mode enabled.
 *
 * S	Start Pin number.   If not given, will default to 0
 *
 * E	End Pin number.   If not given, will default to 127 
 *
 * W    Wait time (in miliseconds) between scans.  If not given will default to 500
 *
 * R    Repeat range scan this many times
 */

inline void gcode_M44() {
char c;
int p, i, j, s=0, e=127, w=500, repeat_cnt=1; 

  if (code_seen('R')) 
    repeat_cnt = code_value_int();

  if (code_seen('S')) 
    s = code_value_int();

  if (code_seen('E')) 
    e = code_value_int();

  if (code_seen('W')) 
    w = code_value_int();

  for(p=s; p<=e; p++) {
      if ( sensitive_pin(p) )
	  SERIAL_PROTOCOL(".");
      else {
	  j = p % 10;
	  c = '0' + j;
	  SERIAL_PROTOCOL(c);
      }
  }
  SERIAL_PROTOCOL("\n");

  for(i=0; i<repeat_cnt; i++) {
	  for(p=s; p<=e; p++) {
	      pinMode(p, INPUT_PULLUP);
              if ( sensitive_pin(p)) {
		  SERIAL_PROTOCOL(".");
	      } else {
		  j = digitalRead( p );
		  if ( j & 0x01 )
		     SERIAL_PROTOCOL("1");
		  else
		     SERIAL_PROTOCOL("0");
	      }
      }
      SERIAL_ECHO("\n");
      idle();
      delay(w);
  }
} 
```

Bob-the-Kuhn, 2016-09-07:
I'm definitely **NOT** asking to remove anything from the sensitive pin list.

I just don't see a reason to use the list when all you're doing is reading the status of the pins.

Roxy-3D, 2016-09-07:
If you put that code into your Marlin_main.cpp,   you can enter a M43 S68 E74 command.   That will only read the status of those pins.    It will lock up the processor.    Just reading certain I/O addresses causes things to happen.  In the case of pins 68 to 74, I don't currently know what reading those addresses does.   But it does some thing serious.

thinkyhead, 2016-09-07:
It's true. With certain pins you can't even "touch" them or it disrupts the electronics. Just as in quantum physics, you can't "look" at something without interacting with it and affecting the outcome.

Bob-the-Kuhn, 2016-09-14:
I'm in favor of just closing this request.  Way more work than I envisioned.

closed 2016-09-14 by Roxy-3D

Bob-the-Kuhn, 2016-10-25:
I think I found one source of the "can't touch pins" concern.  In digitalRead they turn off the PWM before doing a read but don't turn it back on.

thinkyhead, 2016-10-26:
@Bob-the-Kuhn Aha, good to know!