MarlinFirmware/Marlin #4770: Auto Level Hot Bed problem.

genzo76, 2016-09-07:
I continue to have problems with the auto level, after many tests I realized that does not seem to consider or is treated incorrectly, the test point near X: 0 Y: 0.

The defect which I note is that the closer I get to the maximum X / Y as the head approaches the plate, while at X: 0 / Y: 0 The head is far from the bed.

This happens even when the plate is perfectly level, so if you do not use the auto level  the first layer is ok, but if I active the auto level have shown the problem.

I use the switch system with servo motor, I checked that the position is correct and that all mechanics parts is stable. For each measurement always I get the  same values.
I'm sure the problem is not mechanical, but probably there is some problem in the interpolation points.

I also tried to increase the points of the test grid, but the result does not change much.
Consider that step of the points in which the head is pressed , but on the home plate point the head is away from the plate of almost 1 mm.

If I can be of help to solve the problem just ask, I am at your disposal, even to do the tests, I'm a programmer, I don't have study your code, but if you want illustrate at me the point for  calculation, I can do tests to see if i can solve.

Thank you all.

thinkyhead, 2016-09-07:
Thanks for the report. We're looking at some fixes. Try this branch and see if it works any better:

https://github.com/thinkyhead/Marlin/tree/rc_fix_leveling_maths

genzo76, 2016-09-08:
I think I understand the problem but do not know where to put hands to solve it you can help me?

If I am not mistaken, the measurements are correct but when applied inverts the X axis (perhaps it also does with Y, but I do not know).

In practice, if the point X = 0 measurement 4 and the point X = 100 Measure 8, during the movement reverses the point X = 0 X = 100 X = 0, and then the head is raised while at X = 100 the head is pressed on the plate.

thinkyhead, 2016-09-08:
So you found the same problem exists in the branch I linked to for your benefit?

genzo76, 2016-09-10:
I tried the branch you have indicated to me, the level car seems to work, but the offset between the plate does not work, I can not adjust it.
The value that worked before now leaves 8mm away, if I try to adjust the space increases to mail decrease.

I went from -9.30 to -14, but space is increased to mail decrease. To be safe I also tried to pass the values in a positive but I get the same result, no crendo make mistakes, there may yet be some problems.

thinkyhead, 2016-09-10:
@genzo76 Please post your configuration files so I can check that they are ok.

genzo76, 2016-09-13:
This is my configuration:

#ifndef CONFIGURATION_H
#define CONFIGURATION_H

/**
 *
 *  ***********************************
 *  **  ATTENTION TO ALL DEVELOPERS  **
 *  ***********************************
 *
 * You must increment this version number for every significant change such as,
 * but not limited to: ADD, DELETE RENAME OR REPURPOSE any directive/option.
 *
 * Note: Update also Version.h !
 */
#define CONFIGURATION_H_VERSION 010100

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/delta directory.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a Scara printer replace the configuration files with the files in the
// example_configurations/SCARA directory.
//

// @section info

// User-specified version info of this build to display in [Pronterface, etc] terminal window during
// startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
// build by the user have been successfully uploaded into firmware.
#define STRING_CONFIG_H_AUTHOR "(none, default config)" // Who made the changes.
#define SHOW_BOOTSCREEN
#define STRING_SPLASH_LINE1 SHORT_BUILD_VERSION // will be shown during bootup in line 1
#define STRING_SPLASH_LINE2 WEBSITE_URL         // will be shown during bootup in line 2

//
// *** VENDORS PLEASE READ *****************************************************
//
// Marlin now allow you to have a vendor boot image to be displayed on machine
// start. When SHOW_CUSTOM_BOOTSCREEN is defined Marlin will first show your
// custom boot image and them the default Marlin boot image is shown.
//
// We suggest for you to take advantage of this new feature and keep the Marlin
// boot image unmodified. For an example have a look at the bq Hephestos 2
// example configuration folder.
//
//#define SHOW_CUSTOM_BOOTSCREEN
[... 1199 lines cut ...]
 * 34 - RAMPS1.4 - uses Analog input 5 on the AUX2 connector
 * 81 - Printrboard - Uses Analog input 2 on the Exp1 connector (version B,C,D,E)
 * 301 - Rambo  - uses Analog input 3
 * Note may require analog pins to be defined for different motherboards
 **********************************************************************/
// Uncomment below to enable
//#define FILAMENT_WIDTH_SENSOR

#define DEFAULT_NOMINAL_FILAMENT_DIA 3.00  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation

#if ENABLED(FILAMENT_WIDTH_SENSOR)
  #define FILAMENT_SENSOR_EXTRUDER_NUM 0   //The number of the extruder that has the filament sensor (0,1,2)
  #define MEASUREMENT_DELAY_CM        14   //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

  #define MEASURED_UPPER_LIMIT         3.30  //upper limit factor used for sensor reading validation in mm
  #define MEASURED_LOWER_LIMIT         1.90  //lower limit factor for sensor reading validation in mm
  #define MAX_MEASUREMENT_DELAY       20     //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

  #define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially

  //When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
  //#define FILAMENT_LCD_DISPLAY
#endif

#endif // CONFIGURATION_H

thinkyhead, 2016-09-13:
It's ok that you have `X_MIN_POS` as -5 and `Y_MIN_POS` as -16 but you probably want to set `MANUAL_X_HOME_POS` and `MANUAL_Y_HOME_POS` to 0.

thinkyhead, 2016-09-13:
I've made some changes that you can try at this branch:

https://github.com/thinkyhead/Marlin/tree/rc_final_z_correction

closed 2017-10-31 by thinkyhead