MarlinFirmware/Marlin #4772: UBL Z probe offset

OrbitAVI, 2016-09-07:
I've asked this question before, but I cannot seem to find the thread, so maybe I asked it somewhere else. Anyway I had to reset my machine to fix something. Once I powered it back on my z probe offset is no longer having any affect at all. I did not change anything in the firmware and I've had this issue before, but can't remember how to fix it.

My z probe offset is having no affect at all on my printer. I can set it to -6 and it will not move the nozzle any closer to the bed. I've cleared the mesh, inactivated the mesh, activated the mesh, reran the whole procedure etc.. nothing I do seem's to work. I can print fine if I baby step it down enough, but the z probe offset should work. Am I missing something?

Roxy-3D, 2016-09-07:
I did find a bug in the handling of the z_offset in the UBL code.   I believe I have it fixed.  And in a day or two, I'll have the updates up on GitHub for you to grab.   The problem is, I can't put them there yet because I'm in the middle of converting the code to use the new higher speed mesh_buffer_line() routine.

In the mean time, there is a viable work around.   First, turn on Z_BABYSTEPPING.  Then make sure your Z_PROBE_OFFSET_FROM_EXTRUDER is set very accurately.   Then verify your z_offset is at 0.000 with a G29 W command.    And if you do need to adjust things, you can just use the LCD Panel to raise or lower the nozzle a tiny bit.    

In a couple days, the z_offset issues should be gone.

OrbitAVI, 2016-09-07:
Yep baby stepping is what I've been doing thanks.

closed 2016-09-07 by OrbitAVI