MarlinFirmware/Marlin #4773: Z safe homing incompatible with negative X endstop pos

VanessaE, 2016-09-08:
If you have `X_MIN_POS` set to a negative value (-13.5 in my case), as with a dual extruder setup, uncommenting `Z_SAFE_HOMING` throws a compiler error.  I do not (yet) use auto-bed-leveling, I just want to keep the nozzles away from my bulldog clips.

This is Marlin from RCBugFix branch, commit 3b2f7ba22544bafb7e0e437ed06bfd66e29dc7aa (plus my settings).   Full copy of configs available on request.

```
Generating function prototypes...
"/home/vanessa/RepRap/arduino-1.6.11/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/home/vanessa/RepRap/arduino-1.6.11/hardware/arduino/avr/cores/arduino" "-I/home/vanessa/RepRap/arduino-1.6.11/hardware/arduino/avr/variants/mega" "/tmp/buildd292ea8cc2300388faf33ba75d961aab.tmp/sketch/Marlin.ino.cpp" -o "/tmp/buildd292ea8cc2300388faf33ba75d961aab.tmp/preproc/ctags_target_for_gcc_minus_e.cpp"
In file included from /tmp/buildd292ea8cc2300388faf33ba75d961aab.tmp/sketch/Marlin.ino.cpp:1:0:
Configuration.h:647: error: floating constant in preprocessor expression
 #define X_MIN_POS -13.5
                    ^
/home/vanessa/RepRap/arduino-1.6.11/hardware/arduino/avr/cores/arduino/Arduino.h:85:20: note: in definition of macro 'max'
 #define max(a,b) ((a)>(b)?(a):(b))
                    ^
/tmp/buildd292ea8cc2300388faf33ba75d961aab.tmp/sketch/Conditionals_post.h:136:28: note: in expansion of macro 'X_MIN_POS'
   #define MIN_PROBE_X (max(X_MIN_POS, X_MIN_POS + X_PROBE_OFFSET_FROM_EXTRUDER))
                            ^
/tmp/buildd292ea8cc2300388faf33ba75d961aab.tmp/sketch/SanityCheck.h:471:31: note: in expansion of macro 'MIN_PROBE_X'
   #if Z_SAFE_HOMING_X_POINT < MIN_PROBE_X || Z_SAFE_HOMING_X_POINT > MAX_PROBE_X
                               ^
Configuration.h:647: error: floating constant in preprocessor expression
 #define X_MIN_POS -13.5
                    ^
/home/vanessa/RepRap/arduino-1.6.11/hardware/arduino/avr/cores/arduino/Arduino.h:85:24: note: in definition of macro 'max'
 #define max(a,b) ((a)>(b)?(a):(b))
                        ^
/tmp/buildd292ea8cc2300388faf33ba75d961aab.tmp/sketch/Conditionals_post.h:136:39: note: in expansion of macro 'X_MIN_POS'
   #define MIN_PROBE_X (max(X_MIN_POS, X_MIN_POS + X_PROBE_OFFSET_FROM_EXTRUDER))
                                       ^
/tmp/buildd292ea8cc2300388faf33ba75d961aab.tmp/sketch/SanityCheck.h:471:31: note: in expansion of macro 'MIN_PROBE_X'
   #if Z_SAFE_HOMING_X_POINT < MIN_PROBE_X || Z_SAFE_HOMING_X_POINT > MAX_PROBE_X
                               ^
Configuration.h:647: error: floating constant in preprocessor expression
 #define X_MIN_POS -13.5
                    ^
/home/vanessa/RepRap/arduino-1.6.11/hardware/arduino/avr/cores/arduino/Arduino.h:85:28: note: in definition of macro 'max'
 #define max(a,b) ((a)>(b)?(a):(b))
                            ^
/tmp/buildd292ea8cc2300388faf33ba75d961aab.tmp/sketch/Conditionals_post.h:136:28: note: in expansion of macro 'X_MIN_POS'
   #define MIN_PROBE_X (max(X_MIN_POS, X_MIN_POS + X_PROBE_OFFSET_FROM_EXTRUDER))
                            ^
/tmp/buildd292ea8cc2300388faf33ba75d961aab.tmp/sketch/SanityCheck.h:471:31: note: in expansion of macro 'MIN_PROBE_X'
   #if Z_SAFE_HOMING_X_POINT < MIN_PROBE_X || Z_SAFE_HOMING_X_POINT > MAX_PROBE_X
                               ^
Configuration.h:647: error: floating constant in preprocessor expression
 #define X_MIN_POS -13.5
                    ^
/home/vanessa/RepRap/arduino-1.6.11/hardware/arduino/avr/cores/arduino/Arduino.h:85:32: note: in definition of macro 'max'
 #define max(a,b) ((a)>(b)?(a):(b))
                                ^
/tmp/buildd292ea8cc2300388faf33ba75d961aab.tmp/sketch/Conditionals_post.h:136:39: note: in expansion of macro 'X_MIN_POS'
   #define MIN_PROBE_X (max(X_MIN_POS, X_MIN_POS + X_PROBE_OFFSET_FROM_EXTRUDER))
                                       ^
/tmp/buildd292ea8cc2300388faf33ba75d961aab.tmp/sketch/SanityCheck.h:471:31: note: in expansion of macro 'MIN_PROBE_X'
   #if Z_SAFE_HOMING_X_POINT < MIN_PROBE_X || Z_SAFE_HOMING_X_POINT > MAX_PROBE_X
                               ^
exit status 1
floating constant in preprocessor expression
```

Blue-Marlin, 2016-09-08:
Trz ti use -13 or -14 instead of -13.5.

VanessaE, 2016-09-08:
Already set it to that, works fine so long as the value is an integer.

thinkyhead, 2016-09-08:
This is by design. We'd love it if the C++ pre-compiler could deal with floats, but it can't. If you really need to use 13.5 you'll just have to comment out the sanity check.

closed 2016-09-08 by thinkyhead

VanessaE, 2016-09-08:
No worries, it's not like half a millimeter actually matters here since I have the limits set 3mm smaller than the real size of the bed. :-)