MarlinFirmware/Marlin #4774: Silent Y-Axis creaping to endstop

sangthip, 2016-09-08:
Silent Y creaping to endstop

Ill keep this as short as possible
My printer has a fully shielded aluminium frame and fully shielded belkin cables, 
X and Y cables run up the left column and Z along the bottom
3 Extruders, Hotend, thermistor proximity sensor and fans run up a centre rear column
X,Y,Z endstops run up Right Rear Column, I am fairly certain this is not a crosstalk or stray signal problem
It is setup as a 3 Extruder mixing system.

With #define PROBE_DOUBLE_TOUCH, #define Z_MIN_PROBE_ENDSTOP,
#define Z_MIN_PROBE_PIN 32, #define Z_MIN_PROBE_REPEATABILITY_TEST,
#define ENDSTOPS_ALWAYS_ON_DEFAULT,
#define NOZZLE_PARK_FEATURE, #define NOZZLE_CLEAN_FEATURE,
#define HOME_Y_BEFORE_X,
#define SHOW_TEMP_ADC_VALUES, #define EXTRUDER_RUNOUT_PREVENT,
#define MILLISECONDS_PREHEAT_TIME 3000, #define FAN_KICKSTART_TIME 100

The problem seems to be involved with #define HOME_Y_BEFORE_X and/or #define EXTRUDER_RUNOUT_PREVENT.

Recreatable always with this process from fresh power on
Preset PLA
G28
G29
after 30odd seconds the screen no longer displays The Prepare option and the Y axis starts to slowly creep towards the max endstop (my printers homes to x0,y220,z0)
its printhead position is not updated on the screen or by querying its posn with gcode.  after 5 minutes the Y axis has travelled from Y0 to Y220 and still thinks it is at Y0, sometimes but not always if i do a g29 it crashes the printhead into the Y endstop other times it tells me to do a g28 first (the choice seems to be random, but it will always be one or the other)

I know you enjoy solving problems, I only wish I was more familiar with Marlin software , I may be able to fix it myself, but Im not !

thinkyhead, 2016-09-08:
Hey @sangthip. I hope to solve your problem as soon as possible. I'm experiencing something similar on a machine I'm working on, so I'm pretty sure there's a bug. In my case I'm seeing strange and extremely slow movements after trying to move the E axis, but I'm not using `EXTRUDER_RUNOUT_PREVENT` so perhaps the issue is different

thinkyhead, 2016-09-08:
If possible, can you try building with a different version of Arduino IDE to see if the problem persists? It's a stretch, but best to eliminate this factor. Are you using the RCBugFix version of Marlin?

sangthip, 2016-09-09:
Hi to you
I am using a version downloaded 2-Sept I think this is the latest! IDE 1.6.9, illl try a compile with 1.6.10 and 1.6.7 and see if that fixes it.  The E axis is most likly related to this problem and the firmware retract problem Im having and your E Axis problem as they all relate to E axis .. Just my thinking thou :)

thinkyhead, 2016-09-10:
Thank you for doing the test. I will look at this issue very closely over the weekend as I work on general code cleanup ahead of RC8.

boelle, 2016-11-02:
@sangthip is the issue still present in latest bugfix?

closed 2016-11-04 by thinkyhead