MarlinFirmware/Marlin #4775: Compile error while compiling the UBL branch (devel-ubl)

lrpirlet, 2016-09-08:
@Roxy-3D 
I cloned the devel-ubl and compiled it without any problem at all... So far so good.

Then I changed a few things in the configuration.h... and it does not compile... The arduino comes back with 

```
Arduino : 1.6.8 (Windows 10), Carte : "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch\G29_Unified_Bed_Leveling.cpp: In function 'void gcode_G29()':

sketch\G29_Unified_Bed_Leveling.cpp:345:45: warning: iteration 5u invokes undefined behavior [-Waggressive-loop-optimizations]

    bed_leveling_mesh.z_values[i][i] += 10.0 ;     // cells thick that is raised

                                             ^

sketch\G29_Unified_Bed_Leveling.cpp:344:10: note: containing loop

  case 1: for(i=0; i<MESH_NUM_X_POINTS; i++ ) {      // Create a diagonal line several Mesh

          ^

sketch\Marlin_main.cpp: In function 'void set_bed_level_equation_3pts(float, float, float)':

Marlin_main.cpp:2287: error: 'class Planner' has no member named 'bed_level_matrix'

       planner.bed_level_matrix.set_to_identity();

               ^

Marlin_main.cpp:2291: error: 'class Planner' has no member named 'adjusted_position'

           vector_3 uncorrected_position = planner.adjusted_position();

                                                   ^

Marlin_main.cpp:598: error: no matching function for call to 'print_xyz(const char [24], vector_3&)'

   #define DEBUG_POS(PREFIX,VAR) do{ SERIAL_ECHOPGM(PREFIX); print_xyz(" > " STRINGIFY(VAR), VAR); }while(0)

                                                                                                ^

sketch\Marlin_main.cpp:2292:11: note: in expansion of macro 'DEBUG_POS'

           DEBUG_POS("set_bed_level_equation_3pts", uncorrected_position);

           ^

sketch\Marlin_main.cpp:598:96: note: candidates are:

   #define DEBUG_POS(PREFIX,VAR) do{ SERIAL_ECHOPGM(PREFIX); print_xyz(" > " STRINGIFY(VAR), VAR); }while(0)

                                                                                                ^

sketch\Marlin_main.cpp:2292:11: note: in expansion of macro 'DEBUG_POS'

           DEBUG_POS("set_bed_level_equation_3pts", uncorrected_position);

           ^

sketch\Marlin_main.cpp:583:8: note: void print_xyz(const char*, float, float, float)

   void print_xyz(const char* prefix, const float x, const float y, const float z) {

        ^

sketch\Marlin_main.cpp:583:8: note:   candidate expects 4 arguments, 2 provided

sketch\Marlin_main.cpp:590:8: note: void print_xyz(const char*, const float*)

   void print_xyz(const char* prefix, const float xyz[]) {

        ^

sketch\Marlin_main.cpp:590:8: note:   no known conversion for argument 2 from 'vector_3' to 'const float*'

Marlin_main.cpp:2296: error: 'ABL_PROBE_PT_1_X' was not declared in this scope

       vector_3 pt1 = vector_3(ABL_PROBE_PT_1_X, ABL_PROBE_PT_1_Y, z_at_pt_1);

                               ^

Marlin_main.cpp:2296: error: 'ABL_PROBE_PT_1_Y' was not declared in this scope

       vector_3 pt1 = vector_3(ABL_PROBE_PT_1_X, ABL_PROBE_PT_1_Y, z_at_pt_1);

                                                 ^

Marlin_main.cpp:2297: error: 'ABL_PROBE_PT_2_X' was not declared in this scope

       vector_3 pt2 = vector_3(ABL_PROBE_PT_2_X, ABL_PROBE_PT_2_Y, z_at_pt_2);

                               ^

Marlin_main.cpp:2297: error: 'ABL_PROBE_PT_2_Y' was not declared in this scope

       vector_3 pt2 = vector_3(ABL_PROBE_PT_2_X, ABL_PROBE_PT_2_Y, z_at_pt_2);

                                                 ^

Marlin_main.cpp:2298: error: 'ABL_PROBE_PT_3_X' was not declared in this scope

       vector_3 pt3 = vector_3(ABL_PROBE_PT_3_X, ABL_PROBE_PT_3_Y, z_at_pt_3);

                               ^

Marlin_main.cpp:2298: error: 'ABL_PROBE_PT_3_Y' was not declared in this scope

       vector_3 pt3 = vector_3(ABL_PROBE_PT_3_X, ABL_PROBE_PT_3_Y, z_at_pt_3);

                                                 ^

Marlin_main.cpp:2307: error: 'class Planner' has no member named 'bed_level_matrix'

       planner.bed_level_matrix = matrix_3x3::create_look_at(planeNormal);

               ^

Marlin_main.cpp:2308: error: 'class Planner' has no member named 'adjusted_position'

       vector_3 corrected_position = planner.adjusted_position();

                                             ^

Marlin_main.cpp:598: error: no matching function for call to 'print_xyz(const char [22], vector_3&)'

   #define DEBUG_POS(PREFIX,VAR) do{ SERIAL_ECHOPGM(PREFIX); print_xyz(" > " STRINGIFY(VAR), VAR); }while(0)

                                                                                                ^

sketch\Marlin_main.cpp:2315:34: note: in expansion of macro 'DEBUG_POS'

         if (DEBUGGING(LEVELING)) DEBUG_POS("set_bed_level_equation_3pts", corrected_position);

                                  ^

sketch\Marlin_main.cpp:598:96: note: candidates are:

   #define DEBUG_POS(PREFIX,VAR) do{ SERIAL_ECHOPGM(PREFIX); print_xyz(" > " STRINGIFY(VAR), VAR); }while(0)

                                                                                                ^

sketch\Marlin_main.cpp:2315:34: note: in expansion of macro 'DEBUG_POS'

         if (DEBUGGING(LEVELING)) DEBUG_POS("set_bed_level_equation_3pts", corrected_position);

                                  ^

sketch\Marlin_main.cpp:583:8: note: void print_xyz(const char*, float, float, float)

   void print_xyz(const char* prefix, const float x, const float y, const float z) {

        ^

sketch\Marlin_main.cpp:583:8: note:   candidate expects 4 arguments, 2 provided

sketch\Marlin_main.cpp:590:8: note: void print_xyz(const char*, const float*)

   void print_xyz(const char* prefix, const float xyz[]) {

        ^

sketch\Marlin_main.cpp:590:8: note:   no known conversion for argument 2 from 'vector_3' to 'const float*'

exit status 1
'class Planner' has no member named 'bed_level_matrix'

Ce rapport pourrait être plus détaillé avec
l'option "Afficher les résultats détaillés de la compilation"
activée dans Fichier -> Préférences.
```

The ONLY changes I made are to the configuration.h ...
Please open https://github.com/lrpirlet/Marlin/commit/7b4087f48e96b28c84be3ea0975224d376e2fb60
to see the changes between devel-ubl and my lrp-devel-ulb branch...
(yes I know, I am dyslexic and ubl is, for me, the same as ulb)

The complete repository of the lrp-devel-ulb is https://github.com/lrpirlet/Marlin/tree/lrp-devel-ulb

Any Idea what could be wrong? I must admit that I am really lost here...

Roxy-3D, 2016-09-08:
For right now...  You need to have these values defined with reasonable numbers:
```cpp
#define UNIFIED_BED_LEVELING_GRID		// Enable the Grid Leveling portion of Unified Bed Leveling
#define UNIFIED_BED_LEVELING_GRID_POINTS 5 	// Max size of the n x n grid to be probed
```
Soon, when the Grid Based portion of UBL is included and working, we will be able to turn it on and off.   But for right now...  just leaves these #define's the way they are.

lrpirlet, 2016-09-09:
@Roxy-3D 

Damn, I guess I was trying to be too intelligent... 

Ref: #4764, this for others who would be failing the same way...my idea in commenting it was to make sure it would use a 3 point probing and NOT a grid for the tilt correction...

Maybe I should have asked why you defined it and answered that grid is not used... So obviously, there was/is a reason. I have no time to examine it now, but I sure will later. This path smells a lot like there is something I can learn from.. (thanks for that learning opportunity) 

I'll close this when it compiles successfully, or at least when it stops complaining about undefined functions... :-)

Roxy-3D, 2016-09-09:
> Maybe I should have asked why you defined it and answered that grid is not used... So obviously, there was/is a reason. I have no time to examine it now, but I sure will later. 

Well, I guess the best answer is "There are a couple of reasons why it is that way."   The first part of the answer is the goal was to "Unify all of the different bed leveling algorithms and make them all work on all machine types."   Different algorithms are turned on or off by the selections made in Configuration.h.   This project did not start with a "Clean Slate" and no bed leveling in the branch.  Instead, it started (deliberately) with the most recent  code base in RC-7 (as of July 4th).  So existing pieces of the existing auto bed leveling needed to be cut out and replaced.

The second part of the answer is the Grid Based part of UBL is not written yet and not in the code base.  But we know it is going to be there.  And it will need those #define's to control its behavior.  So, rather than delete them, they were just left in place.    What you stumbled on was a fact those #define's were used in the previous auto bed leveling scenarios to control what code got included.  And because I never turned them off since starting the UBL work, I didn't realize there was code dependent on them.

Remember, this is a 'Development Branch'.  It works.  And it shows great promise.  But it isn't fully baked yet.  There are going to be some issues like that which have to be worked through.

lrpirlet, 2016-09-09:
@Roxy-3D 
Thanks, it compiles fine with ``` #define UNIFIED_BED_LEVELING_GRID ```

I could not find any UNIFIED_BED_LEVELING_GRID_POINTS used anywhere so I left it uncommented...

Results in my branch (https://github.com/lrpirlet/Marlin/tree/lrp-devel-ulb) in the logs directory...
G28 lands exactly as I want it when geting to 0,0,0 

M100 show a HUGE reservation for global variable ( is that for the coordinate of the mesh points? if yes does it depend on the number of point probed)?

Last but not least... I admire your job, please do NOT feel guilty, of leftover in a development environment... I only ask to understand :-) 
Remember, thanks for your work...

And, count on me to ask when I feel something is wrong or needs improvement or...

Roxy-3D, 2016-09-09:
> M100 show a HUGE reservation for global variable ( is that for the coordinate of the mesh points? if yes does it depend on the number of point probed)?

Yes.  If you left the Mesh at a 10 x 10 grid, that alone eats up 400 bytes of RAM memory.  (Each float is 4 bytes)   And there are probably almost half of that again in variables in the bed leveling object.   (There are actually two copies of the state information in the bed leveling object.  One that is used, and one that just holds the pre-initialized image of the state if things need to be reset.)   If RAM gets tight, both the Mesh and the state information can be shrunk down.   But for right now, there is plenty of RAM and in a year we should be on a 32-Bit platform where memory foot print is less important.   

The storage space of the EEPROM is a similar story.   There was over 3.5 KB available in most of the AVR processors just sitting there unused.  So, I just let people store as many Mesh's as they have room to hold.  If space gets tight in the EEPROM, we can shrink the storage requirements at least in half by using fixed point numbers (stored as integers).  With any run length type compression algorithm,  probably that can be cut in half again.    But there really is no need...  7 or 8 Mesh's is kind of excessive and any normal growth of the EEPROM storage will just make one or two Mesh slots go away.   And once again, moving to a 32-Bit processor probably removes the 4KB limit on EEPROM too.

> Last but not least... I admire your job, please do NOT feel guilty, of leftover in a development environment... I only ask to understand :-)   Remember, thanks for your work...

Thanks!

closed 2016-09-12 by lrpirlet