MarlinFirmware/Marlin #4777: Build error

mperdue, 2016-09-10:
I needed to reload my printer so I loaded the Arduino package and the RC6 version that I know was working well. When I try to compile I get an error. If anyone can help me figure this out I'd be grateful.

macOS Sierra
Arduino current release
Azteeg X3 Pro

Error messages:

Arduino: 1.6.11 (Mac OS X), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

failed MSpanList_Insert 0x323f80 0x3d4f647c6438 0x0
fatal error: MSpanList_Insert

runtime stack:
runtime.MSpanList_Insert(0x2cae38, 0x323f80)
	/usr/local/go/src/runtime/mheap.c:692 +0x8f
runtime.MHeap_Alloc(0x2cad60, 0x2, 0x10000000026, 0xf989)
	/usr/local/go/src/runtime/mheap.c:240 +0x66
runtime.MCentral_CacheSpan(0x2d4158, 0x2d62f8)
	/usr/local/go/src/runtime/mcentral.c:85 +0x167
runtime.MCache_Refill(0x30c000, 0xc200000026, 0xc208012000)
	/usr/local/go/src/runtime/mcache.c:90 +0xa0

goroutine 1 [running]:
runtime.switchtoM()
	/usr/local/go/src/runtime/asm_amd64.s:198 fp=0xc2087a0ee8 sp=0xc2087a0ee0
runtime.mallocgc(0x900, 0x1a2480, 0x0, 0xc208843200)
	/usr/local/go/src/runtime/malloc.go:178 +0x849 fp=0xc2087a0f98 sp=0xc2087a0ee8
runtime.newarray(0x1a2480, 0x8, 0xa5ca)
	/usr/local/go/src/runtime/malloc.go:365 +0xc1 fp=0xc2087a0fd0 sp=0xc2087a0f98
runtime.hashGrow(0x1b69a0, 0xc20885cc90)
	/usr/local/go/src/runtime/hashmap.go:744 +0x86 fp=0xc2087a1000 sp=0xc2087a0fd0
runtime.mapassign1(0x1b69a0, 0xc20885cc90, 0xc2087a1118, 0xc2087a1108)
	/usr/local/go/src/runtime/hashmap.go:456 +0x568 fp=0xc2087a10a0 sp=0xc2087a1000
arduino.cc/properties.Map.Merge(0xc20885cc90, 0xc2087a11c8, 0x1, 0x1, 0xc2085098c0)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/properties/properties.go:165 +0x153 fp=0xc2087a1198 sp=0xc2087a10a0
arduino.cc/properties.Map.Clone(0xc2080d96e0, 0x9)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/properties/properties.go:173 +0xad fp=0xc2087a11f0 sp=0xc2087a1198
arduino.cc/builder.prepareGCCPreprocRecipeProperties(0xc208068000, 0xc2088782a0, 0x52, 0x1d8650, 0x9, 0x4, 0x0, 0x0, 0x0, 0x0)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/gcc_preproc_runner.go:112 +0xcb fp=0xc2087a12e8 sp=0xc2087a11f0
arduino.cc/builder.(*GCCPreprocRunnerForDiscoveringIncludes).Run(0xc208509920, 0xc208068000, 0x0, 0x0)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/gcc_preproc_runner.go:79 +0x82 fp=0xc2087a13f8 sp=0xc2087a12e8
arduino.cc/builder.runCommand(0xc208068000, 0x323788, 0xc208509920, 0x0, 0x0)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/container_find_includes.go:100 +0x6c fp=0xc2087a1430 sp=0xc2087a13f8
arduino.cc/builder.findIncludesUntilDone(0xc208068000, 0xc2088782a0, 0x52, 0x0, 0x0)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/container_find_includes.go:118 +0x371 fp=0xc2087a1590 sp=0xc2087a1430
arduino.cc/builder.(*ContainerFindIncludes).Run(0x2c49b0, 0xc208068000, 0x0, 0x0)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/container_find_includes.go:80 +0x8b6 fp=0xc2087a1800 sp=0xc2087a1590
arduino.cc/builder.runCommands(0xc208068000, 0xc2087a1a98, 0x22, 0x22, 0x1, 0x0, 0x0)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/builder.go:189 +0x139 fp=0xc2087a1890 sp=0xc2087a1800
arduino.cc/builder.(*Builder).Run(0xc2087a1ce0, 0xc208068000, 0x0, 0x0)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/builder.go:124 +0x1227 fp=0xc2087a1cc0 sp=0xc2087a1890
arduino.cc/builder.RunBuilder(0xc208068000, 0x0, 0x0)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/builder.go:220 +0x49 fp=0xc2087a1ce8 sp=0xc2087a1cc0
main.main()
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/arduino-builder/main.go:316 +0x171b fp=0xc2087a1f98 sp=0xc2087a1ce8
runtime.main()
	/usr/local/go/src/runtime/proc.go:63 +0xf3 fp=0xc2087a1fe0 sp=0xc2087a1f98
runtime.goexit()
	/usr/local/go/src/runtime/asm_amd64.s:2232 +0x1 fp=0xc2087a1fe8 sp=0xc2087a1fe0
/Applications/Arduino.app/Contents/Java/arduino-builder returned 2
Error compiling for board Arduino/Genuino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

mperdue, 2016-09-10:
Looks like a macOS issue...

thinkyhead, 2016-09-10:
Are you using Arduino 1.6.10? Try 1.6.9.

mperdue, 2016-09-10:
It does the same thing as 1.6.11. I also tried to reload Arduino IDE 1.6.9 from scratch after removing all the libraries and such. This all works properly on another Mac that doesn't have macOS Sierra installed. Unfortunately that Mac isn't in the room with the printer...

I'm open to other ideas if anyone has them.

mperdue, 2016-09-10:
I decided to reopen this since no one can see it otherwise. This is a bad time for me to have these problems as I'm converting my printer to a CoreXY setup and I need to be able to easily update the firmware as I make changes.

thinkyhead, 2016-09-10:
I suggest you check the Arduino forums and see how others are dealing with Sierra. Post your questions there and perhaps someone has already worked out a solution.

mperdue, 2016-09-10:
I'm waiting for the email to authorize my account there. I'll certainly post once I have it. Browsing it, I see others with the same issue but I haven't discovered a solution yet.

thinkyhead, 2016-09-11:
I will also keep [an eye on that topic](https://forum.arduino.cc/index.php?topic=423094.0).

thinkyhead, 2016-09-11:
@mperdue After you installed macOS did it prompt you to install Java?

mperdue, 2016-09-11:
No, it didn't prompt me to install java. I'll look into that tomorrow.

mperdue, 2016-09-11:
I started a topic on their support forum as well. 

https://github.com/arduino/Arduino/issues/5360

thinkyhead, 2016-09-12:
@mperdue See my post on that topic. It might require a new build of Arduino, or it might only require a rebuild of some component of macOS Sierra. I'm not sure where the Go language gets involved.

mperdue, 2016-09-16:
This issue is fixed.

https://github.com/arduino/Arduino/pull/5374

closed 2016-09-17 by thinkyhead