MarlinFirmware/Marlin #4778: UBL - Z offset doesnt seem to be applied when printing

printerguykw, 2016-09-10:
Hi,

I'm able to successfully probe a good chunk of my print bed. And when the print starts I can see the Z axis compensating according to the stored valudes.

However, I'm not sure if it's a firmware issue or G-Code issue, but when i start a print the nozzle remains at a significantly higher position than the bed.

I've set in my firmware to set Z to -1.7, I even added that value again prior to my print, however the nozzle remains at a significantly higher position than I would expect it to.

My configuration.h is pretty close to @Roxy-3D 's settings: https://github.com/printerguykw/mymarlin/blob/master/Marlin/Configuration.h

Here is my startup gcodes (full source is @ https://raw.githubusercontent.com/printerguykw/mymarlin/master/samplegcode/pikachu.gcode )

M104 S195.000000
M109 S195.000000
G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops
G29 P1 M
G26 H195 B25
G29 P4
M851 Z-1.7

Roxy-3D, 2016-09-10:
I found a bug in the way the UBL code was dealing with Z-Offsets.   I think I have it fixed, but I'm not quite ready to upload the code for distribution.   In a couple of days I expect to be able to finish testing the code and get it uploaded.   Please just live with the Z-Offset issue for a few more days.  One good work around is to get your Z-Probe-Offset-From-Extruder dialed in exactly correct.  Then you don't need the Z-Offset.   The second good work around is to enable the Z-BABYSTEPPING and you can dial in an offset when the print starts.

 (I replaced the mesh_buffer_line() routine with a faster version but that was a very invasive change.  I believe the Z-Offset is fixed in this next version of code, but I can't even try it right now because my RAMPS board died.   I've got it replaced and I'm trying to recalibrate my printer at this very moment.  So hopefully...  in a couple of days...  there will be a new and improved version of UBL for you to load.)

printerguykw, 2016-09-10:
@Roxy-3D thanks :)

epatel, 2016-09-10:
@printerguykw It looks like you are doing homing in two separate steps. Before this PR https://github.com/MarlinFirmware/Marlin/pull/3835 MBL didn't support that. It looks like this hasn't been added to UBL yet. Try do a plain `G28` instead of doing to steps.

Roxy-3D, 2016-09-10:
I missed that.   In theory, you should be OK doing the G28 in two steps.   But I can tell you that has never been tested by me.   I always do a generic, single G28.     But I do know there is a bug in the Z-Offset.   So a single G28 might not help.

Roxy-3D, 2016-09-10:
@epatel   I must have missed that thread or I most certainly would have read it.  (I try to read everything you post!)   At comment:  https://github.com/MarlinFirmware/Marlin/pull/3835#discussion_r64839180

You said:
> Yes, saving and reading active can cause it to enable MBL at an unknown position (where the sync of compensation down to the steppers "position" isn't correctly made)...so now MBL can only be activated at the end of a G28 where all axis are known (thus the MBL position can calculated and MBL can be turned on)

This is interesting!  I've never seen a problem with this.  But I'm wondering if I need to dig deeper and verify the UBL System is not vulnerable to this issue.   Maybe the reason I haven't seen an issue is the UBL System's state gets loaded at power up along with a Mesh if a Mesh Slot is active.  It may be the UBL System is always in a deterministic state and that is insulating me from what you saw.   

I'm going to keep an eye open looking for issues in this area!

epatel, 2016-09-10:
@Roxy-3D Some of these things require quite some thinking I believe. Personally I need to visualise the steps that are done. Sometimes the code is very minor but have a huge effect (thats why I am reluctant to change too many things). If you look at the code changes to add it https://github.com/MarlinFirmware/Marlin/pull/3835/files it basically removes the MBL compensation distance from the starting point (start of G28), and then if the nozzle is at a known position after the homing sequence it can be turned on again. Well, I haven't looked too deep into UBL, only saw that pre_homing_z probably wasn't in place.

epatel, 2016-09-10:
Yeah, and not seeing problems can be deceiving ;) as some problems will only come into effect if one prints close to the borders (not this though). Or not be visible at all if the bed is very flat (then MBL has very little to do). My printer is pretty bad, I needed MBL to be able to use it at all ;)

thinkyhead, 2016-09-10:
Turning MBL and other forms of leveling on and off should be a simpler matter soon. I have two functions in place in a branch I've been working on for some clients. There are two functions. In the planner we have `apply_leveling` which takes a point in the "working space" (i.e., `current_position`) and applies leveling to it before storing it in the planner and applying it to the stepper `count_position`. It applies whatever leveling system is activated.

Then there's `unapply_leveling` which reverses the operation, taking the current stepper `count_position` that represents the actual physical position of the axes, and —after using forward kinematics for Delta or SCARA— applying the inverse of the matrix to XYZ or simply subtracting the MBL or the grid-based "non-linear" Z-alteration from the position.

For any future leveling system these functions simply need to be extended. They can be tested by taking the output of one and sending it to the other, to ensure that after an application and a reversal, you always get back the point you started with.

Roxy-3D, 2016-09-10:
> If you look at the code changes to add it https://github.com/MarlinFirmware/Marlin/pull/3835/files it basically removes the MBL compensation distance from the starting point (start of G28), and then [...]

Agreed!  This is very important!  We turn off the correction for the duration of G28 and then restore its state at the end:

```cpp
void gcode_G28() {
bool UBL_state_when_G28_was_entered;	// We probably need to add some configuration control to this.  For
					// me, I do not want G28 to turn off the UBL System.   But for people
					// that use multiple pieces of glass, it is very possible they want
					// explicit control of when the UBL System is active.   
					//
					// Right now, it is nice for me because at power up (or reset) the active
					// Mesh gets loaded and the state of the UBL System is restored.   It can't
					// be on during probing, but other than that, I always want it enabled.

  #if ENABLED(DEBUG_LEVELING_FEATURE)
    if (DEBUGGING(LEVELING)) SERIAL_ECHOLNPGM(">>> gcode_G28");
  #endif

  // Wait for planner moves to finish!
  stepper.synchronize();

#ifdef UNIFIED_BED_LEVELING_FEATURE
  UBL_state_when_G28_was_entered = blm.state.active;  // Remember the state so we can restore it

  blm.state.active = 0;		// Turn off the Unified Bed Leveling System.  But
#endif						// do not invalidate the matrix.   We don't know what
						// the user's plans are yet.
```

Roxy-3D, 2016-09-13:
> thanks :) I have a bunch of spare parts sitting around as well, give me the word if you need me to ship anything printerguykw@gmail.com

@printerguykw  Thank You so much for the kind offer!   I think I have everything I need to keep making progress.   I have my printer back together again and it seems to be working.   I'm putting the finishing touches on the latest version of the UBL System right now.   There are a couple of small details I want to change but in the next day or two there will be a new version for you to load and try.

Many of the changes are transparent and were aimed at using less CPU resources.  The whole reason for that was to leave some head room so I can start the work getting the Delta's running with it.   But the biggest issue was the Z-Offset not working correctly.   I think I have that correct now.   But if there are problems, it will be much easier for me to duplicate what the user is seeing and stamp on the bugs.

printerguykw, 2016-09-13:
Thanks @Roxy-3D for working on ubl.

Roxy-3D, 2016-09-13:
I think I have the Z-Offset bug fixed.   As soon as it is check out it will be uploaded.   But in the mean time, I just realized, there is a simple way to artificially do the exact same thing.   You can just do a G29 P6 C your-z-offset     That will push or pull your mesh to what ever offset you want to put it at:

```cpp
      P6    Phase 6   Shift Mesh height.  The entire Mesh's height is adjusted by the height specified 
      		      with the C parameter.  Being able to adjust the height of a Mesh is useful tool.  It
		      can be used to compensate for poorly calibrated Z-Probes and other errors.  Ideally,
		      you should have the Mesh adjusted for a Mean Height of 0.00 and the Z-Probe measuring
		      0.000 at the Z Home location.
```

HikariNoKitsune, 2016-09-16:
Its not fixed. Just tried it. G28 itself is fixed, but after you run G29, the offset is over exaggerated. For instance: I use an offset of -1.6 and this now works with the bugfix rc that was released about 8-10 hours ago at the time of writing this. But after you run auto-leveling, it just slams the nozzle into the bed when you tell it to go to Z0..... I nearly broke off my X axis because of this. Everything seemed to be working, so when i told it to level, at the end just before it was to start the test print, it slammed the nozzle under the bed and then tried to pull the nozzle 'over' the bed...

I've tried almost everything i can think of and nothing works.... I'm about to just set an offset inside of Slic3r to automatically offset the nozzle nearly 3mm off the bed...

HikariNoKitsune, 2016-09-17:
Setting the offset inside of Slic3r worked for a single print. The very next print, the same exact thing happened yet again... So idk what is going on, but now my printer is basically just a lump of useless parts until its fixed...

drorgl, 2016-09-17:
Same thing happened to me, after digging in the code, I've found that commenting this line made the auto leveling behave better:
```
        // 4. Use the last measured distance to the bed, if possible
        if ( NEAR(current_position[X_AXIS], xProbe - (X_PROBE_OFFSET_FROM_EXTRUDER))
          && NEAR(current_position[Y_AXIS], yProbe - (Y_PROBE_OFFSET_FROM_EXTRUDER))
        ) {
          float simple_z = z_real - (measured_z - (-zprobe_zoffset));
          #if ENABLED(DEBUG_LEVELING_FEATURE)
            if (DEBUGGING(LEVELING)) {
              SERIAL_ECHOPAIR("Z from Probe:", simple_z);
              SERIAL_ECHOPAIR("  Matrix:", new_z);
              SERIAL_ECHOLNPAIR("  Discrepancy:", simple_z - new_z);
            }
          #endif
          **new_z = simple_z;**
        }
```

line 3672 https://github.com/MarlinFirmware/Marlin/blob/d41eeb62af4bfff0094a6c293761dfbfc3bf0f75/Marlin/Marlin_main.cpp

Since I'm not familiar with Marlin code, I'm not sure what side effects it has, but I've printed a part successfully after that.

It raised my suspicion after executing G29 twice and seeing the exaggerated offset double.

Roxy-3D, 2016-09-17:
@HikariNoKitsune @drorgl 

Are you running the code from the devel_ubl branch?  That code isn't even in the UBL branch.   

If you are using the UBL branch, you should be able to set your z_offset at 0.00  but move the mesh up or down using:  **G29 P6 C your-z-offset**  

And...    I've got one more issue to resolve dealing with extrusion, and when that is 'right', I'll be uploading the latest code.   (Maybe today...  But hopefully not later than tomorrow.)

drorgl, 2016-09-17:
Sorry, I didn't notice you're talking about the UBL branch, I had the same symptom on RCBugFix.

HikariNoKitsune, 2016-09-17:
To he honest idk what ubl is. I just know that I went to the rcbugfix section and typed "Z probe offset error" and it returned this thread.

Roxy-3D, 2016-09-17:
> To he honest idk what ubl is. I just know that I went to the rcbugfix section and typed "Z probe offset error" and it returned this thread.

The UBL code is an attempt to make all the various bed leveling features available to all printer types.  It is currently in a 'Development' branch.  You can find it here:  https://github.com/MarlinFirmware/Marlin/tree/devel-ubl

HikariNoKitsune, 2016-09-17:
@Roxy-3D

So that branch has the described fix from above? Just double checking lol. This bed leveling thing is frustrating me because the place I work for insisted on using the rcbugfix firmware because they believe it has all the "updated features" and I have commissions that need to be printed x.x it's been an uphill battle for me..

Roxy-3D, 2016-09-17:
I have a fix for the Z-Offset issue in my code, but it is mixed in with other changes.  But that 'fix' is for the UBL branch.   In general, 'Development' branches are not a good choice for production type equipment.   

What I would suggest is stop trying to use the Z-Offset.   Set it to 0.00 and focus on getting your Z_PROBE_OFFSET_FROM_EXTRUDER set exactly correct.  It might require you to adjust the number up or down 3 or 4 times and reflash the firmware, but it doesn't take that long.  (Remember to do a M502 and M500 each time you flash!!!)   And then...  Once things are pretty close to what you want, go ahead and turn on the Z-BABY-STEPPING.   You will be able to dynamically adjust the nozzle as it is printing the skirt at the start of the print.    

Really...   That should get things so you are not frustrated.

HikariNoKitsune, 2016-09-17:
Alright I'll give it a shot, thanks. And yea I know it's not good for production stuff but the boss is dead set on using it since the stable release is "so old" according to him... That's the real frustrating part haha. But I'll give your stuff a shot and see if it works.

Roxy-3D, 2016-09-17:
> But I'll give your stuff a shot and see if it works.

I hope you mean the idea of setting z_offset to 0.00 and getting the Z_PROBE_OFFSET_FROM_EXTRUDER exactly correct.   And then turning on Z-BABYSTEPPING.  That path should work for you.   But if you mean bringing up the devel_UBL branch, it should work also.  The problem is there will be a learning curve you have to go up and that will take time.

HikariNoKitsune, 2016-09-17:
I'll give both a shot to see. I looked before for the baby stepping but couldn't find it, so maybe I'll have to just dig deeper in. Its so odd for me tho lol. When I had my printer first built, it worked flawlessly for several months. Then it just suddenly stopped working right. Then I get told I have to use the new firmware and it starts working for a day, then starts messing up. It seems I get one "good print" at the start of the day, then after that all other prints fail horribly. 

It's also odd to me, as an offhand note, that sometimes G1 and G0 commands travel at various speeds randomly. The start of my gcode tells the printer G1 X100 Y100 and the first time it goes pretty quick. Then the rest of the day it either goes super dead slow, or moderately slow. Idk haha, I really need to convince the boss to just break and let me use stable.

Roxy-3D, 2016-09-17:
> I'll give both a shot to see. I looked before for the baby stepping but couldn't find it, 

It is in Configuration_adv.h

'''cpp
// Babystepping enables the user to control the axis in tiny amounts, independently from the normal printing process
// it can e.g. be used to change z-positions in the print startup phase in real-time
// does not respect endstops!
#define BABYSTEPPING
#if ENABLED(BABYSTEPPING)
//#define BABYSTEP_XY  //not only z, but also XY in the menu. more clutter, more functions
                       //not implemented for deltabots!
  #define BABYSTEP_INVERT_Z false  //true for inverse movements in Z
  #define BABYSTEP_MULTIPLICATOR 1 //faster movements
#endif
```

closed 2017-10-31 by thinkyhead