MarlinFirmware/Marlin #4780: Use only "raw" position in Stepper / Planner?

thinkyhead, 2016-09-10:
**Problem**: As I'm working on kinematics and leveling, the existence of coordinate space offsets seems to be adding extra work to the process of conversions.

**Background**: The two variables that can affect the coordinate space are `home_offset[]` as applied by `M206`/`M428`, and `position_shift[]` as applied by `G92`.

Historically, Marlin has always kept the stepper/planner step-based count positions in sync with the `current_position` so that if you simply divided the steps by `steps_per_mm` you would get back the position in terms of the shifted coordinate space. However, with both bed leveling and kinematics this methodology seems to add more work than if we always removed these offsets before sending them to the stepper/planner.

**For example**: when performing linear bed leveling compensation, we subtract the offsets and the tilt fulcrum position, then apply the matrix to the point, and then re-add the offsets and tilt fulcrum before storing the point in the planner and stepper block counts. If instead we always stored the "raw" position, then the coordinate space offsets would always be subtracted first as they entered the planner, and the leveling would only have to subtract and re-add the XY of the tilt fulcrum.

**Proposal**: Remove the coordinate space offsets as points are added to the planner/stepper counts. Compare to see how much this reduces and simplifies the code, as it surely will.

Roxy-3D, 2016-09-10:
> Proposal: Remove the coordinate space offsets as points are added to the planner/stepper counts. Compare to see how much this reduces and simplifies the code, as it surely will.

Actually...   I'm debugging this right now.   Originally, the Mesh Bed Leveling put the correction down in planner.buffer_line().   The problem is, I could not optimize the speed of the mesh_buffer_line() routine with the correction done at that low of level.   For moves that stay within the same X row or Y column it is possible to cut out a lot of the floating point operations because you know you are only crossing one or the other type of mesh lines.   And as a result, you know the two end points of the segment you need to interpolate position on.   Instead of 3 linear interpolations, it turns into only one.   The problem being....  You have to do the correction at a higher level.

So...  I'm debugging that right now.    It is much 'cleaner' to just have the single place where the correction is applied (way down at the bottom of planner.buffer_line() ) but this routine is too close to the hardware to know what is going on with the bigger line being broken up into sections that terminate on mesh lines.

thinkyhead, 2016-09-10:
Seems like it would be prudent to combine non-linear (delta) bed leveling with MBL since they essentially do the very same procedure. At least the correct-for-point part.

closed 2018-02-26 by thinkyhead