MarlinFirmware/Marlin #4783: UBL on corexy testing issues

adamfilip, 2016-09-11:
@Roxy-3D im trying to configure UBL on my CoreXY machine, probe is fixed type. but when i enable just the fixed probe.. it says cant have more then one type. 

i have attact my configuration.h file incase your curious

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/465949/Configuration.txt)

Roxy-3D, 2016-09-11:
This may not be a CoreXY issue.   My guess is that if it is complaining about probe types, it doesn't like fixed probes in general.   I'll try to take a look at it as soon as I can.   But it will be at least a few days before I can get setup to look at it.   (I'm in the middle of a debug session and can't change things right now.)

adamfilip, 2016-09-11:
no worries i will load up RC7 Bugfix for now.

thinkyhead, 2016-09-12:
Your problem is this:
```cpp
#define Z_ENDSTOP_SERVO_NR 0
```
You shouldn't define a Z endstop servo unless you have a servo-based Z probe.

closed 2016-09-12 by thinkyhead