MarlinFirmware/Marlin #4784: Z safe homing

adamfilip, 2016-09-11:
using RC7 Bugfix
I have enabled Z home homing, but its not moving the gantry to the middle of the bed.

my printer homes to Y max instead of Y Min, not sure if this is causing an issue

my travel limits are
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 310
#define Y_MAX_POS 300
#define Z_MAX_POS 300

// ENDSTOP SETTINGS:
// Sets direction of endstops when homing; 1=MAX, -1=MIN
// :[-1,1]
#define X_HOME_DIR -1
#define Y_HOME_DIR 1
#define Z_HOME_DIR -1

thinkyhead, 2016-09-12:
What do your Z safe homing settings look like? The defaults look like this:
```cpp
#define Z_SAFE_HOMING

#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT ((X_MIN_POS + X_MAX_POS) / 2)    // X point for Z homing when homing all axis (G28).
  #define Z_SAFE_HOMING_Y_POINT ((Y_MIN_POS + Y_MAX_POS) / 2)    // Y point for Z homing when homing all axis (G28).
#endif
```

thinkyhead, 2016-09-12:
Nevermind, I have your config from another issue. The settings seem ok.

thinkyhead, 2016-09-12:
Note that currently Z_SAFE_HOMING only applies to `G28` or `G28 X Y Z`. It doesn't apply if you use `G28 Z` to home only the Z axis.

thinkyhead, 2016-09-12:
Ah yes, I just remembered. There's a bug in `Z_SAFE_HOMING`. If your probe is outside the bed when you try to use it, it will not perform the move. The problem is this test:
```cpp
float cpx = RAW_CURRENT_POSITION(X_AXIS), cpy = RAW_CURRENT_POSITION(Y_AXIS);
if (   cpx >= X_MIN_POS - (X_PROBE_OFFSET_FROM_EXTRUDER)
    && cpx <= X_MAX_POS - (X_PROBE_OFFSET_FROM_EXTRUDER)
    && cpy >= Y_MIN_POS - (Y_PROBE_OFFSET_FROM_EXTRUDER)
    && cpy <= Y_MAX_POS - (Y_PROBE_OFFSET_FROM_EXTRUDER)) { . . .
```
To fix the issue, change the first line to:
```cpp
float cpx = RAW_X_POSITION(destination[X_AXIS]), cpy = RAW_Y_POSITION(destination[Y_AXIS]);
```

boelle, 2016-11-02:
@adamfilip was the issue fixed?

closed 2016-11-04 by thinkyhead