MarlinFirmware/Marlin #4785: Z not raising between probe instances

rektbink, 2016-09-11:
It seems that in the latest Firmware, 1.1.0-RC7 , if the Z probe travel < Z probe height then the probe will not lift in between probe instances.

thinkyhead, 2016-09-12:
Do you see this also with `RCBugFix`? Note that the settings `Z_PROBE_TRAVEL_HEIGHT` and `Z_PROBE_DEPLOY_HEIGHT` refer to the height of the nozzle. So they need to be greater than your probe Z offset. For example, if your `Z_PROBE_OFFSET_FROM_EXTRUDER` is -10 then you would need to make these values 10 plus the raise you want. Taking this example, you might then use:
```cpp
#define Z_PROBE_OFFSET_FROM_EXTRUDER -10
. . .
#define Z_PROBE_DEPLOY_HEIGHT -(Z_PROBE_OFFSET_FROM_EXTRUDER) + 10
#define Z_PROBE_TRAVEL_HEIGHT -(Z_PROBE_OFFSET_FROM_EXTRUDER) + 5
```

hexxter, 2016-11-19:
on 1.1.0-RC7 i have this settings:
```
#define Z_PROBE_OFFSET_FROM_EXTRUDER -20
. . .
#define Z_PROBE_DEPLOY_HEIGHT -(Z_PROBE_OFFSET_FROM_EXTRUDER) + 10
#define Z_PROBE_TRAVEL_HEIGHT -(Z_PROBE_OFFSET_FROM_EXTRUDER) + 5
```
but the probe does not lift between the instances. 
I use auto bed leveling with 5x5 grid. Befor it move to last row it lift a little bit the probe.

Blue-Marlin, 2016-11-19:
Likely you z-max is triggered.
Please do a M119 in mid air.

hexxter, 2016-11-19:
no z-max is not the problem (is not triggered).
I go to 
```
#define Z_PROBE_DEPLOY_HEIGHT 75
#define Z_PROBE_TRAVEL_HEIGHT 65

```
then it lift for 5mm between the points. I think it must be absolute to the homing z value. i found at a single probe the distance is about -58mm to the zero from the homing on Z Max (380mm down)

rektbink, 2016-11-19:
Your offset is -20, therefore you need to set Z probe deploy height to (+20) to essentially zero out, then you need to add the desired height.

hexxter, 2016-11-19:
i think it works not correct:
1) plain bed
```
RECV: Eqn coefficients: a: 0.00111371 b: -0.00135153 d: 58.45009613
RECV: Bed Level Correction Matrix:
RECV: +0.999999 +0.000000 +0.001114
RECV: +0.000002 +0.999999 -0.001352
RECV: -0.001114 +0.001352 +0.999998
RECV: X:348.08 Y:339.91 Z:70.16 E:0.00 Count A: 92702 B:1078 Z:56071
```
2) 4mm lift on the front side (right corner)
```
RECV: Eqn coefficients: a: -0.00320320 b: 0.00822463 d: 59.21489715
RECV: Bed Level Correction Matrix:
RECV: +0.999995 +0.000000 -0.003203
RECV: +0.000026 +0.999966 +0.008224
RECV: +0.003203 -0.008224 +0.999961
RECV: X:347.78 Y:340.58 Z:68.06 E:0.00 Count A: 92702 B:1078 Z:55795
```

thinkyhead, 2016-11-20:
Please test with `RCBugFix`. And note: the Z raise value now only refers to the amount to raise, so you no longer need to include the probe Z offset.

closed 2017-10-31 by thinkyhead