MarlinFirmware/Marlin #4786: Azteeg X3 Pro

Psyc0Flyer, 2016-09-11:
Some lines of code need added/fixed in pins_AZTEEG_X3_PRO.h in order to properly read temps for bed and hotend and MIN/MAX pin #s. All of the pin defines for MIN/MAX are for a Delta. They should be for either or so they need fixed. Here is what I added to make things work.

`//This section is to swap the MIN and MAX pins because the X3 Pro comes with only
//MIN endstops soldered onto the board. Delta code wants the homing endstops to be
//the MAX so I swapped them here.
//
#if ENABLED(DELTA)
  #undef X_MIN_PIN
  #undef X_MAX_PIN
  #undef Y_MIN_PIN
  #undef Y_MAX_PIN
  #undef Z_MIN_PIN
  #undef Z_MAX_PIN

  #define X_MIN_PIN         3
  #define X_MAX_PIN         2
  #define Y_MIN_PIN        14
  #define Y_MAX_PIN        15
  #define Z_MIN_PIN        18
  #define Z_MAX_PIN        19
#endif

#ifndef Z_MIN_PROBE_PIN
  #define Z_MIN_PROBE_PIN  19
#endif

#define E2_STEP_PIN        23
#define E2_DIR_PIN         25
#define E2_ENABLE_PIN      40

#define E3_STEP_PIN        27
#define E3_DIR_PIN         29
#define E3_ENABLE_PIN      41

#define E4_STEP_PIN        43
#define E4_DIR_PIN         37
#define E4_ENABLE_PIN      42

#undef HEATER_1_PIN
#undef HEATER_2_PIN
#undef HEATER_3_PIN
#define HEATER_1_PIN        10
#define HEATER_2_PIN        9
#define HEATER_3_PIN        16
#define HEATER_4_PIN        17
#define HEATER_5_PIN        4
#define HEATER_6_PIN        5
#define HEATER_7_PIN        6
**#define HEATER_8_PIN        11**

#undef TEMP_2_PIN
#undef TEMP_3_PIN

**#define TEMP_BED_PIN    14    
#define TEMP_0_PIN         13   
#define TEMP_1_PIN         15**   
#define TEMP_2_PIN         12   // ANALOG NUMBERING
#define TEMP_3_PIN         11   // ANALOG NUMBERING
#define TEMP_4_PIN         10   // ANALOG NUMBERING
#define TC1                 4   // ANALOG NUMBERING Thermo couple on Azteeg X3Pro
#define TC2                 5   // ANALOG NUMBERING Thermo couple on Azteeg X3Pro`

thinkyhead, 2016-09-12:
The code has already been changed in `RCBugFix`. It now defines the pins as `X_STOP_PIN`, `Y_STOP_PIN`, `Z_STOP_PIN` and they are remapped to min or max based on your `USE_[XYZ](MIN|MAX)_PLUG` settings.

Psyc0Flyer, 2016-09-12:
Ok, thank you for that info :)

-- 
Best regards!

Prometheus-3D
610-780-2714

This electronic transmission contains information which may be confidential
and/or privileged. This information is intended for use only by the
individual or entity named above and may not be shared elsewhere or
otherwise. If you are not the intended recipient (or the employee or agent
responsible for delivering this information to the intended recipient), you
are hereby notified that any use, dissemination, distribution, or copying
of this communication is prohibited. If you have received this information
in error, please notify me by electronic mail and delete all copies of the
transmission.

Thank you.

thinkyhead, 2016-09-12:
Hold the phone. I may have jumped ahead. I changed the pins for a project but haven't updated `RCBugFix` yet. I'll merge the changes shortly.

thinkyhead, 2016-09-12:
So let's review.

The current Azteeg X3 Pro pins file first includes `pins_RAMPS.h` which pre-assigns the pins for temp sensors and heaters... They are given the following values
```cpp
#define TEMP_0_PIN         13   // ANALOG NUMBERING
#define TEMP_1_PIN         15   // ANALOG NUMBERING
#define TEMP_BED_PIN       14   // ANALOG NUMBERING
. . .
#define HEATER_0_PIN     RAMPS_D10_PIN // 10
#define HEATER_1_PIN     RAMPS_D9_PIN // 9
#define HEATER_BED_PIN RAMPS_D8_PIN // 8
```
The heater pins shown above actually depend on the configuration, and they may be assigned to the PWM FAN_PIN if the configuration specifies no heated bed. The RAMPS file _**does not assign**_ `HEATER_3_PIN`, so you shouldn't override it. And take note! The heaters are numbered **starting from 0**, not starting from 1.

So currently, the X3 pins file sets additional heater pins like so:
```cpp
#define HEATER_2_PIN       16
#define HEATER_3_PIN       17
#define HEATER_4_PIN        4
#define HEATER_5_PIN        5
#define HEATER_6_PIN        6
#define HEATER_7_PIN       11
```
Your change above reassigns the 2nd, 3rd, and 4th heaters in such a way that `HEATER_1_PIN` conflicts with `HEATER_0_PIN` (both on 10). And all the other heater pins are being moved up. I'm not sure that is totally correct.

Note also: If the first three PWM heater connectors are in a different order or are on different pins, there's a more correct way (in `RCBugFix`) to reassign them. Namely, ahead of the #include line you should set the pins for the MOSFETs:
```cpp
#define RAMPS_D10_PIN ??
#define RAMPS_D9_PIN ??
#define RAMPS_D8_PIN ??

#include "pins_RAMPS.h"
```
As for the funky DELTA section… The note in the pins file says only the headers for one set are soldered. But I assume there are spots for the other three, if desired. So they're actually just being swapped here. We don't want to presume that a board with 6 endstop connectors will only have 3 ever used, so I actually decided against assigning them to `X_STOP_PIN`, etc.

thinkyhead, 2016-09-12:
Recommend you read the thread because I've edited posts after they may have been emailed.

Psyc0Flyer, 2016-09-12:
The X3 Pro has 8 spots for either fans or hotends or both. It goes from
1-8, thermistors 0-4 and a dedicated bed pin. As for the min/max, yes,
there are spots for either setup. You just have to solder connectors where
you want them.
http://files.panucatt.com/datasheets/x3pro_wiring.pdf
http://files.panucatt.com/datasheets/x3pro_pins_reva.pdf

-- 
Best regards!

Prometheus-3D
610-780-2714

This electronic transmission contains information which may be confidential
and/or privileged. This information is intended for use only by the
individual or entity named above and may not be shared elsewhere or
otherwise. If you are not the intended recipient (or the employee or agent
responsible for delivering this information to the intended recipient), you
are hereby notified that any use, dissemination, distribution, or copying
of this communication is prohibited. If you have received this information
in error, please notify me by electronic mail and delete all copies of the
transmission.

Thank you.

thinkyhead, 2016-09-12:
Right. So the original author of the pins file decided to swap the pin assignments for `DELTA` rather than force users to solder the max pins onto the board.

Psyc0Flyer, 2016-09-12:
LOL, correct

-- 
Best regards!

Prometheus-3D
610-780-2714

This electronic transmission contains information which may be confidential
and/or privileged. This information is intended for use only by the
individual or entity named above and may not be shared elsewhere or
otherwise. If you are not the intended recipient (or the employee or agent
responsible for delivering this information to the intended recipient), you
are hereby notified that any use, dissemination, distribution, or copying
of this communication is prohibited. If you have received this information
in error, please notify me by electronic mail and delete all copies of the
transmission.

Thank you.

thinkyhead, 2016-09-12:
If not using DELTA the pins are assigned in the other order.

thinkyhead, 2016-09-12:
According to the datasheet all the heater pins are already correct.

closed 2016-09-12 by thinkyhead

Psyc0Flyer, 2016-09-12:
No clue. I was up till 5am Sunday morning cause I couldnt get my readings
from bed and hotend. I changed lines of code and reloaded and it worked.

-- 
Best regards!

Prometheus-3D
610-780-2714

This electronic transmission contains information which may be confidential
and/or privileged. This information is intended for use only by the
individual or entity named above and may not be shared elsewhere or
otherwise. If you are not the intended recipient (or the employee or agent
responsible for delivering this information to the intended recipient), you
are hereby notified that any use, dissemination, distribution, or copying
of this communication is prohibited. If you have received this information
in error, please notify me by electronic mail and delete all copies of the
transmission.

Thank you.

thinkyhead, 2016-09-12:
Weird. Double-check the order of your connections.