MarlinFirmware/Marlin #4787: No SDSupport with FULL GRAPHIC Smart Controller

BVd2e, 2016-09-11:
Hey

I want to update marlin but
I have problems with verify'ing the marlin code is I able  #define SDSUPPORT
than I get a error message" 'lcd_set_custom_characters' was not declared in this scope "
(full copy error below)
 I use the Motherboard board_K8200
with a RepRapDiscount FULL GRAPHIC Smart Controller
http://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller

This setup used the work perfectly with the older version of marlin

can someone help to fix this issue ?

copy error message
-------------------------------------

Arduino: 1.6.11 (Mac OS X), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch/ultralcd.cpp: In function 'void lcd_goto_screen(screenFunc_t, bool, uint32_t)':
ultralcd.cpp:384: error: 'lcd_set_custom_characters' was not declared in this scope
         lcd_set_custom_characters(screen == lcd_status_screen);
                                                              ^
In file included from sketch/MarlinConfig.h:27:0,
                 from sketch/Marlin.h:36,
                 from sketch/ultralcd.h:26,
                 from sketch/ultralcd.cpp:23:
sketch/ultralcd.cpp: In function 'void lcd_status_screen()':
ultralcd.cpp:436: error: 'progress_bar_ms' was not declared in this scope
       if (ELAPSED(ms, progress_bar_ms + PROGRESS_BAR_MSG_TIME + PROGRESS_BAR_BAR_TIME)) {
                       ^
sketch/macros.h:114:40: note: in definition of macro 'PENDING'
 #define PENDING(NOW,SOON) ((long)(NOW-(SOON))<0)
                                        ^
sketch/ultralcd.cpp:436:11: note: in expansion of macro 'ELAPSED'
       if (ELAPSED(ms, progress_bar_ms + PROGRESS_BAR_MSG_TIME + PROGRESS_BAR_BAR_TIME)) {
           ^
ultralcd.cpp:492: error: too many arguments to function 'void lcd_implementation_init()'
       );
       ^
In file included from sketch/ultralcd.cpp:48:0:
sketch/ultralcd_impl_DOGM.h:223:13: note: declared here
 static void lcd_implementation_init() {
             ^
sketch/ultralcd.cpp: In function 'void lcd_update()':
ultralcd.cpp:2543: error: too many arguments to function 'void lcd_implementation_init()'
       );
       ^
In file included from sketch/ultralcd.cpp:48:0:
sketch/ultralcd_impl_DOGM.h:223:13: note: declared here
 static void lcd_implementation_init() {
             ^
sketch/ultralcd.cpp: In function 'void lcd_finishstatus(bool)':
ultralcd.cpp:2738: error: 'progress_bar_ms' was not declared in this scope
     progress_bar_ms = millis();
     ^
exit status 1
'lcd_set_custom_characters' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

thinkyhead, 2016-09-12:
Do you have more than one display enabled in the configuration? If you only have `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER` enabled, it should not even be including any of that code.

BVd2e, 2016-12-11:
> @BVd2e <https://github.com/BVd2e> how did you connect the display?
>
> I have this printer and just wonder how you did it as there are no
> connectors for it

On the german wiki page for the k8200 there is a guide for the wiring
http://k8200-wiki.de/Grafisches_Display

that is how i did it

boelle, 2016-12-11:
@BVd2e do you know of any ready  made adapters ?

closed 2017-10-31 by thinkyhead