MarlinFirmware/Marlin #4788: build error when enable Z_SAFE_HOMING

prophetmaster, 2016-09-11:
ccldbfoc.ltrans0.ltrans.o*: In function process_next_command()
ccldbfoc.ltrans0.o*: (.text+0xede): relocation truncated to fit: R_AVR_7_PCREL against no symbol
 
collect2.exe*: error: ld returned 1 exit status

Error compiling for board Arduino/Genuino Mega w/ ATmega2560 (Mega 2560)

thinkyhead, 2016-09-12:
What version of Arduino IDE is that? It seems to be more an internal compiler error rather than a code error.

prophetmaster, 2016-09-12:
Latest IDE 1.6.11

thinkyhead, 2016-09-12:
I presume you're testing `RCBugFix`, yes?
Please post your configuration. I haven't seen this error.

prophetmaster, 2016-09-12:
the latest release candidate 1.1.0-RC7

thinkyhead, 2016-09-13:
Great. When you have a chance, please post your configuration.

thinkyhead, 2016-09-14:
Building on Linux?

thinkyhead, 2016-09-14:
It seems to be a spurious internal compiler problem. Try moving the `process_next_command()` function to the end of `Marlin_main.cpp`, or shuffling other functions around. Eventually the compiler will magically be able to compile it without any other changes to the code. I recommend you start with the current `RCBugFix`, where we have already shuffled some functions around.

alejandro-anv, 2016-09-15:
Thank you: Just moved the declaration of codenum:

uint16_t codenum = 0; // define ahead of goto

a few lines up and it compiled ok.

alejandro-anv, 2016-09-19:
I am having the same problem but only with this config:
[config_anv.zip](https://github.com/MarlinFirmware/Marlin/files/480807/config_anv.zip)

thinkyhead, 2016-09-20:
Problem seems to be gone with the current `RCBugFix` so switch to that and see if it works for you.

alejandro-anv, 2016-09-20:
Where do I find RCBugFix? Is it in MarlinDev?

esenapaj, 2016-09-20:
Here it is.
MarlinFirmware/Marlin at RCBugFix
https://github.com/MarlinFirmware/Marlin/tree/RCBugFix

thinkyhead, 2016-09-21:
See also #4808. The compiler / linker are just trying to be too clever and failing.

boelle, 2016-11-02:
since this is invalid and a dublicate why not close?

closed 2016-11-04 by thinkyhead