MarlinFirmware/Marlin #4790: Man Mesh Levelling issue

adamfilip, 2016-09-12:
My printers homes on Y max
when I try and use Manual Mesh leveling it tries to go beyond the Y Max endstop and rams the gantry into the limit.

is there a way to switch the direction manual mesh levelling starts to probe?

thinkyhead, 2016-09-12:
That would have to be a bug someplace. MBL (G29 S1) is supposed to move to the first point on the bed by way of its XY coordinate.

adamfilip, 2016-09-12:
@Roxy-3D tried with UBL and its basically same issue.. since it homes to Y Max it just sits there at Y max
but it should move to Y min first.

epatel, 2016-09-12:
@adamfilip I can have a look-see later today, see if there is anything that stands out.

Roxy-3D, 2016-09-12:
>  tried with UBL and its basically same issue.. since it homes to Y Max it just sits there at Y max
but it should move to Y min first.

I think what this implies is this issue was in the RC-7 code base as far back as July 4th, 2016.  Because this problem really is outside of the bed leveling portion of the code base.   And that is the date I forked.   If the fix is not too invasive, I'll bring it over to the UBL branch when it is figured out.

thinkyhead, 2016-09-12:
@adamfilip If you do a simple G1 to a point after G28 all is normal?

adamfilip, 2016-09-12:
If I do a G28 then a G1 X75 Y75 it moves to the correct position

adamfilip, 2016-09-13:
Hmm disregard my last comment. Ran into a weird issue. Will elaborate soon

epatel, 2016-09-13:
@adamfilip Have been looking at the manual parts in `ultralcd.cpp`. Nothing pops out. Though, have you set a "home_offset" (done a `M206`)?

adamfilip, 2016-09-13:
I have uploaded a video to demonstrate what im experiencing.

[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/468651/Configuration_adv.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/468652/Configuration.txt)
Video Link. [https://youtu.be/G3-TqGbnxSQ](url)

Using Simplify
3d control panel. after it Homes with a G28 X0 Y0 Z0
I can issue +Y100 or +X100 moves and it moves properly
If I send a G1 X150 Y150 it rams into the endstops.

epatel, 2016-09-13:
@adamfilip Looks like Y is traveling the wrong direction then I imagine, shouldn't Y+100 go "out" and Y-100 be back into the bed?

adamfilip, 2016-09-13:
Noticed that in simplify3d I had y axis inverted.

I will try and reverse you direction on marlin and see if that corrects things

adamfilip, 2016-09-13:
after i turned off the Simpliy3d Y invert setting, I Messed around with Stepper wire orientation and marlin settings and managed to get it to move properly again. but it still doesnt home properly.

when I have Y_HOME_DIR 1 it tried to home away from the Y Max endstop
when I have Y_Home_DIR -1 it doesnt move at all, as if the endstop was already triggered. but its not (when checked with M119,)
also my Y Min endstop is disabled.

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/468898/Configuration.txt)

thinkyhead, 2016-09-13:
CoreXY? Wish you had mentioned that sooner. I'm aware that there's an issue ongoing with endstops on CORE setups, but haven't yet worked it out. The general issue seems to be that endstops are checked on the basis of raw motor direction, but with CORE systems we can't rely on that. We have to look at the combined motor directions, and possibly their relative rates of motion, to determine which direction each axis is actually moving.

thinkyhead, 2016-09-13:
Are you still using `RC6` for your testing? We're not going to be able to debug against that version. Please use `RCBugFix` so we can test against the latest code. You may or may not experience the same issue, because there have been a slew of patches since RC6, including several related to CoreXY.

adamfilip, 2016-09-13:
I will switch back to Bugfix

adamfilip, 2016-09-13:
Ok i switch back to current Bugfix. error i get now is to use "use Y min plug when homing Y to min" but my Y home DIR is -1 so it should be going to max.. not min

is there another place im not seeing that sets... home to Max vs Min?

adamfilip, 2016-09-13:
If i set Y Home Dir to 1 , error goes away but it homes in opposite direction.

thinkyhead, 2016-09-13:
> my Y home DIR is -1 so it should be going to max

No, no, no. That's exactly the opposite of true. -1 is homing to MIN (the negative direction).

> If i set Y Home Dir to 1 , error goes away but it homes in opposite direction.

That's correct for homing to MAX. Now it sounds like your XY motors are set up backward. On a CoreXY machine, the two core-axis motors need to be wired in the same way. Make triple certain that they are wired so that they coordinate properly.

adamfilip, 2016-09-14:
My machine has one stepper upside down. As I run stacked belts.  One Above and one below the gantry. Right now motion moves properly when I send job moves with control panel

Why does it give me an error for y min if it tries to home for your max?

adamfilip, 2016-09-14:
Maybe I should just add the y min endstop and save everyone the hassle lol

thinkyhead, 2016-09-14:
If you are homing Y to a MAX endstop (with `Y_HOME_DIR 1`) then you should have `USE_YMAX_PLUG` enabled.

adamfilip, 2016-09-14:
Thats what im doing

Y Max endstop is plugged in and enabled
Y Min is disabled

#define USE_XMIN_PLUG
//#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG
//#define USE_XMAX_PLUG
#define USE_YMAX_PLUG
//#define USE_ZMAX_PLUG

thinkyhead, 2016-09-14:
> Why does it give me an error for y min if it tries to home for your max?
> Thats what im doing

You said:
> error i get now is to use "use Y min plug when homing Y to min" 

If that error is appearing then you must still have `Y_HOME_DIR` set to -1. The sanity check goes like this:
```cpp
  #elif Y_HOME_DIR < 0 && DISABLED(USE_YMIN_PLUG)
    #error "Enable USE_YMIN_PLUG when homing Y to MIN."
```

adamfilip, 2016-09-14:
Sorry, ive tried so many combination of settings and stepper wire orientations that ive lost track.
Yes when I set Y_Home_DIR to -1 it give me the compile error, when I set Y_Home_DIR 1 it tried to home away from the Y Max endstop

adamfilip, 2016-09-14:
I am installing a Y Min endstop to simplify things.

adamfilip, 2016-09-14:
I think part of my problem was that I had the orientation of the Y Axis mixed up in my head.
I assumed the back of the bed was the Y Max and the front was the Y Min
but based on this image the back is the MIN and the front is the MAX
[image]

Blue-Marlin, 2016-09-14:
As soon as the bed is moving, not the nozzle, it's going to be complicated - for most of us.
The coordinate system must be right handed to work!

epatel, 2016-09-14:
@adamfilip well I think the image is fine. It is just pointing out the location for the end stops, y min trigger when the front of the bed is under the nozzle.

adamfilip, 2016-09-14:
Here is how my machine currently is setup. 
its a fixed bed CoreXY configuration

[image]

adamfilip, 2016-09-14:
> Let me add that your corexy looks very nice. Could fancy one myself :)

Thank you, its been a great project.. but also a money pit! I could have bought a pretty nice premade machine :)

thinkyhead, 2016-09-14:
> I assumed the back of the bed was the Y Max and the front was the Y Min

I'm extraordinarily relieved that you sorted it out.

closed 2016-09-14 by thinkyhead