MarlinFirmware/Marlin #4794: UBL "G29 M" behave unexpectingly under repetier host

lrpirlet, 2016-09-12:
I am using repetier host version 1.6.2 to test the UBL branch...

@Roxy-3D, I think the use of M as a parameter to G29 is not going to work smoothly with Repetier... and maybe other host software... see further

@repetier I do issue the command Gnn M... Repetier changes this and send to the printer M0Gnn... It issues a wait code...
Is that an intended behavior???

```
20:34:39.084 : ok
20:34:46.960 : N45 I will Issue G29 M*96
20:34:46.960 : echo:N45 I will Issue G29 M*96
20:34:46.960 : echo:Unknown command: "I will Issue G29 M"
20:34:46.960 : ok
```
Here I issue G29 M in the repetier "Manual Control" box labeled G-Code:
Repetier send M0G29 as you can see
```
20:34:54.445 : N49 M0G29*82
20:34:54.445 : echo:N49 M0G29*82
20:35:34.512 : Communication timeout - reset send buffer block
```
Here, everythings seems blocked... G29 shows in the display... Repetier shows "1 command waiting"
I press the button to proceed, the display clears... Sometimes Repetier comes back with Idle sometimes NOT... In this case Repetiercomes back "Idle" and allow me to issue another G-code...
```
20:35:38.153 : TEST12ok
20:35:38.184 : ok
```
I did try to replace G29 with any G command.. I get same behavior with G55 (that does NOT exist in Marlin

```
20:52:53.385 : N89 M111 S255*82
20:52:53.385 : echo:DEBUG:ECHO,INFO,ERRORS,DRYRUN,COMMUNICATION,LEVELING,SEGMENTS,ADJUSTMENTS
20:52:53.385 : ok
20:52:57.542 : N91 I will Issue G55 M*98
20:52:57.542 : echo:N91 I will Issue G55 M*98
20:52:57.542 : echo:Unknown command: "I will Issue G55 M"
20:52:57.542 : ok
20:53:06.574 : N95 M0G55*88
20:53:06.574 : echo:N95 M0G55*88
20:53:13.153 : TEST12ok
```
Note TEST12 is output by M0 in debug mode...

The "funny" thing is that when I insert V4 or W between the command and the M, the command behave as I expect it....

```
20:56:35.478 : N148 I will Issue G29 W M*43
20:56:35.478 : echo:N148 I will Issue G29 W M*43
20:56:35.478 : echo:Unknown command: "I will Issue G29 W M"
20:56:35.478 : ok
20:56:38.290 : N150 G29 W M*12
20:56:38.290 : echo:N150 G29 W M*12
20:56:38.290 : Unified Bed Leveling System Inactive.
20:56:38.290 : No Mesh Loaded.
20:56:38.306 : G29_Correction_Fade_Height : 10.00
20:56:38.306 : z_offset: 0.000000
20:56:38.306 : UBL_state_at_invokation :0
20:56:38.306 : UBL_state_recursion_chk :0
20:56:38.322 : Free EEPROM space starts at: 0x0120
20:56:38.337 : end of EEPROM              : 0FFF
20:56:38.337 : z_value[][] size: 354  220
20:56:38.353 : EEPROM free for UBL: 0x0EDF
20:56:38.368 : EEPROM can hold 0x0011 meshes.
20:56:38.384 : sizeof(stat)     :0043
20:56:38.384 : MESH_NUM_X_POINTS  11
20:56:38.384 : MESH_NUM_Y_POINTS  5
20:56:38.384 : MESH_MIN_X         0
20:56:38.384 : MESH_MIN_Y         0
20:56:38.384 : MESH_MAX_X         190
20:56:38.384 : MESH_MAX_Y         190
20:56:38.400 : MESH_X_DIST        19.000000
20:56:38.400 : MESH_Y_DIST        47.500000
20:56:38.478 : Unified Bed Leveling sanity checks passed.
20:56:38.478 : Bed Topography Report:
20:56:38.493 : (0,4)                                                                                                                                                                              (10,4)
20:56:38.509 : (0,190)                                                                                                                                                                          (190,190)
20:56:38.509 : 0.00000     0.00000     0.00000     0.00000     0.00000     0.00000     0.00000     0.00000     0.00000     0.00000     0.00000
20:56:38.509 : 
20:56:38.525 : 0.00000     0.00000     0.00000     0.00000     0.00000     0.00000     0.00000     0.00000     0.00000     0.00000     0.00000
20:56:38.525 : 
20:56:38.540 : 0.00000     0.00000     0.00000     0.00000     0.00000     0.00000     0.00000     0.00000     0.00000     0.00000     0.00000
20:56:38.540 : 
20:56:38.556 : 0.00000     0.00000     0.00000     0.00000     0.00000     0.00000     0.00000     0.00000     0.00000     0.00000     0.00000
20:56:38.556 : 
20:56:38.572 : [ 0.00000]   0.00000     0.00000     0.00000     0.00000     0.00000     0.00000     0.00000     0.00000     0.00000     0.00000
20:56:38.587 : (0,0)                                                                                                                                                                              (190,0)
20:56:38.603 : (0,0)                                                                                                                                                                                 (10,0)
20:56:38.603 : 
20:56:38.603 : ok
```

Roxy-3D, 2016-09-12:
> I am using repetier host version 1.6.2 to test the UBL branch...

@lrpirlet I have only tested using PronterFace.  The M option works flawlessly with PronterFace.  Even the weird cases where you do something like G29 P1 R M and it generates map after map.  They keep just scrolling up the screen showing you what has been discovered about the bed's topology.

> I think the use of M as a parameter to G29 is not going to work smoothly with Repetier... and maybe other host software... see further

As stated, it works flawlessly with PronterFace.  But Repetier Host is very important.  We will have to figure out what is keeping it from working there and get it fixed.

@repetier Is there anything you need from me to get this working better?   If so, just let me know.   It is possible the problem is on the Marlin side of things.   But if so, I'll need some guidance on what I should change to make it so Marlin is not getting Repetier Host angry with us.   Please let me know!!!

thinkyhead, 2016-09-13:
We can't use "M" as the letter for a parameter. Verboten.

Roxy-3D, 2016-09-13:
> We can't use "M" as the letter for a parameter. 

@thinkyhead   Can you give more detail?   I've been using that since the start of UBL and it works fine with PronterFace.    M is easy to remember for 'Map'.   So if possible, I would like to keep using it for that purpose.

thinkyhead, 2016-09-13:
M, G, and T are GCode reserved letters. The more we diverge from standard GCode, the more grief we get fro the CNC world.

Roxy-3D, 2016-09-13:
> The more we diverge from standard GCode, the more grief we get from CNC world.

Maybe.   But I think if we put it up to a vote from the people already using UBL, they are going to say they want the 'M' option to stay there.    And $#!%, take a look at the parameter (or option) list for G29.   I really do need option letters that "Make Sense".

thinkyhead, 2016-09-13:
UBL is hardly finalized, and "map" isn't the only word in the dictionary that could be applied. You'll find not a single commit from me where I chose to use `N`, `M`, `G`, etc. as parameters because I've been assiduously avoiding their use. Specifically, because I researched GCode and examined several GCode parsers to get a sense of why it is the way it is.

In essence, each of these letters indicates a "mode change" in the character stream, and in fact standard GCode doesn't even require line-breaks. I realize that RepRap G-code represents a divergence from the standard, but I think in this case we ought to defer.

> take a look at the parameter (or option) list

That's actually a really over-the-top list of options. I'm sure this will have to be seriously reformed before it will be acceptable. I don't have time to get into alternatives, but clearly it needs a re-design.

repetier, 2016-09-13:
Parses check for either M or G in commandline and match it. You can not mix. Also N is line number for every command so it also forbidden for all codes! T is an exception so you can use it as extruder for M104, also Tx is normally a command to switch tools.

I guess host/server may drop M or G or at least parse it wrong so you are provoking problems mixing M and G. Also host will expect a number after M also it would now replace all missing values by 0 I think.

lrpirlet, 2016-09-13:
@Roxy-3D 
Well, I issued "g29 m" in the G-Code box... the result is M0G29... Just for the fun I did input "hello there"... Repetier did output ``` 20:53:49.341 : echo:Unknown command: "Hello There" ```...

So, it looks like, maybe, the first letter of each word is changed to uppercase.

That said, I can still use the G29 M by writing G29 V4 M or G29 W M... This seems to work, FOR ME and probably up to when I install a next version of Repetier... So, I will never trust myself and I will chose a different approach.

If you ask my advice, I would say that I feel this command to be much too complex for a G-Code... This looks like a UNIX command with lots and lots of modifier... I have a fairly good feeling for UNIX commands, but I will NEVER be fluent enough with this G29 G-Code to use it fully... Especially given the fact that I have no -h modifier

The only thing I ask to g29 is to transform a bed, 3 axes and one tip  into a coherent system where, for chosen Z, the tip moves over the bed at a constant height relative to the bed... A set of 4 to 6 modifiers is about what I can and will remember... ( I do NOT use m48 to the full... I learned that m48 P12 V4  will give me the mean, a significant Sigma .. I know that I can move the tip first, I know that I can make the servo go up and down, I know that I can move the tip between probing... Without doc, I will NEVER trust my memory... and to be honest, when I have some time it is NOT to play with modifiers, but to print the part I want...)

Please understand that the above is MY opinion.... This is NO complaint against your choices... I will use what you produce or I will write it myself (If I can :-) )

thinkyhead, 2016-09-13:
Don't get me wrong, either. The geeks writing host software seem to be making some "brittle" choices in their parsers and don't seem to appreciate the power of regular expressions. As developers of the firmware, we should probably be providing code to the host developers to help them along – so-called "reference implementations" to demonstrate how GCode and the output of various commands should be parsed. Of course, this extra work takes time and effort, and I know we're all pretty constrained lately.

Roxy-3D, 2016-09-13:
> Well, I issued "g29 m" in the G-Code box... the result is M0G29... Just for the fun I did input "hello there"... Repetier did output 20:53:49.341 : echo:Unknown command: "Hello There"...

> So, it looks like, maybe, the first letter of each word is changed to uppercase.

OK, in the new code the Map display is also enabled with an 'O' command.  I'll let either one of them display the map.

thinkyhead, 2016-09-14:
> do it for every machine type

But it will only be installed on one machine at a time. So it's not necessary to provide every option on every machine. Likewise, I also presume that not every leveling method will be enabled at once, but just one. So again, we should keep it tight. Ideally just a bare `G29` that will probe the bed according to previous configuration, and perhaps preserving the options we already have, to specify a smaller region (although I'm disinclined to keep that because we still have to extrapolate across the entire print region).

thinkyhead, 2016-09-14:
I note that [G32](http://reprap.org/wiki/GCode#G32:_Probe_Z_and_calculate_Z_plane) is now tending towards being preferred for bed probing…

thinkyhead, 2016-09-14:
[M320](http://reprap.org/wiki/GCode#M320:_Activate_autolevel_.28Repetier.29) : Activate Autolevel
[M321](http://reprap.org/wiki/GCode#M321:_Deactivate_autolevel_.28Repetier.29) : Deactivate Autolevel
[M322](http://reprap.org/wiki/GCode#M322:_Reset_autolevel_matrix_.28Repetier.29) : Reset Autolevel Matrix
[M323](http://reprap.org/wiki/GCode#M323:_Distortion_correction_on.2Foff_.28Repetier.29) : "Distortion correction" on/off

Roxy-3D, 2016-09-14:
> M320 : Activate Autolevel
> M321 : Deactivate Autolevel
> M322 : Reset Autolevel Matrix
> M323 : "Distortion correction" on/off

At least for me...  It is much more intuitive to know G29 is used for the Auto Bed Leveling.   It is easy for me to remember a G29 A   activates it.   And it is easy for me to remember that G29 D    deactivates it.  I don't see any harm in 'over loading' G29 that way.

thinkyhead, 2016-09-14:
Customarily in G-Code "A" represents a value for the "A Axis."

Roxy-3D, 2016-09-14:
Well, GCode is fine for what it is intended to do.   But it is a very limited and constrained syntax.     And...   I'm right now, every thing is in Cartesian coordinates, even if you are on a Scara or Delta.   

Please load up the UBL System and use it.   I think you will find it has a nice feel to it and it makes sense.

thinkyhead, 2016-09-14:
Actually, I've been making some allowances for raw moves on SCARA, so you can directly set the A and B angles. For my client's firmware it's G6 with A B C. A and B are angles in degrees. C is the Z stepper. It also allows uninterpolated moves with G0, something which is fine on SCARA but which we can't allow on Delta because it produces dips in the Z position.

Roxy-3D, 2016-09-14:
> which we can't allow on Delta because it produces dips in the Z position.

If it is desirable to allow it on Delta's,  there are a couple of ways to do it and not have the dip.   I'm really not trying to be argumentative.  But here is the way I see it:   When the term 'uninterpolated' is used, it is talking about the mapping of the coordinate system for the start and end points.   It really isn't saying you can't do an intelligent trajectory to get from one to the other.   The problem with Delta's is the simple case is an arc that dips.   So...   You still don't map (interpolate) the start and end point.  But you do a 'smart' path that is physically a straight line.   One simple technique to do that is just break the line segment up into many smaller line segments (with very small dips).

thinkyhead, 2016-09-14:
>  there are a couple of ways to do it and not have the dip

Nope. You have to split up any line into small moves and interpolate. There isn't another option, because in any given straight cartesian movement, the tower carriages always move parabolically.

thinkyhead, 2016-09-14:
> …break the line segment up…

Hmm, I'm going to continue to ruminate on this in a general way. I have been thinking that for a given straight line, an individual tower carriage might move up a bit, then down a bit. But that isn't always the case, so I think we could possibly do some heuristics ahead of time. I do know for sure that they move non-linearly, gaining speed and then reducing speed. So, we might be able to build on the Bresenham technique to achieve that kind of movement.

Meanwhile, check it out. I added a hidden setting that you should try on a Delta at the next opportunity. If you enable `DELTA_FAST_SQRT` it will use the fast inverse square root algorithm (as used in Quake 3 Arena) instead of the float library's `sqrt` function. The accuracy is supposed to be pretty good. For the moment, it completely supplants the `sqrt` function in `inverse_kinematics`, but it could periodically use `sqrt` to make a correction, if it turns out to be too far off.

Roxy-3D, 2016-09-14:
> As mentioned, the tower carriages move non-linearly, gaining speed when the effector is closer to the tower, and reducing speed when the effector is farther away. So, we just might be able to build on the Bresenham technique to achieve the necessary parabolic movement. We just need to know the ratios of effector-to-tower for the start, middle, and end of any given linear movement (for all three towers).

I guess this is obvious but I didn't realize that was the cause of the dip until you pointed it out.   With that realization, here is a question:   Is the dip truly parabolic?  Because, if it is, that type of curve can be approximated very accurately with just three data points.    We already have the start and end point.   I wonder if we could have a table (coming straight out from each tower) to get the third data point.   Depending upon how close the effector is passing by the tower, we grab the third number out of a table and can exactly (and quickly) predict the expected dip????  

> Meanwhile, check it out. I added a hidden setting that you should try on a Delta at the next opportunity. If you enable DELTA_FAST_SQRT it will use the Q_rsqrt fast inverse square root algorithm (as used in Quake 3 Arena) instead of the float library's sqrt function. It's supposed to be significantly faster than sqrt — but since we need the reciprocal, there's one additional float division per tower

I'll definitely check out the faster square root algorithm.   But  I've got my head too full of stuff right now with a bug I'm trying to chase down.    When I get to the point where I start trying to bring up the UBL code on Delta's I will be in a position to both look at and use the new square root code.

esenapaj, 2016-09-14:
>`DELTA_FAST_SQRT`

It's interesting.
<details>
<summary>test program: ([reference](http://forum.arduino.cc/index.php?topic=74656.0))</summary>
```cpp
float Q_rsqrt(float number) {
        long i;
        float x2, y;
        const float threehalfs = 1.5F;

        x2 = number * 0.5F;
        y  = number;
        i  = *(long *)&y;                     // evil floating point bit level hacking
        i  = 0x5f3759df - (i >> 1);           // what the fuck?
        y  = *(float *)&i;
        y  = y * (threehalfs - (x2 * y * y)); // iteration

        return y;
}

float Q_rsqrt2(float number) {
        long i;
        float x2, y;
        const float threehalfs = 1.5F;

        x2 = number * 0.5F;
        y  = number;
        i  = *(long *)&y;                     // evil floating point bit level hacking
        i  = 0x5f3759df - (i >> 1);           // what the fuck?
        y  = *(float *)&i;
        y  = y * (threehalfs - (x2 * y * y)); // 1st iteration
        y  = y * (threehalfs - (x2 * y * y)); // 2nd iteration, this can be removed

        return y;
}

float Q_rsqrt3(float number) {
        long i;
        float x2, y;
        const float threehalfs = 1.5F;

        x2 = number * 0.5F;
        y  = number;
        i  = *(long *)&y;                     // evil floating point bit level hacking
        i  = 0x5f3759df - (i >> 1);           // what the fuck?
        y  = *(float *)&i;
        y  = y * (threehalfs - (x2 * y * y)); // 1st iteration
        y  = y * (threehalfs - (x2 * y * y)); // 2nd iteration, this can be removed
        y  = y * (threehalfs - (x2 * y * y)); // 3nd iteration, for test

        return y;
}

void setup() {
  Serial.begin(250000);
}

void loop() {
  float dummy = 0;
  uint32_t t;
  double err_tmp = 0;
  double max_err = 0;
  double ave_err = 0;
  uint32_t count = 0;

  Serial.println("Starting long calculation:");
  Serial.print("\n");  

  t = micros();
  for (uint16_t a = 0; a < 6000; a++) {
    dummy += sqrt(a);
  }
  t = micros() - t;

  Serial.print("calculation time of sqrt(): ");
  Serial.println(t);
  Serial.print("\n");

  t = micros();
  for (uint16_t a = 0; a < 6000; a++) {
    dummy += sqrtf(a);
  }
  t = micros() - t;

  Serial.print("calculation time of sqrtf(): ");
  Serial.println(t);
  Serial.print("\n");

  t = micros();
  for (uint16_t a = 0; a < 6000; a++) {
    dummy += Q_rsqrt(a);
  }
  t = micros() - t;

  for (float a = 0; a < 1000; a += 0.01) {
    count++;
    err_tmp = (float)(sqrt(a)) - 1.0 / Q_rsqrt(a);
    ave_err += err_tmp;
    if (abs(err_tmp) > max_err) {
      max_err = abs(err_tmp);
    }
  }

  Serial.print("calculation time of Q_rsqrt(): ");
  Serial.println(t);
  Serial.print("maximum error with sqrt(): ");
  if (max_err != 0)
    Serial.println(max_err, 7);
  else
    Serial.println("nodiff");
  Serial.print("average error with sqrt(): ");
  Serial.println(ave_err / count, 7);
  Serial.print("\n");  

  t = micros();
  for (uint16_t a = 0; a < 6000; a++) {
    dummy += Q_rsqrt2(a);
  }
  t = micros() - t;

  max_err = 0;
  ave_err = 0;
  for (float a = 0; a < 1000; a += 0.01) {
    err_tmp = (float)(sqrt(a)) - 1.0 / Q_rsqrt2(a);
    ave_err += err_tmp;
    if (abs(err_tmp) > max_err) {
      max_err = abs(err_tmp);
    }
  }

  Serial.print("calculation time of Q_rsqrt2(): ");
  Serial.println(t);
  Serial.print("maximum error with sqrt(): ");
  if (max_err != 0)
    Serial.println(max_err, 7);
  else
    Serial.println("nodiff");
  Serial.print("average error with sqrt(): ");
  Serial.println(ave_err / count, 7);
  Serial.print("\n");  

  t = micros();
  for (uint16_t a = 0; a < 6000; a++) {
    dummy += Q_rsqrt3(a);
  }
  t = micros() - t;

  max_err = 0;
  ave_err = 0;
  for (float a = 0; a < 1000; a += 0.01) {
    err_tmp = (float)(sqrt(a)) - 1.0 / Q_rsqrt3(a);
    ave_err += err_tmp;
    if (abs(err_tmp) > max_err) {
      max_err = abs(err_tmp);
    }
  }

  Serial.print("calculation time of Q_rsqrt3(): ");
  Serial.println(t);
  Serial.print("maximum error with sqrt(): ");
  if (max_err != 0)
    Serial.println(max_err, 7);
  else
    Serial.println("nodiff");
  Serial.print("average error with sqrt(): ");
  Serial.println(ave_err / count, 7);
  Serial.print("\n");  

  Serial.print("dummy value: ");
  Serial.println(dummy);
  Serial.println("\n");

  delay(1000);
}
```
</details>

Result of Arduino MEGA:
```
calculation time of sqrt(): 260108

calculation time of sqrtf(): 266968

calculation time of Q_rsqrt(): 314992
maximum error with sqrt(): 0.0542393
average error with sqrt(): -0.0204092

calculation time of Q_rsqrt2(): 535780
maximum error with sqrt(): 0.0001450
average error with sqrt(): -0.0000408

calculation time of Q_rsqrt3(): 741392
maximum error with sqrt(): 0.0000038
average error with sqrt(): -0.0000000

```

Result of Arduino Due:
```
calculation time of sqrt(): 158269

calculation time of sqrtf(): 55673

calculation time of Q_rsqrt(): 35595
maximum error with sqrt(): 0.0542397
average error with sqrt(): -0.0204093

calculation time of Q_rsqrt2(): 58142
maximum error with sqrt(): 0.0001452
average error with sqrt(): -0.0000408

calculation time of Q_rsqrt3(): 79701
maximum error with sqrt(): 0.0000047
average error with sqrt(): -0.0000000
```

In case of MEGA, It looks like that  `Q_rsqrt` is slower than default `sqrt()` and `sqrtf()` for some reason.
In case of Due, `Q_rsqrt` is x2 ~ x4 faster than default `sqrt()`, but `Q_rsqrt2()` and `Q_rsqrt3()` are slower than `sqrtf()`.
IDE is 1.6.12 HourlyBuild 2016/09/08 10:33.

Roxy-3D, 2016-09-14:
@esenapaj  Do you know if the Due is doing 4 byte or 8 byte floating point math?   If I remember right, the Due has a floating point co-processor built into the chip.  If that is true, this all makes sense.   The Q_rsqrt() would be a software algorithm that is having to loop and do a bunch of work to get its number calculated.

Roxy-3D, 2016-09-14:
> Also have a look at G33 which is basically Mesh Leveling. @repetier is adding support for various GCodes, so we will have to watch the RepRap Wiki and write a bot to watch their Github to keep abreast of these changes.

We already have something very similar with M421.   We can warm that over and change it to be G33 so it is exactly compliant with that description.    But really...   I haven't been using M421.   The Mesh Editor built into the UBL G29 code is much nicer to use.

thinkyhead, 2016-09-14:
@esenapaj That is interesting, alright. It sure would have been nice if `Q_rsqrt` was faster. We really need a new trick to make Delta better. My guess it that `Q_rsqrt` was faster for desktop platforms, but obviously it's not too hot on AVR.

Blue-Marlin, 2016-09-14:
@Roxy-3D Roxy
No. The Due has no FPU but much wider then 8arbit registers and a integer division in hardware.

AnHardt, 2016-09-15:
Floats are 32bit. Doubles are 32bit on the 8 bit processors and 64bit at the DUE.
https://www.arduino.cc/en/Reference/Double

esenapaj, 2016-09-15:
I've altered the above test program for more information and experiment.
<details>
<summary>altered test program:</summary>
```cpp
float Q_rsqrt(float number, uint_fast8_t n) {
  float x2, y;
  int32_t i;
  char* num_ptr;
  const float threehalfs = 1.5f;

  x2 = number * 0.5f;
  num_ptr = (char*)&number;                   // suppress warning for 32bit architecture
  i = 0x5f3759df - (*(int32_t*)num_ptr >> 1); // evil floating point bit level hacking, what the fuck?
  num_ptr = (char*)&i;                        // suppress warning for 32bit architecture
  y = *(float*)num_ptr;

  for (uint_fast8_t i = 0; i < n; i++) y *= (threehalfs - (x2 * y * y)); // iteration

  return y;
}

void setup() {
  Serial.begin(250000);
}

void loop() {
  uint_fast32_t t,
  count = 0;
  float dummy[8] = { 0.0f };
  double err_tmp = 0.0,
  max_err = 0.0,
  ave_err = 0.0;

  Serial.println("Starting long calculation:");
  Serial.print("\n");  

  t = micros();
  for (uint_fast16_t a = 0; a < 6000; a++) dummy[0] += sqrt(a);
  t = micros() - t;
  Serial.print("calculation time of sqrt(): ");
  Serial.println(t);
  Serial.print("\n");

  t = micros();
  for (uint_fast16_t a = 0; a < 6000; a++) dummy[1] += sqrtf(a);
  t = micros() - t;
  Serial.print("calculation time of sqrtf(): ");
  Serial.println(t);

  for (float a = 0.0f; a < 1000.0f; a += 0.01f) {
    count++;
    err_tmp = sqrt(a) - sqrtf(a);
    ave_err += err_tmp;
    if (abs(err_tmp) > max_err) max_err = abs(err_tmp);
  }
  Serial.print("maximum error with sqrt(): ");
  if (max_err != 0)
    Serial.println((float)max_err, 7);
  else
    Serial.println("nodiff");
  Serial.print("average error with sqrt(): ");
  Serial.println((float)(ave_err / count), 7);
  Serial.print("\n");

  for (uint_fast8_t i = 1; i <= 3; i++) {
    max_err = 0.0;
    ave_err = 0.0;

    Serial.print("Q_rsqrt() (");
    Serial.print(i);
    Serial.println(" time iteration): ");

    t = micros();
    for (uint16_t a = 0; a < 6000; a++) dummy[i * 2] += Q_rsqrt(a, i);
    t = micros() - t;
    Serial.print("calculation time of Q_rsqrt(): ");
    Serial.println(t);

    t = micros();
    for (uint16_t a = 0; a < 6000; a++) dummy[i * 2 + 1] += 1.0f / Q_rsqrt(a, i);
    t = micros() - t;
    Serial.print("calculation time of 1.0f / Q_rsqrt(): ");
    Serial.println(t);

    for (float a = 0.0f; a < 1000.0f; a += 0.01f) {
      err_tmp = sqrt(a) - 1.0f / Q_rsqrt(a, i);
      ave_err += err_tmp;
      if (abs(err_tmp) > max_err) max_err = abs(err_tmp);
    }
    Serial.print("maximum error with sqrt(): ");
    if (max_err != 0)
      Serial.println((float)max_err, 7);
    else
      Serial.println("nodiff");
    Serial.print("average error with sqrt(): ");
    Serial.println((float)(ave_err / count), 7);
    Serial.print("\n");
  }

  Serial.println("dummy value for preventing excess optimization by compiler: ");
  for (uint_fast8_t i = 0; i < 8; i++) Serial.println(dummy[i]);

  Serial.println("\n");
  delay(1000);
}
```
</details>

Result of Arduino MEGA:
```
calculation time of sqrt(): 255580

calculation time of sqrtf(): 255568
maximum error with sqrt(): nodiff
average error with sqrt(): 0.0000000

Q_rsqrt() (1 time iteration): 
calculation time of Q_rsqrt(): 337148
calculation time of 1.0f / Q_rsqrt(): 536100
maximum error with sqrt(): 0.0542393
average error with sqrt(): -0.0204092

Q_rsqrt() (2 time iteration): 
calculation time of Q_rsqrt(): 541444
calculation time of 1.0f / Q_rsqrt(): 740512
maximum error with sqrt(): 0.0001450
average error with sqrt(): -0.0000408

Q_rsqrt() (3 time iteration): 
calculation time of Q_rsqrt(): 745168
calculation time of 1.0f / Q_rsqrt(): 944372
maximum error with sqrt(): 0.0000038
average error with sqrt(): -0.0000000
```

Result of Arduino Due:
```
calculation time of sqrt(): 150328

calculation time of sqrtf(): 61020
maximum error with sqrt(): 0.0000010
average error with sqrt(): 0.0000000

Q_rsqrt() (1 time iteration): 
calculation time of Q_rsqrt(): 40406
calculation time of 1.0f / Q_rsqrt(): 61214
maximum error with sqrt(): 0.0542400
average error with sqrt(): -0.0204093

Q_rsqrt() (2 time iteration): 
calculation time of Q_rsqrt(): 64150
calculation time of 1.0f / Q_rsqrt(): 84956
maximum error with sqrt(): 0.0001452
average error with sqrt(): -0.0000408

Q_rsqrt() (3 time iteration): 
calculation time of Q_rsqrt(): 87700
calculation time of 1.0f / Q_rsqrt(): 108507
maximum error with sqrt(): 0.0000046
average error with sqrt(): -0.0000000
```
In case of MEGA, It looks like that default `sqrt()` or `sqrtf()` is best.
In case of Due, It looks like that default `sqrtf()` is best.

thinkyhead, 2016-09-17:
@esenapaj I'm looking at the possibility of using 1/2 as many `sqrt` calls and interpolating between them, rather than doing `sqrt` for every single segment. Do you think this will lead to too much error? The general idea I'm thinking of…

```cpp
float prev_delta[ABC];
for (uint16_t s = 0; s < segments; s++) {
  LOOP_XYZE(i) logical[i] += segment_distance[i];
  inverse_kinematics(logical);
  planner.buffer_line(delta[A_AXIS], delta[B_AXIS], delta[C_AXIS], logical[E_AXIS], _feedrate_mm_s, active_extruder);

  // Get the delta two steps ahead, if possible
  if (s < segments - 2) {
    LOOP_XYZE(i) logical[i] += 2 * segment_distance[i];
    inverse_kinematics(logical);
    planner.buffer_line(
      (prev_delta[A_AXIS] + delta[A_AXIS]) * 0.5,
      (prev_delta[B_AXIS] + delta[B_AXIS]) * 0.5,
      (prev_delta[C_AXIS] + delta[C_AXIS]) * 0.5,
      logical[E_AXIS], _feedrate_mm_s, active_extruder
    );
    ++s;
  }
  planner.buffer_line(delta[A_AXIS], delta[B_AXIS], delta[C_AXIS], logical[E_AXIS], _feedrate_mm_s, active_extruder);
  prev_delta[A_AXIS] = delta[A_AXIS];
  prev_delta[B_AXIS] = delta[B_AXIS];
  prev_delta[C_AXIS] = delta[C_AXIS];
}
```

thinkyhead, 2016-09-17:
Also, there are 6 float subtract operations that can easily be eliminated for every Delta segment, by converting Delta moves to "raw" coordinates in advance of segmentation.

Currently we have this:
```cpp
void inverse_kinematics(const float logical[XYZ]) {

  const float cartesian[XYZ] = {
    RAW_X_POSITION(logical[X_AXIS]),
    RAW_Y_POSITION(logical[Y_AXIS]),
    RAW_Z_POSITION(logical[Z_AXIS])
  };

  // Macro to obtain the Z position of an individual tower
  #define DELTA_Z(T) cartesian[Z_AXIS] + _SQRT( \
    delta_diagonal_rod_2_tower_##T - HYPOT2(    \
        delta_tower##T##_x - cartesian[X_AXIS], \
        delta_tower##T##_y - cartesian[Y_AXIS]  \
      )                                         \
    )

  delta[A_AXIS] = DELTA_Z(1);
  delta[B_AXIS] = DELTA_Z(2);
  delta[C_AXIS] = DELTA_Z(3);
}
```
By pre-subtracting the coordinate space offsets in `prepare_kinematic_move_to`, this can be replaced by…
```cpp
void inverse_kinematics(const float raw[XYZ]) {

  // Macro to obtain the Z position of an individual tower
  #define DELTA_Z(T) raw[Z_AXIS] + _SQRT( \
    delta_diagonal_rod_2_tower_##T - HYPOT2(    \
        delta_tower##T##_x - raw[X_AXIS], \
        delta_tower##T##_y - raw[Y_AXIS]  \
      )                                         \
    )

  delta[A_AXIS] = DELTA_Z(1);
  delta[B_AXIS] = DELTA_Z(2);
  delta[C_AXIS] = DELTA_Z(3);
}
```
This will have the greatest effect on long straight lines, but probably not too much effect on lines under ~5mm. It will depend on delta speed.

Some other questions: Are small (and slow) moves too aggressively segmented by `prepare_kinematic_move_to`? Does it make sense to split up lines into segments smaller than 0.2mm? Would vertical deviation at that scale make any noticeable difference in the print quality?

esenapaj, 2016-09-18:
>Do you think this will produce too much error?

I'm not familiar with it because it's a advanced content for me... Is this OK?
<details>
<summary>test program:</summary>
```cpp
enum AxisEnum {
  NO_AXIS = -1,
  X_AXIS  = 0,
  A_AXIS  = 0,
  Y_AXIS  = 1,
  B_AXIS  = 1,
  Z_AXIS  = 2,
  C_AXIS  = 2,
  E_AXIS  = 3,
  X_HEAD  = 4,
  Y_HEAD  = 5,
  Z_HEAD  = 6,
  ALL_AXES = 100
};
#define LOOP_XYZE(VAR) for (uint8_t VAR=X_AXIS; VAR<=E_AXIS; VAR++)
#define NUM_AXIS 4
#define XYZE 4
#define ABC  3
#define XYZ  3
volatile float position_shift[XYZ] = { 0 },
               home_offset[XYZ] = { 0 };
#define RAW_POSITION(POS, AXIS)     (POS - home_offset[AXIS] - position_shift[AXIS])
#define RAW_X_POSITION(POS)         RAW_POSITION(POS, X_AXIS)
#define RAW_Y_POSITION(POS)         RAW_POSITION(POS, Y_AXIS)
#define RAW_Z_POSITION(POS)         RAW_POSITION(POS, Z_AXIS)
#define RAW_CURRENT_POSITION(AXIS)  RAW_POSITION(current_position[AXIS], AXIS)
#define UNEAR_ZERO(x) ((x) < 0.000001)
volatile int feedrate_percentage = 100;
#define MMM_TO_MMS(MM_M) ((MM_M)/60.0)
#define MMS_SCALED(MM_S) ((MM_S)*feedrate_percentage*0.01)
volatile static float feedrate_mm_s = MMM_TO_MMS(1500.0);
#define ARG_X float lx
#define ARG_Y float ly
#define ARG_Z float lz
#define HYPOT2(x,y) (sq(x)+sq(y))
#define HYPOT(x,y) sqrt(HYPOT2(x,y))
#define NOLESS(v,n) do{ if (v < n) v = n; }while(0)
volatile static float destination[NUM_AXIS] = { 0.0 };
volatile float offset[NUM_AXIS] = { 0.0 },
               target[NUM_AXIS], // Destination position
               current_position[NUM_AXIS] = { 0.0 },
               radius = HYPOT(offset[X_AXIS], offset[Y_AXIS]),
               center_X = current_position[X_AXIS] + offset[X_AXIS],
               center_Y = current_position[Y_AXIS] + offset[Y_AXIS],
               linear_travel = target[Z_AXIS] - current_position[Z_AXIS],
               r_X = -offset[X_AXIS],  // Radius vector from center to current location
               r_Y = -offset[Y_AXIS],
               rt_X = target[X_AXIS] - center_X,
               rt_Y = target[Y_AXIS] - center_Y,
               angular_travel = atan2(r_X * rt_Y - r_Y * rt_X, r_X * rt_X + r_Y * rt_Y),
               mm_of_travel = HYPOT(angular_travel * radius, fabs(linear_travel)),
               segment_distance[NUM_AXIS] = { 0.0 },
               delta[ABC] = { 0.0 };
#define MM_PER_ARC_SEGMENT 1
volatile uint16_t segments = floor(mm_of_travel / (MM_PER_ARC_SEGMENT));
volatile uint8_t active_extruder = 0;
#define DELTA_DIAGONAL_ROD 250.0 // mm
#define DELTA_RADIUS_TRIM_TOWER_1 0.0
#define DELTA_RADIUS_TRIM_TOWER_2 0.0
#define DELTA_RADIUS_TRIM_TOWER_3 0.0
#define DELTA_DIAGONAL_ROD_TRIM_TOWER_1 0.0
#define DELTA_DIAGONAL_ROD_TRIM_TOWER_2 0.0
#define DELTA_DIAGONAL_ROD_TRIM_TOWER_3 0.0
#define DELTA_SEGMENTS_PER_SECOND 200
[... 317 lines cut ...]
average error with prepare_kinematic_move_to() (not optimized):
delta[A_AXIS] -0.0000002
delta[B_AXIS] -0.0000001
delta[C_AXIS] 0.0000001

calculation time of prepare_kinematic_move_to() (USE_DELTA_IK_INTERPOLATION enabled): 17734760
maximum error with prepare_kinematic_move_to() (not optimized):
delta[A_AXIS] 0.0000153
delta[B_AXIS] 0.0000305
delta[C_AXIS] 0.0000305
average error with prepare_kinematic_move_to() (not optimized):
delta[A_AXIS] -0.0000003
delta[B_AXIS] -0.0000002
delta[C_AXIS] 0.0000002

calculation time of prepare_kinematic_move_to() (USE_RAW_KINEMATICS USE_DELTA_IK_INTERPOLATION enabled): 16438640
maximum error with prepare_kinematic_move_to() (not optimized):
delta[A_AXIS] 0.0000153
delta[B_AXIS] 0.0000305
delta[C_AXIS] 0.0000305
average error with prepare_kinematic_move_to() (not optimized):
delta[A_AXIS] -0.0000003
delta[B_AXIS] -0.0000002
delta[C_AXIS] 0.0000002
```

esenapaj, 2016-09-18:
>Are small (and slow) moves too aggressively segmented by prepare_kinematic_move_to?
>Does it make sense to split up lines into segments smaller than 0.2mm?
>Would vertical deviation at that scale make any noticeable difference in the print quality?

Sorry, I haven't knowledge for answering about those at this time...

thinkyhead, 2016-09-18:
From 23,953,812µs down to 16,438,640µs saves 7,515,172µs, or 7.5 whole seconds. Or, about 32% faster. Not too bad.

lrpirlet, 2016-09-23:
@thinkyhead @Roxy-3D @esenapaj 
This thread "slipped" away from the original subject... This is fine, as I did learn some , but I will let anyone of you close it, because I am not too sure that the subject has been exhausted...

Roxy-3D, 2016-09-23:
Yeah...   That does seem to happen!  :)     I think we have resolution on what caused the problem with the 'M' (Map) function, and I already have the system changed to accept either a lower case 'm' or an upper case 'O'.   So...  The option can be used on Repetier Host now.

closed 2016-09-23 by Roxy-3D

repetier, 2016-09-24:
" I already have the system changed to accept either a lower case 'm' or an upper case 'O'"

O is fine, but comment on lower case m: users are lazy and often write lower letters meaning someone will fix it and since older firmwares did not fix, host uppercases gcode letters normally if nor forced to send original command. As I understand GCode specification all letters were supposed uppercase anyway so relying on upper/lower case is a dangerous path.

thinkyhead, 2016-09-24:
I concur. No lower-case, please. That's just not good. I shouldn't need to explain why.

Roxy-3D, 2016-09-25:
Alright..    I agree...  Let's not cause any problems needlessly!