MarlinFirmware/Marlin #4795: UBL G29 P1 does NOT raise between move

lrpirlet, 2016-09-12:
@Roxy-3D My probe is a switch ( smb ) that a servo brings in between the hot tip and the bed...

Sorry the pictures refuses to rotate, please rotate 90° clockwise...

[image]
 is the probe retracted...

[image]
 is the probe engaged...

The issue is that the thing that hold the swith is very light (it needs to act as a tiny spring to close on the tip)... 
That thing is ripped of with the smallest force... 
I could probe 4 points before it was forced away... 
The sequence is 
- go down to the bed,
- trigger the switch,
- move in the XY plqne,
- raise away from the bed along z,
- probe down...

Is there a way to set configure.h so that I have:
- go down to the bed,
- trigger swith,
- raize along Z,
- move in the raised XY plane,
- probe down

Should I modify the code?

marobol, 2016-09-12:
i have a same problem, but if i try with #define Z_MIN_PROBE_ENDSTOP in Config.h, the sequence for auto level is ok, but can´t make homing.
I need help to...

lrpirlet, 2016-09-12:
BTW, the parameters documented in the G29_Unified_Bed_Leveling.CPP misses the V modifier... The "V" modifier is in the code :-)...

Roxy-3D, 2016-09-12:
@thinkyhead    I wonder if this has the same root issue as #4790 ?   I'm just using the homing and probing code as it was July 4th, 2016 with no changes.

Roxy-3D, 2016-09-12:
@lrpirlet 
I'm not positive, and @thinkyhead can correct me if this not true, but I think the magnitude of these numbers:
```cpp
#define Z_RAISE_PROBE_DEPLOY_STOW 5 // Raise to make room for the probe to deploy / stow
#define Z_RAISE_BETWEEN_PROBINGS 2  // Raise between probing points.
```
must be greater than your
```cpp
#define Z_PROBE_OFFSET_FROM_EXTRUDER -4.33   // Z offset: -below +above  [the nozzle]
```
Please try changing that 2 to a 5 and see what happens.  In fact, it might be best to go with 7mm just because 5 will only give you less than 1mm of space as you move.

marobol, 2016-09-12:
i was put
// Use with caution and do your homework.
//#define Z_MIN_PROBE_ENDSTOP 
//SI HABILITO ESTE FUNCIONA BIEN PERO NO BAJA EL SERVO CUANDO HACE EL G28
//IF ENEABLE define Z_MIN_PROBE_ENDSTOP  THE AUTO LEVEL IS OK BUT SERVO NOT MOVE THE SWITCH FOR HOMING...

// Enable Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN to use the Z_MIN_PIN for your Z_MIN_PROBE.
// The Z_MIN_PIN will then be used for both Z-homing and probing.
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN 
// SI HABILITO ESTA ARRASTRA EL SWITCH AL NIVELAR LA CAMA
// IF  ENEABLE #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN, THE SWITCH RUN ON THE ALL BED STILL AUTOLEVEL FINISH.

// To use a probe you must enable one of the two options above!

// This option disables the use of the Z_MIN_PROBE_PIN
// To enable the Z probe pin but disable its use, uncomment the line below. This only affects a
// Z probe switch if you have a separate Z min endstop also and have activated Z_MIN_PROBE_ENDSTOP above.
// If you're using the Z MIN endstop connector for your Z probe, this has no effect.
//#define DISABLE_Z_MIN_PROBE_ENDSTOP

// Enable Z Probe Repeatability test to see how accurate your probe is

#define Z_MIN_PROBE_REPEATABILITY_TEST

//
// Probe Raise options provide clearance for the probe to deploy, stow, and travel.
//
#define Z_PROBE_DEPLOY_HEIGHT 5 // Raise to make room for the probe to deploy / stow
#define Z_PROBE_TRAVEL_HEIGHT 2  //5 Raise between probing points.

//
// For M851 give a range for adjusting the Z probe offset
//
#define Z_PROBE_OFFSET_RANGE_MIN -20
#define Z_PROBE_OFFSET_RANGE_MAX 20

Roxy-3D, 2016-09-12:
> i was put

I don't understand what that means or what I'm supposed to infer from that comment.   You will have to try again with more text.

marobol, 2016-09-12:
in this file, i c somethid diferences, but i think that is an old version...
https://github.com/lrpirlet/Marlin/blob/lrp-devel-ulb/Marlin/Configuration.h

marobol, 2016-09-12:
I that is a bug, i try found in the source code what diference when we define:
#define Z_MIN_PROBE_ENDSTOP  or #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN 
because with #define Z_MIN_PROBE_ENDSTOP the auto level sequence is ok..

marobol, 2016-09-12:
i found a solution for this... you need send this code G29 E to retract de switch

Roxy-3D, 2016-09-12:
> i found a solution for this... you need send this code G29 E to retract de switch

I think we have some confusion here.    The UBL Auto Bed Leveling System does not currently have an E option available.    The title of this thread is: UBL G29 P1 does NOT raise between move

marobol, 2016-09-12:
If you like change this , you need replace this code parts: (line 3455 of Marlin_main.cpp)
    // bool  stow_probe_after_each = code_seen('E'); // put rem at this line and...
    // +++++++++++++++++*************************************************
    bool stow_probe_after_each; // Así queda fijo el comando que levanta e switch en cada medición
    if (code_seen('E')||code_seen('e')){
     stow_probe_after_each = code_seen('E');
    }
    else{
     stow_probe_after_each = true; 
    }  //++++++++++++++++++++++++++++++++++++++++ fin

lrpirlet, 2016-09-13:
@Roxy-3D OK, Thanks, I'll try that... 

The implication of your advice is that this Z_PROBE_TRAVEL_HEIGHT variable has become absolute from a relative displacement... The version I used (RC6+ a few PRs) was relative displacement to add from the switch trigger height... I only copied my values from there...

I guess I need to search the issues to verify that thinking... Then I should read the Configuration.h to see if I missed some explanatory comments...

Will try that later tonight if (or later when) I get time

Thanks again for your help attitude...

thinkyhead, 2016-09-13:
> The implication of your advice is that this `Z_PROBE_TRAVEL_HEIGHT` variable has become absolute

Not just implication. It was _explicitly_ renamed from `*_RAISE_*` to indicate this new behavior. I agree, it no longer ensures the raise will always be over the probe height. But @AnHardt especially seems to think the new setting is clearer than the old one.

Roxy-3D, 2016-09-13:
> But in fact there has been a great increase in complaints and apparent confusion.

Yes, but people have been trained for 2 years that *_RAISE_* goes up from the current location.   Probably a few lines of comments in the Configuration.h file immediately above the new *_HEIGHT_* parameters will help people do the 'right' thing.

lrpirlet, 2016-09-13:
@Roxy-3D @thinkyhead 

Thanks, latest configuration.h in RCBugFix says:
```C++
//
// Minimum heights for the probe to deploy/stow and travel.
// These values specify the distance from the NOZZLE to the BED.
//
#define Z_PROBE_DEPLOY_HEIGHT 15 // Z position for the probe to deploy/stow
#define Z_PROBE_TRAVEL_HEIGHT  5 // Z position for travel between points
```
That should be clear... 

Yet I would have written ```// Z absolute  position for the probe to deploy/stow``` for people like me who are too much in a hurry... 
I read to the end of the line (most often) but 'hey, do not hope that I will read 2 lines above when I have modified that configuration file at least 24982 times already" 😃 

Sorry, joking but real.... I am not a good reader when I know (I think I know)

Will test the proposed solution later, no time today...

thinkyhead, 2016-09-14:
> Yes, but people have been trained for 2 years that _RAISE_ goes up from the current location.

Hence the reason it was changed to _HEIGHT_ — as already pointed out.

thinkyhead, 2016-09-14:
We can't catch this at compile time, and besides it also depends on the M851 setting. So either we revert to the old methodology or add code to catch the issue at runtime in `do_probe_raise`…
```cpp
if (zprobe_zoffset < 0 && z_dest < -zprobe_zoffset) {
  SERIAL_ERROR_START;
  SERIAL_ECHOPGM(MSG_ERR_PROBE_HEIGHT);
  LCD_MESSAGEPGM(MSG_ERR_PROBE_HEIGHT_LCD);
  zdest = -zprobe_zoffset + 5;
  BUZZ(500, 220);
}
```

thinkyhead, 2016-09-14:
As another option, make the defaults more explicit:
```cpp
//
// Minimum heights for the probe to deploy/stow and travel.
// These values specify the distance from the NOZZLE to the BED.
//
#define Z_PROBE_DEPLOY_HEIGHT min(-Z_PROBE_OFFSET_FROM_EXTRUDER, 0) + 10 // Z for Deploy/Stow
#define Z_PROBE_TRAVEL_HEIGHT min(-Z_PROBE_OFFSET_FROM_EXTRUDER, 0) + 5  // Z between probe points
```

Roxy-3D, 2016-09-14:
> As another option, make the defaults more explicit:
```cpp
//
// Minimum heights for the probe to deploy/stow and travel.
// These values specify the distance from the NOZZLE to the BED.
//
#define Z_PROBE_DEPLOY_HEIGHT max(-Z_PROBE_OFFSET_FROM_EXTRUDER, 0) + 10 // Z for Deploy/Stow
#define Z_PROBE_TRAVEL_HEIGHT max(-Z_PROBE_OFFSET_FROM_EXTRUDER, 0) + 5  // Z between probe points
```
My vote would be that we do it this way.  The one change I would suggest is let's hide the extra complexity in Conditionals.h or SanityCheck.h   Let's keep the expression in Configuration.h just a simple number.  That has the best chance of the user understanding things and helping us do the 'right' thing.

thinkyhead, 2016-09-14:
>  let's hide the extra complexity in Conditionals.h or SanityCheck.h

Not an option. We might as well then revert to the old method. Which I am most inclined to do at this point anyway.

Roxy-3D, 2016-09-14:
I'm not really pushing hard on this...   But please explain why it is not an option?   I'm just trying to understand.   

From my perspective most of the problems that users have where they end up with non-functional configurations is because they don't understand the big picture.   Keeping these #define's very simple and just a number would help the user make intelligent choices for the number they put there.

thinkyhead, 2016-09-14:
@Roxy-3D Think about it. If the user sets "height" values and then we "hide" the actual value by sticking the `max` macro on it, adding their "height" value to it, then it's no longer a "height" value. It's now a "raise" value. So it's already reverting the behavior. Therefore, if that's the proposal, we might as well rename them as "raise" values and automatically add the Z probe offset in the `do_probe_raise` function, as it was before.

lrpirlet, 2016-09-14:
@thinkyhead 
Please do NOT make yet another change without some more thinking please... People are starting to learn... If we revert now, we will confuse all who hit that variable already for the benefit of a few that have NOT seen the change...

What we should do, IMHO, is observing from a height all the variables used... If the majority of the variable are absolute then we introduce the concept of absolute everywhere. Writing in the configuration.h at the begin that all variables are absolute  will push people to have an educated guess rather than push them to think deeply at every steps... 
(Replace absolute with relative if the finding points to relative)
Sorry, I have no time to do that before a few weeks... I would need to plan 2 to 4 hours...

I do NOT want to talk for anyone, but I perceive that this is probably @AnHardt thinking...

AnHardt, 2016-09-14:
Please stay with height.
'Z_PROBE_OFFSET_FROM_EXTRUDER' is simply wrong here.
'Z_PROBE_OFFSET_FROM_EXTRUDER' is the difference between nozzle tip and the point where the probe triggers, For any mechanical switch this is not the same as the difference between the nozzles tip and the point where the switch has no contact with the bed anymore!
We could make a new constant with the right amount - but would this make sense? Would it simplify things?

lrpirlet, 2016-09-14:
@Roxy-3D @thinkyhead 
Well, The discussion we had about the variable was sort of... SKEWED :-) even if very interresting...

The ULB configuration.h file does NOT contains any ```Z_PROBE_TRAVEL_HEIGHT```  but only a  ```Z_RAISE_BETWEEN_PROBINGS ``` ...
it reads:
```C++
//
// Probe Raise options provide clearance for the probe to deploy, stow, and travel.
//
#define Z_PROBE_DEPLOY_HEIGHT 5 // Raise to make room for the probe to deploy / stow
#define Z_PROBE_TRAVEL_HEIGHT 2  //5 Raise between probing points.
```

LOL

So even if I had read, I would have set it wrong....

The nice thing is that the solution proposed by @Roxy-3D works perfectly. Thanks. I used 7 as the value.

For the small story, I tried to be smart and first set
 ``` #define Z_RAISE_BETWEEN_PROBINGS Z_PROBE_OFFSET_FROM_EXTRUDER + 2) ``` but the compiler came back with a message to the effect that the preprocessor cannot handle floating point... Old known problem, I had it already when setting X and Y offset in the classic bed leveling approach... 

NB, I would close this, I left it open in case someone wants to add anything...

thinkyhead, 2016-09-14:
@AnHardt I respectfully disagree that "height" values are in any sense better. It's far too easy to overlook the relationship of the setting made by `M851` or the default Z probe distance, and set too low a value. We've had a good dozen confused users needing to have this explained. We tried using a height value, and the result was confusion. There will continue to be confusion, and I for one am tired of having to explain to users that they need to be careful and attentive and understand the relationship of these settings. Frankly, we should make it so that you don't have to take a master course in Marlin configuration to get your probe to raise high enough to clear the bed.

Roxy-3D, 2016-09-14:
>> Please stay with height.
> @AnHardt Why? Seriously. I respectfully disagree that "height" values are in any sense better. It's far too easy to overlook the relationship of the setting made by M851 or the default Z probe offset setting, and virtually everyone sets too low a value, thinking they can just transfer their old settings.

I think the answer is 'height' is absolute and 'raise' is relative.  An absolute number gets rid of one level of indirection and is more likely to produce the correct result.

thinkyhead, 2016-09-14:
What users of 1.0.x currently set is "how far Z will raise from where it is" which is the relative raise amount. They really don't care that the NOZZLE will raise to the specific height Z=20.0. They just want there to be enough clearance, and they want it to be adaptable to their `M851` settings.

lrpirlet, 2016-09-15:
@thinkyhead 
Reading this exchange, I have the feeling that we see here more a language problem than anything else... Could we name this variable completely differently.

If we chose to go absolute, we could call it something like **ABSOLUTE_Z_HEIGHT_TO_TRAVEL WITH_PROBE_ENGAGED** instead of ```Z_PROBE_TRAVEL_HEIGHT```

If we chose to go relative, we could call it something like **RELATIVE_Z_CLEARANCE_BETWEEN_PROBINGS** instead of ```Z_RAISE_BETWEEN_PROBINGS```

To be quite honest I would prefer the **relative** proposal.. 

I hate the idea of absolute when I have NO IDEA what is the ZERO reference position...

So, when the bed is tilted... 
I can estimate the tilt an decide that the maximum difference is about 7 mm... I would then use a relative 8 mm to lift the probe... and that would work

On the other hand, what will be the absolute height needed when my G28 goes to some obscure calculated location in the middle of the bed to find a ZERO reference??? would it be 7+3,5 or 7-3,5 to add  to the negated value of my Z_PROBE_OFFSET_FROM_EXTRUDER ??? 
Sorry too complex... 
So I'll end up with a finger on the big red button pressing to early at least twice before to be confident (or crazy) enough ;-)

thinkyhead, 2016-09-16:
@lrpirlet The previous behavior was to take the position where a probe triggered (i.e., the `Z_PROBE_OFFSET_FROM_EXTRUDER` or `M851 Z` value) and apply the raise over and above that. Thus, if you had a servo-based probe, then even with the raise configured to be zero you were guaranteed to have (just about) enough clearance for the probe to deploy. In other words, the minimum height would be `-zprobe_zoffset`. Naturally, with a servo probe, the clearance for deploying it needs to be greater than zero. So users naturally set their "raise" values to something like 10mm to deploy, and then 5mm between probes.

I have proposed at various times to add a "clearance" setting instead, but inductive probes are more popular now and so "physical clearance" isn't so much the issue any more, as much as having enough "height" so the probe won't trigger.

The only difference in the code between a "raise" and a "height" value is adding one line to the `do_probe_raise` function, adding the probe's Z distance from the nozzle (negated, of course) before doing the raise. (For probes with a positive z offset, there no extra raise.) That seemed the most sensible thing from my perspective, but it was suggested we try HEIGHT instead, because (from a programmer's perspective it makes things a little easier).

If you've ever done database design or anything like that, then you know techies think in the reverse way that normal humans do. A human thinks about things being inside of containers, while a geek thinks in terms of child nodes having parents. I think it's important that we consider things from the end-user perspective.

So, if there's a setting relating to the height of a probe relative to the bed, I think we should respect the intuition of the user rather than the cold logic of the code. Therefore I do believe we should return to the concept of a "raise" and have it be relative to the probe, inasmuch as that is possible. No one is going to leave their raise set to zero, especially if they have a servo probe. They know it needs clearance to deploy. But end-users will continually make the same error — of not including the probe offset in their "height" value if we leave it this way. It's simply too easy to overlook, no matter how many warnings we put in ALL CAPS in the configuration files.

So… that's my whole thinking on the subject. I would much prefer to see end-users complain that it's raising too high than hear about broken servo arms and scraped beds for the next 6 months.

lrpirlet, 2016-09-16:
@thinkyhead 
I think we share the same view, even if yours is much wider... Thanks to make me aware...

Let me push forward the only one other idea I exposed that may not have been discussed enough... I have not seen it either accepted or pushed back...
 
What about being very specific in the variable name and start it with the word RELATIVE ? (or ABSOLUTE if the choice goes with absolute :-( ).

The English word RAISE does not quite mean RELATIVE to me (my mother language is French), Raise means to me "go up"... and that is twitching when I think of my CORE XY (in project) when the Z plate goes down for a RAISE :-)
OK, I bet that you'll tell me think of height of the hot tip relative to the bed... :-)
And, see, the word height can be either absolute OR relative..

To include RELATIVE (or ABSOLUTE) in a variable name would make it clearer to me (and maybe to others).

Of course this is MY view, and I will not even try to impose anything... The core team developing the Marlin certainly have a much better overview of the whole.

P.S. the only advantage of being dyslexic is the fact that each element of a chain can be presented in any order, since the mind can go any direction,  I can see both the car containing the driver and the driver dominating the car... and have endless discussion about who is the one protecting the other..

thinkyhead, 2016-09-18:
@lrpirlet Not to diminish serious dyslexia, but when it comes to remembering what's up and what's down, whether an offset will go in one direction or another, too often I find my head spinning. Hence all the recent issues with G29, I suppose…

Anyway, I hear what you're saying about the word "raise" in cases where the bed moves. Another term that I proposed, and has been hovering around, is CLEARANCE, which would preclude the term RELATIVE. So the settings could instead be `Z_CLEARANCE_DEPLOY_PROBE` and `Z_CLEARANCE_BETWEEN_PROBES`.

lrpirlet, 2016-09-19:
@thinkyhead 
CLEARANCE would be perfect for me, it give an idea of a gap between 2 objects (no sign needed since it describe a distance between 2 objects)

Roxy-3D, 2016-09-19:
I would be OK with 'CLEARANCE' type names.  And that does eliminate the confusion with beds that travel down for greater Z's.

thinkyhead, 2016-09-20:
I've added a commit to #4821 with the "clearance" verbiage. Have a look at the changes and see if it "feels" right.

lrpirlet, 2016-09-23:
@thinkyhead 
That looks fine... thanks for the resulting change .
Closing issue

closed 2016-09-23 by lrpirlet