MarlinFirmware/Marlin #4798: 1.1.0 RC7 - G02/G03 not as smooth as Marlin 1.0.2

jasoroony, 2016-09-13:
For a laser CNC: I'm seeing slowdowns on every arc end (or start?) just sending the tool around a large circle at speed.  1.0.2 does sometimes have a slowdown once or twice on the full circle, but it's much worse with the latest Marlin.  The very simple circle gcode is generated by InkScape, 70cm x 70cm and using 133 for the x/y max speed in Marlin - not particularly fast.  Acceleration/speed doesn't seem to change things.

Changing the arc fine tune settings in conf_adv from 1mm to 4mm makes it slightly better - comparable to the old Marlin version.  So I guess my question is, is there now more code in the buffer filling routines that can no longer keep up?  Is this something worth investigating and/or fixing?

thinkyhead, 2016-09-13:
There are a few changes that could be having this effect, including a change to the way acceleration is limited. Do you see any positive effect if you reduce the acceleration rate for XY?

jasoroony, 2016-09-13:
[Conf.zip](https://github.com/MarlinFirmware/Marlin/files/469500/Conf.zip)

I was already using RCBugFix, sorry for not mentioning it.

Yes, I do see smooth circular motion at a very slow x/y: 33 (but not @ 100, the two settings I just tried, I also reduced acceleration to 100 instead of 400 for both these new tests).

My config has 5 drivers for x,x2,y,y2,z - but I've tried with just the usual 3 without seeing an improvement with the normal 133 speed.  The machine is fairly configurable with sockets for both settings.  Also, the Z min and max end stops are switched around (so I can use GRBL to move Z up as home and Marlin down as home) but I don't expect you'll actually be using these config files.  Just a warning.

thinkyhead, 2016-09-14:
@jasoroony Do you see any positive or negative effect in changing the Jerk settings?

> …more code in the buffer filling routines…?

Are you using any form of bed leveling? For the most part the code is not much different from 1.0.2 in the planner and stepper, but we haven't really done any comprehensive profiling to see how they differ in timing, CPU load, etc. Unfortunately I don't have hardware to test with right now, or I would investigate this myself.

jasoroony, 2016-09-14:
No change, I've tried with jerk set to 4 and 20 but it was the same.  That was with the conservative v & a both set to 100.  Quite slow for what this machine can do.

I'm not using any bed leveling, other than my physical screws! :)

thinkyhead, 2016-09-14:
@jasoroony Hmm. We could try using the old (supposedly incorrect) acceleration code and see if it makes any difference. To do that, replace this block in `planner.cpp` around line 1010…
```cpp
block->acceleration_steps_per_s2 = ceil((block->steps[E_AXIS] ? acceleration : travel_acceleration) * steps_per_mm);
if (max_acceleration_steps_per_s2[X_AXIS] < (block->acceleration_steps_per_s2 * block->steps[X_AXIS]) / block->step_event_count)
  block->acceleration_steps_per_s2 = (max_acceleration_steps_per_s2[X_AXIS] * block->step_event_count) / block->steps[X_AXIS];
if (max_acceleration_steps_per_s2[Y_AXIS] < (block->acceleration_steps_per_s2 * block->steps[Y_AXIS]) / block->step_event_count)
  block->acceleration_steps_per_s2 = (max_acceleration_steps_per_s2[Y_AXIS] * block->step_event_count) / block->steps[Y_AXIS];
if (max_acceleration_steps_per_s2[Z_AXIS] < (block->acceleration_steps_per_s2 * block->steps[Z_AXIS]) / block->step_event_count)
  block->acceleration_steps_per_s2 = (max_acceleration_steps_per_s2[Z_AXIS] * block->step_event_count) / block->steps[Z_AXIS];
if (max_acceleration_steps_per_s2[E_AXIS] < (block->acceleration_steps_per_s2 * block->steps[E_AXIS]) / block->step_event_count)
  block->acceleration_steps_per_s2 = (max_acceleration_steps_per_s2[E_AXIS] * block->step_event_count) / block->steps[E_AXIS];
```
…with this block of code…
```cpp
block->acceleration_steps_per_s2 = ceil(acceleration * steps_per_mm); // convert to: acceleration steps/sec^2
if(((float)block->acceleration_steps_per_s2 * (float)block->steps[X_AXIS] / (float)block->step_event_count) > max_acceleration_steps_per_s2[X_AXIS])
  block->acceleration_steps_per_s2 = max_acceleration_steps_per_s2[X_AXIS];
if(((float)block->acceleration_steps_per_s2 * (float)block->steps[Y_AXIS] / (float)block->step_event_count) > max_acceleration_steps_per_s2[Y_AXIS])
  block->acceleration_steps_per_s2 = max_acceleration_steps_per_s2[Y_AXIS];
if(((float)block->acceleration_steps_per_s2 * (float)block->steps[E_AXIS] / (float)block->step_event_count) > max_acceleration_steps_per_s2[E_AXIS])
  block->acceleration_steps_per_s2 = max_acceleration_steps_per_s2[E_AXIS];
if(((float)block->acceleration_steps_per_s2 * (float)block->steps[Z_AXIS] / (float)block->step_event_count) > max_acceleration_steps_per_s2[Z_AXIS])
  block->acceleration_steps_per_s2 = max_acceleration_steps_per_s2[Z_AXIS];
```
And, just to see if it makes any difference, you should also try this block of code, which is the same as the current code, but does some extra casting that might, possibly, have some effect....
```cpp
block->acceleration_steps_per_s2 = ceil((block->steps[E_AXIS] ? acceleration : travel_acceleration) * steps_per_mm);
if (max_acceleration_steps_per_s2[X_AXIS] < (block->acceleration_steps_per_s2 * (float)block->steps[X_AXIS]) / (float)block->step_event_count)
  block->acceleration_steps_per_s2 = (max_acceleration_steps_per_s2[X_AXIS] * (float)block->step_event_count) / (float)block->steps[X_AXIS];
if (max_acceleration_steps_per_s2[Y_AXIS] < (block->acceleration_steps_per_s2 * (float)block->steps[Y_AXIS]) / (float)block->step_event_count)
  block->acceleration_steps_per_s2 = (max_acceleration_steps_per_s2[Y_AXIS] * (float)block->step_event_count) / (float)block->steps[Y_AXIS];
if (max_acceleration_steps_per_s2[Z_AXIS] < (block->acceleration_steps_per_s2 * (float)block->steps[Z_AXIS]) / (float)block->step_event_count)
  block->acceleration_steps_per_s2 = (max_acceleration_steps_per_s2[Z_AXIS] * (float)block->step_event_count) / (float)block->steps[Z_AXIS];
if (max_acceleration_steps_per_s2[E_AXIS] < (block->acceleration_steps_per_s2 * (float)block->steps[E_AXIS]) / (float)block->step_event_count)
  block->acceleration_steps_per_s2 = (max_acceleration_steps_per_s2[E_AXIS] * (float)block->step_event_count) / (float)block->steps[E_AXIS];
```

jasoroony, 2016-09-15:
Um, sheepishly I say, no change with any of those code segments.  They all do the same thing, but I think I know what's causing it now.  It's not at the end or start of an arc - that was my assumption (sorry!) it's happening every time the keep-alive pixel changes on the RRD Full Graphic Display.  So the LCD redraw is most probably somehow delaying the needed calculations for arcs.

thinkyhead, 2016-09-15:
Actually I'm kind of relieved to know the acceleration code isn't implicated, especially since the newer code is using more integer maths and should be a little quicker. Nevertheless I'm looking at some other tweaks to help optimize in situations where lines are split up into smaller moves.

I've been thinking for a while that it would be very helpful with some setups to "sleep" the display, or give the option to sleep the display anyway. This would really help Delta performance too. The display would go into some mode where it doesn't draw at all, or draws much less, but you could hit the button to "wake it up" again.

Roxy-3D, 2016-09-16:
> I've been thinking for a while that it would be very helpful with some setups to "sleep" the display, or give the option to sleep the display anyway. This would really help Delta performance too. [...]

Yes!   In fact...   I'll proto-type and test this in devel-ubl.   I'm spending a lot of time to free up CPU cycles.  This will help me do that.    And here is something we can debate:    In UBL and the Mesh Validation, there is the concept of "You can press and hold the encoder wheel to abort the operation."   Let's extend that idea to "You can press and hold the encoder wheel to get attention of the system."

thinkyhead, 2016-09-16:
I'm also thinking, press-and-hold the encoder button to do… _something_… that you can choose from a list of options when you build Marlin.

I'd like to have a "debug" display — sort of an extended Info Screen that would give some helpful diagnostic info about memory, performance, etc. Of course it wouldn't update nearly as often, so as to have less effect on performance.

thinkyhead, 2016-09-16:
> free up CPU cycles for Delta's

It occurred to me that we could do some additional clever trickery to accomplish that. As we've been discussing, we're doing 3 `sqrt` functions for every single kinematic segment when calculating delta motion. According to @esenapaj we're not going to beat the performance of the C math library with any clever hacks at that level.

But our discussion about the idea of "interpolation" sparked an idea in my weary brain today. Namely, I think we can probably get away with doing half as many —or maybe even 1/3 as many— `sqrt` calculations and just interpolate the tower positions in-between those points. The interpolation wouldn't be precisely in-between the two (or 3) tower positions, but would be skewed slightly based on the proportions of the hypotenuses (i.e., the sum of the squares before we take the square root). Or, there may be a simple "momentum-based" way to get the skew without having to do three squarings.

The initial interpolation part should actually be very easy!

adamfilip, 2016-09-16:
Id be happy to sacrifice LCD update speed to keep things running smoothly. I dont need to know the XYZ position, a once per second update would be sufficient for my needs, personally.. perhaps.. remove animations or slow them down.

jasoroony, 2016-09-16:
Unfortunately, the U8GLib sort of requires a full LCD update from an initially clear buffer, so reducing the amount of things changing on the display doesn't save much time since anything you are showing needs to be drawn again and again - and the LCD gets a full set of update signals each refresh (every pixel gets updated).  There are many coded delays in sending that data through.

I've been playing with the newer U8G2 library instead, where you can allocate an initial full buffer for every pixel and only do updates to the parts of it you want, it still sends a full set of signals to the LCD though and that would cost another 512 bytes (Marlin already allocates 512 bytes for a half buffer).  It is also much slower than U8GLib because it doesn't do direct hardware manipulation, going through the Arduino libraries instead now.

My latest adventure with Arduino is using a bit of the Marlin LCD direct hardware code to do the full update, since it's even faster than the original U8GLib but that doesn't help here of course, and all these ideas require a bit of a re-write of the Marlin code.  I say a bit, because it's really only the status screen that needs attention (the menus can remain the same, and U8G2 has backwards compatibility) and my final solution (for my particular project) was to include a "tile" (see U8G2) list that says which tiles are dirty and need updating...  My project will draw tiles as time permits rather than the whole screen at once.

Alas the RRD FGD has issues with "tiles" because it can only "address" the "even" one of two of what U8G2 calls a tile.  So that made things tricky (a delay on acting on the last item in my tile list, so I can tell if the next addition has hit the tile next to it and I can discard that new entry because it's got to happen anyway).  I also then had to find even more CPU for my project!  So that code was modified to draw those adjacent tiles in quarters (16 pixels across, two rows down) a bit at a time, yielding to the main project before coming back for the next bit (lucky I had bits to play with in my tile list bytes!)

thinkyhead, 2016-09-17:
It sounds like, if we know which portion of the screen is being drawn, we can totally skip anything that is wholly outside of that region (e.g., the fan and heaters at the top, the SD progress bar and status line at the bottom, and of course the upper/lower lines in the submenus). I can imagine that this one small trick alone will make a difference. We'd just need to set a pair of globals that specify the top and bottom "scanlines" of the region currently being refreshed.

jasoroony, 2016-09-17:
Well, I thought that was a good idea, so I quickly coded up the status screen check in the draw routine and only allowed y lines to be drawn when the little bit was equal to the keep-alive dot value (half the screen drawn each refresh) but it still stalls going around the circle.

In fact, I then made it every two lines (making my own dot variable that goes from 0 - 3) and it still stalled.

Than again, 0 - 7 for my new dot (note, each time the marlin flash screen [which happens to show when the status screen is set as current] gets more and more "line" thread bare!) and it still stalls.

I'm going to say, the U8GLib draw routines will always take too much CPU unless you have a full buffer and make changes to how that buffer is updated.

jasoroony, 2016-09-17:
Is there some quick way (in code) of preventing an LCD refresh completely when the current or previous command was an arc?  Or are the "delta's" a more important thing to fix here?

jasoroony, 2016-09-18:
I've been going back through the RC versions of Marlin (though I should have only tried RC6!) looking at what they do differently...

Removing the following call to idle() makes the code go back to the way it was:

In Marlin_main.cpp:

void plan_arc () {
...
millis_t now = millis();
if (ELAPSED(now, next_idle_ms)) {
  next_idle_ms = now + 200UL;
  idle();
}
...

However, there is still one or sometimes two stalls going around the circle.  This is the same as I have seen in every other version except I did see RC1 go all the way around without a stall twice before I moved on - that may have been luck though.

jasoroony, 2016-09-18:
I made some changes (the changes below don't include Marlin.h, planner.cpp, planner_bezier.cpp or stepper.cpp changes required which is passing the draw parameter value of 1 in most cases except for "bezier" which passes 0) the plan_arc function passes 0 to initiate a temporary bad on drawing the LCD.  This makes sure everything (housekeeping) is run except the lcd_update().  It also delays the next time it can run in case arcs are bunched up in the file.  Seems to work for me, except the menus become unresponsive, which is OK because I don't want to accidentally ruin my job by forgetting not to use the menus!  Perhaps some #ifdef statements could be inserted so only the graphic displays use this code but for my usage this works.

void idle(uint8_t draw
  #if ENABLED(FILAMENT_CHANGE_FEATURE)
    , bool no_stepper_sleep/*=false*/
  #endif
) {
  static uint8_t hold_off_draw = 0;
  
  if (hold_off_draw)
    hold_off_draw--;

  if (draw == 0)
    hold_off_draw = 10;
  // (This should be for 1/2 second(s) after an arc finishes, in case there's another arc coming up.)

  if ((hold_off_draw == 0) && (draw)) // Prevent drawing the LCD when doing arcs
    lcd_update();
  
  host_keepalive();
  manage_inactivity(
    #if ENABLED(FILAMENT_CHANGE_FEATURE)
...

thinkyhead, 2016-09-18:
@jasoroony Yeah, Deltas and SCARAs are going to have the biggest bottleneck, since every move involves kinematic conversion, lots of small moves, and thus lots of calculation plus lots of planner blocks.

Delta, SCARA, and arc moves are "blocking" because they quickly fill up the planner buffer, and so `Planner::buffer_line` has to wait for the queue to gain space before blocks can be added. While `buffer_line` is waiting for the stepper ISR to finish the oldest block it calls `idle()`, and that invokes a screen draw every 100ms or so.

So essentially, you have to either do fewer screen draws or you have to run the ISR at a slower rate and thus make longer delta segments. The thing is, all the heavy calculation is done in the main thread, not in the ISR. So one key to optimization is going to be in pre-calculating as much as possible so the ISR uses as few cycles as possible. Another optimization yet-to-be is to convert as much as possible to fixed-point integer maths.

thinkyhead, 2016-09-18:
> I made some changes

That's generally the concept I described, with the same problem. Namely that it blocks the lcd update and makes menus unresponsive. A more comprehensive solution is still needed. Something that would only reduce or delay the update of the screen, but would not do so when controller button events are received.

jasoroony, 2016-09-19:
:)  Thanks for fixing up my terrible post above with the <code> segments.

It's going to be a hard issue to fix, I think.  At least without hacking into U8GLib some more.

Perhaps, another good idea is (given, I see with all the different features in Marlin, different amounts of RAM are needed) move to U8G2 and allocate a smaller buffer of 128 bytes (basically a line of text) instead of your 512 bytes or the actual required 1KB for the full screen buffer mode - without telling U8G2 that it's smaller / using the full mode initialiser (and fix the clear buffer code to clear less bytes) then only ever clear + "draw" to that first line.

That gets you back a heap of RAM so a win there.

Then modify your hardware layer LCD output routine to pick which Y row to start outputting the buffered line to.  That way you can really reduce the amount of CPU used (remember U8GLib will process the full set of menu lines multiple times excluding those outside your buffer lines by what I assume is a painstakingly CPU intensive process) by only processing one line each call to idle().  That was the nice thing about U8G2, that it can do away with the firstPage/nextPage in full mode.  Then do it 10 times a second - so over 1 second the whole display gets refreshed fully plus a bit, but you get back to the calculations needed much more quickly.

Maybe you can even do this without changing ultralcd.cpp, by using custom firstPage/nextPage methods that know what you're trying to achieve.  Drawing character row 1, ignore everything except the first MENU_ITEM_XXX that comes in... etc. etc.  Spash and Status screens might need a bit more work, but I think they were always going to.

jasoroony, 2016-09-20:
Hmmm, there's far more edits required for Marlin to use U8G2 than I thought.  Maybe this will work instead?  (I'm not near my machine to test it at the moment.)

ultra_st7920_u8glib_rrd.h: (change to use the smallest buffer)

<code>
#define PAGE_HEIGHT 8   //128 byte framebuffer
//#define PAGE_HEIGHT 16  //256 byte framebuffer
//#define PAGE_HEIGHT 32  //512 byte framebuffer
</code>

ultra.cpp: (break up the page-processing - in lcd_update())

<code>
    if (lcdDrawUpdate) {
      uint8_t lcdOldDrawUpdate = lcdDrawUpdate;

      switch (lcdDrawUpdate) {
        case LCDVIEW_CALL_NO_REDRAW:
          lcdDrawUpdate = LCDVIEW_NONE;
          break;
        case LCDVIEW_CLEAR_CALL_REDRAW: // set by handlers, then altered after (rarely occurs here)
        case LCDVIEW_CALL_REDRAW_NEXT:  // set by handlers, then altered after (never occurs here?)
          lcdDrawUpdate = LCDVIEW_REDRAW_NOW;
        case LCDVIEW_REDRAW_NOW:        // set above, or by a handler through LCDVIEW_CALL_REDRAW_NEXT
        case LCDVIEW_NONE:
          break;
      }

      #if ENABLED(ULTIPANEL)
        #define CURRENTSCREEN() (*currentScreen)()
      #else
        #define CURRENTSCREEN() lcd_status_screen()
      #endif

      #if ENABLED(DOGLCD)  // Changes due to different driver architecture of the DOGM display
        static int8_t next_page_result = 0;
        static int8_t dot_color = 0;
        dot_color = 1 - dot_color;
        if (next_page_result == 0)
          u8g.firstPage();
        //do {
          lcd_setFont(FONT_MENU);
          u8g.setPrintPos(125, 0);
          u8g.setColorIndex(dot_color); // Set color for the alive dot
          u8g.drawPixel(127, 63); // draw alive dot
          u8g.setColorIndex(1); // black on white
          CURRENTSCREEN();
        //} while (u8g.nextPage());
          if ((next_page_result = u8g.nextPage()) != 0) {
            // Want to continue refreshing the screen
            lcd_status_update_delay = 1;
            next_lcd_update_ms = ms;
            lcdDrawUpdate = lcdOldDrawUpdate;
            return;
          }
      #else
        CURRENTSCREEN();
      #endif
    }
</code>

jasoroony, 2016-09-20:
After some testing, I notice it works a lot better when this code is added at the top (added, except that the variables and #defines are moved.)  But the SD file list screen is really slow now - any idea guys?  This does make the "circle" gcode much smoother but not as smooth as it can be.  The keep-alive dot runs really fast too???  Anyway, with a few fixes this might give back a heap of RAM and CPU.

```cpp
void lcd_update() {

  #if ENABLED(ULTIPANEL)
    #define CURRENTSCREEN() (*currentScreen)()
  #else
    #define CURRENTSCREEN() lcd_status_screen()
  #endif
  #if ENABLED(DOGLCD)  // Changes due to different driver architecture of the DOGM display
    static int8_t next_page_result = 0;
    static int8_t dot_color = 0;
    if (next_page_result != 0) {
      lcd_setFont(FONT_MENU);
      u8g.setPrintPos(125, 0);
      u8g.setColorIndex(dot_color); // Set color for the alive dot
      u8g.drawPixel(127, 63); // draw alive dot
      u8g.setColorIndex(1); // black on white
      CURRENTSCREEN();
      if ((next_page_result = u8g.nextPage()) != 0) // Want to continue refreshing the screen
        return;
    }
    static millis_t return_to_status_ms = 0;
    manage_manual_move();
  #endif

  lcd_buttons_update();
...
```

jasoroony, 2016-09-20:
(Note to the Marlin developers - I think the code I've changed now in the two previous posts is about as far as I can take it - as I just don't know the Marlin graphics code well enough to locate the remaining draw issue that I can see in the "normal" Marlin code too, for a moment before it redraws anyway.)

Roxy-3D, 2016-09-20:
It sure would be nice if the Graphic's Panels use less CPU cycles because of this!

jasoroony, 2016-09-21:
I've reused some of the saved bytes to cache the sd card files shown on the LCD which makes it better than before (I think).  Still can't work out why that first line doesn't draw the first time around!

```cpp
  #if ENABLED(SDSUPPORT)

    static uint16_t fileCntCache = 0xFFFF;

    #if !PIN_EXISTS(SD_DETECT)
      static void lcd_sd_refresh() {
        card.initsd();
        encoderTopLine = 0;
        fileCntCache = 0xFFFF;
      }
    #endif

    static void lcd_sd_updir() {
      card.updir();
      encoderTopLine = 0;
      fileCntCache = 0xFFFF;
    }

    /**
     *
     * "Print from SD" submenu
     *
     */
    void lcd_sdcard_menu2() {
      ENCODER_DIRECTION_MENUS();
      if (lcdDrawUpdate == 0 && LCD_CLICKED == 0) return; // nothing to do (so don't thrash the SD card)

      #define MAXFILECACHE LCD_HEIGHT // 5 (Max is 8)
      static uint16_t fileStartCache = 0xFFF0; // Make sure this value + MAXFILECACHE doesn't overflow
      static uint8_t  fileItemsCache = 0;
      static uint8_t  fileTypeCache = 0; // Bit pattern of at most 8 cached items (1=Dir)
      static char     filenameCache [MAXFILECACHE][FILENAME_LENGTH];
      static char longFilenameCache [MAXFILECACHE][LONG_FILENAME_LENGTH];

      uint16_t fileCnt;
      uint16_t j;

      if (fileCntCache == 0xFFFF) {
        fileStartCache = 0xFFF0;
        fileItemsCache = 0;
        fileTypeCache = 0;
        fileCntCache = fileCnt = card.getnrfilenames(); // We want to avoid doing a full directory scan.
      } else
        fileCnt = fileCntCache;

      START_MENU();
      MENU_ITEM(back, MSG_MAIN);
      card.getWorkDirName();
      if (card.filename[0] == '/') {
        #if !PIN_EXISTS(SD_DETECT)
          MENU_ITEM(function, LCD_STR_REFRESH MSG_REFRESH, lcd_sd_refresh);
        #endif
      }
      else {
        MENU_ITEM(function, LCD_STR_FOLDER "..", lcd_sd_updir);
      }

      for (uint16_t i = 0; i < fileCnt; i++) {
        if (_menuLineNr == _thisItemNr) {

          if ((i >= fileStartCache) && (i < fileStartCache + fileItemsCache)) { // Cache hit?
            j = i - fileStartCache;
            if ((fileTypeCache & (1 << j)) != 0)
              MENU_ITEM(sddirectory, MSG_CARD_MENU, filenameCache [j], longFilenameCache [j]);
            else
              MENU_ITEM(sdfile, MSG_CARD_MENU, filenameCache [j], longFilenameCache [j]);

          } else {

            card.getfilename(
               #if ENABLED(SDCARD_RATHERRECENTFIRST)
                 fileCnt-1 -
               #endif
               i
            );

            if ((i < fileStartCache) || (i >= fileStartCache + MAXFILECACHE)) {
              fileStartCache = i;
              fileItemsCache = 0;
              fileTypeCache = 0;
            } 
            strcpy (filenameCache [fileItemsCache], card.filename);
            strcpy (longFilenameCache [fileItemsCache], card.longFilename);
            if (card.filenameIsDir)
              fileTypeCache |= (1 << fileItemsCache);
            fileItemsCache++; // Menu items 'return' from functions when selected

            if (card.filenameIsDir)
              MENU_ITEM(sddirectory, MSG_CARD_MENU, card.filename, card.longFilename);
            else
              MENU_ITEM(sdfile, MSG_CARD_MENU, card.filename, card.longFilename);
          }
        }
        else
          MENU_ITEM_DUMMY();
      }
      END_MENU();
    }

    void lcd_sdcard_menu () {
      fileCntCache = 0xFFFF;
      currentScreen = lcd_sdcard_menu2;
      lcd_sdcard_menu2();
    }

  #endif //SDSUPPORT

...and down further for sddirectory...

  #if ENABLED(SDSUPPORT)

    static void menu_action_sdfile(const char* filename, char* longFilename) {
      UNUSED(longFilename);
      card.openAndPrintFile(filename);
      lcd_return_to_status();
    }

    static void menu_action_sddirectory(const char* filename, char* longFilename) {
      UNUSED(longFilename);
      card.chdir(filename);
      encoderPosition = 0;
      fileCntCache = 0xFFFF;
    }

  #endif //SDSUPPORT
```

thinkyhead, 2016-09-21:
I'd mess with this myself, but unfortunately I'm visiting another city and won't have a printer or display to test with, possibly for a couple of weeks. Maybe I can borrow a display and do some testing this coming weekend.

jasoroony, 2016-09-22:
I put lots of serial logging in and I could see what was going wrong, my copy of lcdDrawUpdate (lcdOldDrawUpdate) in the top post, and restoring it before returning, was causing the issue.  I've edited that out of the post now.  It was really only part of the 1st failed attempt at changing the logic anyway.

It's interesting though, the LCD redraws twice to fix the "top line" missing bit bug, which is part of the original Marlin.  It does hand back control between refreshes though, so it wasn't causing a doubly long delay in the planner, but was probably causing arcs to double-stall with the original full refresh.

That's it from me, unless I decide to try and only redraw the first line (in the 2nd pass) or make it draw properly the first time - I think this code now works well enough to get back a lot of CPU and a bit of RAM too.  It's debatable if #ifdef's need to be put in the new SD card cache code, since it should make scrolling on 20x4 LCDs slightly better too.  Not my call though, it really depends on the RAM needed for other options, but I assume they all work with the graphic displays already so it could stay as-is.

This is still not enough to make arcs at my full speed as smooth as with the LCD blocked totally.  I've tried seeing if a 1 pixel high line buffer works better, but I don't really seem much improvement, if any.  I expect there's a whole lot of processing going on regardless of the buffer size (as I explained earlier) which is why I thought U8G2 would work better (reducing the processing that goes on each call).  Oh well.  The Marlin menu system is fairly integrated in working that way too and not exactly what I had in mind.

jasoroony, 2016-09-22:
Possible double draw fix: (adding in check for - lcdDrawUpdate == LCDVIEW_REDRAW_NOW)

```cpp
  #define _MENU_ITEM_PART_1(TYPE, LABEL, ARGS...) \
    if (_menuLineNr == _thisItemNr) { \
      if (lcdDrawUpdate == LCDVIEW_REDRAW_NOW) \
        lcd_implementation_drawmenu_ ## TYPE(encoderLine == ```

and

```cpp
      #if ENABLED(DOGLCD)  // Changes due to different driver architecture of the DOGM display
        if (lcdDrawUpdate == LCDVIEW_REDRAW_NOW) {
          dot_color = 1 - dot_color;
          u8g.firstPage();
        //do {
          lcd_setFont(FONT_MENU);
          u8g.setPrintPos(125, 0);
          u8g.setColorIndex(dot_color); // Set color for the alive dot
          u8g.drawPixel(127, 63); // draw alive dot
          u8g.setColorIndex(1); // black on white
        }
          CURRENTSCREEN();
        //} while (u8g.nextPage());
        if ((lcdDrawUpdate == LCDVIEW_REDRAW_NOW) && ((next_page_result = u8g.nextPage()) != 0))
          return; // Want to continue refreshing the screen

      #else
```

jasoroony, 2016-09-22:
Unsurprisingly, crafting the status screen to draw only when the smaller buffer lines are "in play" speeds things up considerably (to the point where my machine's full speed no longer stalls, but merely slows slightly when the refresh occurs):

in ultralcd.cpp (add a variable drawingSegment to "count" which area of the screen is being drawn)

```cpp
uint8_t drawingSegment = 0;

void lcd_update() {

  #if ENABLED(ULTIPANEL)
    #define CURRENTSCREEN() (*currentScreen)()
  #else
    #define CURRENTSCREEN() lcd_status_screen()
  #endif
  
  #if ENABLED(DOGLCD)  // Changes due to different driver architecture of the DOGM display
    static int8_t next_page_result = 0;
    static int8_t dot_color = 0;
    if (next_page_result != 0) {
      drawingSegment++;
      lcd_setFont(FONT_MENU);
```

...further down...

```cpp
      #if ENABLED(DOGLCD)  // Changes due to different driver architecture of the DOGM display
        if (lcdDrawUpdate == LCDVIEW_REDRAW_NOW) {
          dot_color = 1 - dot_color;
          u8g.firstPage();
          drawingSegment = 0;
        //do {
          lcd_setFont(FONT_MENU);
          u8g.setPrintPos(125, 0);
          u8g.setColorIndex(dot_color); // Set color for the alive dot
          u8g.drawPixel(127, 63); // draw alive dot
          u8g.setColorIndex(1); // black on white
        }
          CURRENTSCREEN();
        //} while (u8g.nextPage());
        if ((lcdDrawUpdate == LCDVIEW_REDRAW_NOW) && ((next_page_result = u8g.nextPage()) != 0))
          return; // Want to continue refreshing the screen

      #else
        CURRENTSCREEN();
      #endif
    }
```

in file ultralcd_impl_dogm.h (use the new variable to reduce used U8GLib functions to the minimum)

```cpp
extern uint8_t drawingSegment;

static void lcd_implementation_status_screen() {
  u8g.setColorIndex(1); // black on white

  bool blink = lcd_blink();

  // Symbols menu graphics, animated fan
  if (drawingSegment <= 2)
  {
    u8g.drawBitmapP(9, 1, STATUS_SCREENBYTEWIDTH, STATUS_SCREENHEIGHT,
      #if HAS_FAN0
        blink && fanSpeeds[0] ? status_screen0_bmp : status_screen1_bmp
      #else
        status_screen0_bmp
      #endif
    );
  }
[... 120 lines cut ...]

  if (drawingSegment < 7) return;

  // Status line
  #if ENABLED(USE_SMALL_INFOFONT)
    u8g.setPrintPos(0, 62);
  #else
    u8g.setPrintPos(0, 63);
  #endif
  #if DISABLED(FILAMENT_LCD_DISPLAY)
    lcd_print(lcd_status_message);
  #else
    if (PENDING(millis(), previous_lcd_status_ms + 5000UL)) {  //Display both Status message line and Filament display on the last line
      lcd_print(lcd_status_message);
    }
    else {
      lcd_printPGM(PSTR("dia:"));
      lcd_print(ftostr12ns(filament_width_meas));
      lcd_printPGM(PSTR(" factor:"));
      lcd_print(itostr3(100.0 * volumetric_multiplier[FILAMENT_SENSOR_EXTRUDER_NUM]));
      u8g.print('%');
    }
  #endif
}
```

Blue-Marlin, 2016-09-22:
Nice idea, but will not work on a Viki2, what is drawn bottom to top (screenrot_180).
For a proper, but outdated, implementation see: https://github.com/AnHardt/Marlin/pull/13

jasoroony, 2016-09-23:
Good point.  But it can still be pretty easily made to work: (though not for 90 and 270 degrees rotation...)

```cpp
uint8_t drawingSegment = 0xFF;

void lcd_update() {

  #if ENABLED(ULTIPANEL)
    #define CURRENTSCREEN() (*currentScreen)()
  #else
    #define CURRENTSCREEN() lcd_status_screen()
  #endif
  
  #if ENABLED(DOGLCD)  // Changes due to different driver architecture of the DOGM display
    static int8_t next_page_result = 0;
    static int8_t dot_color = 0;
    if (next_page_result != 0) {
      
      #if ENABLED(LCD_SCREEN_ROT_90)
      #elif ENABLED(LCD_SCREEN_ROT_180)
        drawingSegment >>= 1;
      #elif ENABLED(LCD_SCREEN_ROT_270)
      #else
        drawingSegment <<= 1;
      #endif

      lcd_setFont(FONT_MENU);
```

... further down ...

```cpp
      #if ENABLED(DOGLCD)  // Changes due to different driver architecture of the DOGM display
        if (lcdDrawUpdate == LCDVIEW_REDRAW_NOW) {
          dot_color = 1 - dot_color;
          u8g.firstPage();

          #if ENABLED(LCD_SCREEN_ROT_90)
          #elif ENABLED(LCD_SCREEN_ROT_180)
            drawingSegment = 0x80;
          #elif ENABLED(LCD_SCREEN_ROT_270)
          #else
            drawingSegment = 0x01;
          #endif

        //do {
          lcd_setFont(FONT_MENU);
```

In file ultralcd_impl_dogm.h:

```cpp
extern uint8_t drawingSegment;

static void lcd_implementation_status_screen() {
  u8g.setColorIndex(1); // black on white

  bool blink = lcd_blink();

  // Symbols menu graphics, animated fan
  if ((drawingSegment & 0x07) != 0)
  {
    u8g.drawBitmapP(9, 1, STATUS_SCREENBYTEWIDTH, STATUS_SCREENHEIGHT,
      #if HAS_FAN0
        blink && fanSpeeds[0] ? status_screen0_bmp : status_screen1_bmp
      #else
        status_screen0_bmp
      #endif
    );
  }
[... 201 lines cut ...]
```cpp
  void lcd_implementation_drawedit(const char* pstr, const char* value=NULL) {

    if ((drawingSegment & 0x36) == 0)
        return; // Only bother drawnig if drawing in the centre.

    uint8_t rows = 1;
```

```cpp
    static void _drawmenu_sd(bool isSelected, uint8_t row, const char* pstr, const char* filename, char* const longFilename, bool isDir) {
      UNUSED(pstr);
      char c;
      uint8_t n = LCD_WIDTH - (START_COL) - 1;

      if (longFilename[0]) {
        filename = longFilename;
        longFilename[n] = '\0';
      }

      if (lcd_implementation_mark_as_selected(row, isSelected))
        return;

      if (isDir) lcd_print(LCD_STR_FOLDER[0]);
```

jasoroony, 2016-09-23:
Quote: "For a proper, but outdated, implementation see: AnHardt#13"

Thanks for that, I've had a look at this now.  Yes, that branch does very similar things, but as you say it is very outdated now!  It also still uses the two-page update (where I'm using 8) so I'm thinking even though I'm not in a position to measure their timings, the solution here should be giving:

1/ More CPU back on the status screen without changing the look.
2/ More RAM back.
3/ Slightly better CPU usage with normal menus.
4/ A much better SD file selection experience.
5/ Doesn't break 90/270 degree rotation, even though there's not "much gain" when using them.
6/ I'm not totally sure, but it probably doesn't fix the double-draw issue I saw.

Though I would suggest the change to the status screen in the other branch that follows this comment is needed in my solution too, since I've sometimes seen the "tear" in the X/Y/Z position readout that it will solve:

  // precalculate all strings in stripe 0

Not sure if I'll get a chance to look at that for a while though.  There's probably other things in there too, like using better selected library functions to improve speed that could be ported across later.  All in all, it was pretty good work and it's a shame, somewhat, that it's now outdated and abandoned by the fellow who put the effort in.

thinkyhead, 2016-09-24:
I actually noticed something, looking back to your comment about `idle()` being called in `plan_arc`. It's not needed. The command `planner.buffer_line` calls `idle()` anytime the planner buffer is full during the arc loop. And that is certain to be the case a lot. So I don't think the extra logic in `plan_arc` to call `idle()` periodically makes any difference. In the same vein, the logic to make sure it doesn't call it too often ends up being fairly ineffectual. And the call to `manage_heater` there is also therefore likely pretty redundant, and only slows down the loop.

jasoroony, 2016-09-24:
My machine takes 23 seconds to complete the full lap at full speed.  With that code block commented out, the screen refreshes once but sometimes twice.  I ran it a few times.  This is consistent with Marlin 1.0.2 and you really don't have any control via the LCD while it's processing those arcs like this.

I've also now separated the drawing from the nextPage call (draw, yield, nextPage, yield, draw, etc.) and it's getting better still for me.  I've also optimised the menu line drawing so it does 2 or 3 pages of drawing depending on if the text crosses over a page or two.  But I can see the machine slowing down as status refreshes occur, or if in the menus, each time I cause a refresh by turning the selector knob.

I'm interested in why that is, because I've tried with the slow-down feature turned off and it still does the same thing - but I really expected it to stall rather than just slow down.

thinkyhead, 2016-09-24:
Normally the screen refreshes only periodically, such as for the Info display, but when the controller knob is turned it refreshes for every perceptible change of the knob position.

jasoroony, 2016-09-25:
You lose control just with the block commented out, mind you though, the LCD updates are much slower once you start performing the arc.  Considering how much I'm yielding, this is to be expected and I don't consider it a problem.  You have to ask also, how many jobs are fully comprised of arcs?  Yes, I think it needs to work properly in terms of the job, but I can certainly live with trading off performance in the LCD for those rare jobs!

The nice thing is, the menu selection itself still works well because the menus in Marlin are still checking that each time the LCD update routine is called.  It doesn't need to fully redraw (or even half redraw now) before acting on your selection.

BTW: I have not "branched" the project, it's in my own local source management system, but if the code that's here already gets past testing and into RCBugFix, then I'll provide the "extras" I've found.  What's here now stops the machine from tearing itself apart, the "extras" so far have only reduced the slowdown a bit more.  And I think I've described them well enough for anyone with coding skills to implement them anyway.

I suspect though testing is going to be troublesome, it seems like all the Marlin devs have opted for a 20x4 LCD, and I too have a spare standalone one I'm considering wiring up, if I had known about these issues I wouldn't have opted for the full graphic display!  Or maybe I'll drop this approach for my already working 25x8 text (graphic library based on U8G2) that I've already made use next to nothing in terms of CPU...  I'm sure I, at least, have got the 512 bytes spare.  Too many decisions, not enough time!

Take care,
Jason.

thinkyhead, 2016-09-25:
> I have not "branched" the project

Too bad. Things get lost and stale and forgotten without an active fork.

> anyone with coding skills to implement them

Someday. When time allows.

closed 2016-09-26 by jasoroony

jasoroony, 2016-09-26:
Sorry, I've been asked to clean up the code I posted here due to copyright issues.

thinkyhead, 2016-09-26:
Copyright or NDA? 😉

Roxy-3D, 2016-09-27:
> it seems like all the Marlin devs have opted for a 20x4 LCD,

With infinite RAM and CPU cycles, there are some small benefits that a Graphical Display could bring.   And certainly opinions are going to vary.   But a 20x4 LCD does everything I need.  it costs less than $10 on eBay (including shipping),   maybe that is good enough for now?