MarlinFirmware/Marlin #4800: Feature request: Option to display feedrate instead of XY values during printing

Bendimon, 2016-09-13:
When I'm printing, I don't need to see XY values on my LCD, they're changing too fast, and I only find useful to see Z values.

On the other hand, It would be useful to me to see what's the programmed feedrate (F) at every moment, to stablish the relationship between what i'm seeing and the number, to know what I should change on the slicer.

Perhaps it could be an option in the firmware to determine if the user wants to see XY or F.. only during execution, because in manual mode it's good to see XY values.

Kaibob2, 2016-09-13:
Good idea. I would also prefer to have a feedback of the current feedrates, but I think the update frequency of the display is too low. It would be as laggy as the XY positions.

thinkyhead, 2016-09-13:
We have several outstanding requests to give alternative Info Screen display options. I like the layout of the Info Screen in the Prusa i3 MK2 branch. So perhaps we will adopt that.

boelle, 2016-11-02:
since this is a dublicate why not close?