MarlinFirmware/Marlin #4802: AUTO BED LEVELING FEATURE not working with latest RCBugFix

renzaaa, 2016-09-13:
Not working with AUTO_BED_LEVELING_GRID_POINTS 2 (probing 2 points and gantry crash to the x-max side).
Not working with AUTO_BED_LEVELING_GRID_POINTS 3 (probing 6 points and gantry crash to the x-max side).

Mugga6315, 2016-09-13:
Same for me, like already posted here:
https://github.com/MarlinFirmware/Marlin/issues/4662#issuecomment-246774815

thinkyhead, 2016-09-13:
@renzaaa Sorry about that! I've been doing a lot of patches in the last couple days and probably mistyped something. Can you post your configuration so I can test with exactly your settings? Thanks!

thinkyhead, 2016-09-13:
See if this change to line 3454 of `Marlin_main.cpp` makes any difference:
```diff
-       for (int8_t xCount = xStart; xCount != xStop; xCount += xInc) {
+       for (int8_t xCount = xStart; xCount != xStop; xCount += xInc) {
```

thinkyhead, 2016-09-13:
Also… line 3434…
```diff
-     uint8_t zig = auto_bed_leveling_grid_points & 1; //always end at [RIGHT_PROBE_BED_POSITION, BACK_PROBE_BED_POSITION]
+     bool zig = auto_bed_leveling_grid_points & 1; //always end at [RIGHT_PROBE_BED_POSITION, BACK_PROBE_BED_POSITION]
```

Mugga6315, 2016-09-13:
Works with the changes! Thanks.
But the other issue I referred to, in my post stays (x-axis seems to be straight, but y-axis movement is higher at the back and almost crashing into bed in the front)

thinkyhead, 2016-09-13:
@Mugga6315 See my earlier reply on that issue.

renzaaa, 2016-09-14:
@thinkyhead Working now. Thanks.

closed 2016-09-14 by thinkyhead