MarlinFirmware/Marlin #4807: Question: DRV8825 Fault signal

Nocturnal42, 2016-09-14:
Before I go to the trouble of rigging something up, is it possible to have marlin pickup on the fault signal from DRV8825 based motor drivers?

Roxy-3D, 2016-09-14:
Pin 18 of the DRV8825 reflects the Fault status of the chip for over temperature or too much current.  I have the A4988 drivers which do not seem to have a Fault status on them.   I wish they did because I would like to code this up and get it working.

With that said...   It is fairly straight forward to have Marlin watch for a Fault and do something with it.   The harder part is deciding what Marlin should do.   Initially, you could just shut everything down much like a 'Thermal Event' calls Kill() and shuts everything down.    You could define a Fault() function that puts FAULT! on the LCD Panel but does everything else the same as Kill().

To detect a fault, you would find 4 or 5 unused GPIO pins on your AVR chip.   I have code in the devel_ubl branch (M43 and M44) to help you do that.  Once you find the unused pins, you would just wire them up to the Fault pin of your stepstick.    Then in either the timer interrupt or the idle() routine you would write a very fast, quick check that used almost no CPU cycles to verify that no faults existed and continue on with the rest of the code that is already there and doing real work.    If by chance these few lines of code did see a fault, they would call your new Fault() function and shutdown the firmware.

Grogyan, 2016-09-14:
A simple message in the terminal should suffice should there be any actual
problems during printing.

Nocturnal42, 2016-09-15:
Finding unused pins is easy since I have the schematic for my board. Unfortunately, any unused pins are unlikely to have been brought out, and even if they are, the entire board is screened, so following traces is out (though I will have a closer look when I pull the board out, just in case). I also don't want to risk soldering directly to the chip, to much chance of the wires being knocked off and free floating. 

So the easiest signal to get at are for my unused 5th driver.  Which gives me PL0..2, in an easy to access place, conveniently right next to the DRV8825 boards. I don't really think the driver for my extruder is suffering from thermal overload, so 3 pins is enough. Since the fault pin on the drivers is going to be connected to +5V (for A4988 carriers), I'll either have to sever the trace or desolder the pin.

Ideally Marlin would pause the print, wait for the fault to clear, and then continue. Since that's unlikely to be feasible (you can't move the hotend, and having it sitting on top of your print for any length of time is unlikely to be good), I'll settle for killing the print and popping an error message.

Nocturnal42, 2016-09-15:
> Initially, you could just shut everything down much like a 'Thermal Event' calls Kill() and shuts everything down. You could define a Fault() function that puts FAULT! on the LCD Panel but does everything else the same as Kill().

Since kill() take a message pointer to put on the LCD, shouldn't calling it with an appropriate message do the job better than duplicating kill()?

Roxy-3D, 2016-09-15:
> Finding unused pins is easy since I have the schematic for my board. 
> ...
> So the easiest signal to get at are for my unused 5th driver. Which gives me PL0..2, in an easy to access place, 

It is very common to bring out unused pins to a header with connector pins.   What kind of board do you have?  The M43 command in the devel_ubl branch will let you connect a volt meter or LED to the pin and watch for it to be pulsed.

> Ideally Marlin would pause the print, wait for the fault to clear, and then continue. Since that's unlikely to be feasible (you can't move the hotend, and having it sitting on top of your print for [...]

That depends.   If it is not the Z-Motors that over heated, I would say you can move them.   I was thinking you find an unused pin to connect each stepper driver to the AVR chip.  You would monitor all 4 or 5 of them.
     
> Since kill() take a message pointer to put on the LCD, shouldn't calling it with an appropriate message do the job better than duplicating kill()?

Probably...  I didn't reference the code prior to posting.

Nocturnal42, 2016-09-15:
I have a [http://www.geeetech.com/wiki/index.php/GT2560 Geeetech GT2560]. I have a schematic, so I already know where the pins go, and I can count at least 15 that are unconnected. Since there are no unpopulated headers, and all sockets are accounted  for... I'd say they aren't brought out.

Ideally I would monitor all drivers, but since I'm lacking in convenient pins, and the spare driver socket only has 3 GPIO's associated with it, I'll stick with monitoring XYZ for the moment. 

I added a few defines to the pin file for my board, and to Configuration.h, added a few lines to Marlin_main.c to initialize the pins correctly, and this is what I ended up putting in the ISR in stepper.cpp after the call to update the endstops. It should be pretty fast. 
 
    // Check for stepper driver FAULT signal
    #if ENABLED(USE_X_FAULT)
      if (!READ(X_FAULT_PIN)) kill(PSTR("X Fault!"));
    #endif
    #if ENABLED(USE_Y_FAULT)
      if (!READ(Y_FAULT_PIN)) kill(PSTR("Y Fault!"));
    #endif
    #if ENABLED(USE_Z_FAULT)
      if (!READ(Z_FAULT_PIN)) kill(PSTR("Z Fault!"));
    #endif

I had a little trouble with my initial testing, till I realized the steppers had to be moving for the code to run. I didn't look closely, so I don't know if that's because the ISR is disabled when the steppers aren't moving, or if its simple that section doesn't get called.

Roxy-3D, 2016-09-15:
It wouldn't hurt anything to put those same checks in the idle() routine.  Pretty much (not 100% true!!!) idle() only runs when there is nothing happening.   So putting extra checks there would catch the faults even if the stepper motors are not moving.

thinkyhead, 2016-09-16:
Actually, sometimes `idle()` is called when there's something waiting or in a loop. But sometimes it's only calling `manage_inactivity`. So `idle()` isn't a bad place to do periodic general stuff.

closed 2017-10-31 by thinkyhead