MarlinFirmware/Marlin #4808: can't compile 1.1.0-RC7

alejandro-anv, 2016-09-14:
I've compiled 1.1.0-rc6 without problems but when I try to compile RC7 I receive the following error:

For testing I've used all default configuration. Just download and compile.

/tmp/ccRleP5V.ltrans0.ltrans.o: In function `process_next_command()':
ccRleP5V.ltrans0.o:(.text+0x351c): relocation truncated to fit: R_AVR_7_PCREL against `no symbol'
collect2: error: ld returned 1 exit status
exit status 1
Error compilación en tarjeta Arduino/Genuino Mega or Mega 2560.

Tested with Arduino 1.6.8 and 1.6.11 (linux)

thinkyhead, 2016-09-14:
Just like #4788. Please keep an eye on that issue. It seems to be a spurious compiler problem.

alejandro-anv, 2016-09-15:
Thank you: Just moved the declaration of codenum:

uint16_t codenum = 0; // define ahead of goto

a few lines up and it compiled ok.

alejandro-anv, 2016-09-19:
I was testing moving instructions and couldn't compile... 

Can you test if you can compile with this settings?

[config_anv.zip](https://github.com/MarlinFirmware/Marlin/files/480800/config_anv.zip)

thinkyhead, 2016-09-20:
Helpful Tip #1: You no longer need to redefine these to reverse the encoder:
```cpp
#define BTN_EN2 31   //ANV: invertido
#define BTN_EN1 33
```
Instead you should enable one of these:
```cpp
//#define REVERSE_ENCODER_DIRECTION
//#define REVERSE_MENU_DIRECTION
```

thinkyhead, 2016-09-20:
Helpful Tip # 2: You're defining `DELTA_RADIUS` more than once. Don't do that.

thinkyhead, 2016-09-20:
As I suspected, it does seem to be a compiler issue. The compiler doesn't like the `NOZZLE_CLEAN_FEATURE` for some inexplicable reason. I've tried various tricks to get the compiler to accept it, but nothing seems to work. You can see for yourself, if you comment out the following line (2787) it will compile…

```cpp
  Nozzle::clean(pattern, strokes, objects);
```

While `RC7` won't compile with your configuration, `RCBugFix` compiles just fine. So you should update to `RCBugFix` if you want to keep using the nozzle cleaning feature.

Roxy-3D, 2016-09-20:
Would it make sense to make a branch that is ready to compile but won't?   Because with that, the Arduino people could probably chase down the problem and figure out what the problem is.

Is there anybody here that has a good working relationship with the Arduino group that could pass along the link to that branch and get it resolved?

thinkyhead, 2016-09-20:
It's a longstanding issue with certain kinds of branching vs. jumping optimization. It can probably also be worked-around by setting compiler/linker flags instructing it to use jumps instead of tryeing to use branches where possible. The basic issue is that the compiler tries to be predictive about how close together code segments will be, but it gets it wrong in some cases, and this is only discovered when the linker tries to apply the branch distance and discovers that it's too large. If branch code and jump code was the same size, the linker could cleverly substitute a jump, but there's a slight size difference so it just bails out.

boelle, 2016-11-02:
since this is invalid and a dublicate why not close?

closed 2016-11-03 by thinkyhead