MarlinFirmware/Marlin #4809: UBL on CoreXY

adamfilip, 2016-09-15:
@Roxy-3D I finally got my machine behaving to the point where i was able to run a G29 P1
I noticed some funny actions.. such as long pauses (30-40 seconds) once in a while
also when it does a large raise.. it seem to get higher each time, up to the point it was raising up like 200mm and i had to shut it down.

looking forward to testing your latest release :) eta?

Roxy-3D, 2016-09-15:
> looking forward to testing your latest release :) eta?

I think it will be this weekend some time.  I'm reading too many stories about stupid stuff on Yahoo.  So it seems I never get a full day of debugging done. 

I certainly believe you that you are seeing some long pauses, but that doesn't make any sense.   If you see that in the new version we can add some print outs that will isolate exactly what is causing that.   For all practical purposes, I don't do any thing different for CoreXY's than a normal Cartesian machine.  That might be wrong.   I got bit on that with M48 and Delta's.   For the longest time I didn't have a Delta so I couldn't fix it and make M48 work.  (M48 now works on Delta's!!! )   There may be something like that going on with CoreXY's.

adamfilip, 2016-09-15:
M48 works :)

kmbecker13, 2016-09-15:
I also saw this behavior where the bed would drop (+Z direction) rarely during G29 P1, but i also was having some other issues with my endstops ([see 4796](https://github.com/MarlinFirmware/Marlin/issues/4796) at the time so i didn't really worry about it.

It was a very rare event and only happened maybe 2-3 times out of 40 attempts at doing the G29 P1, haven't been able to track down a root cause yet.

adamfilip, 2016-09-21:
@Roxy-3D check out this video.. let me know your thoughts.. ive noticed the Z doing several up down motions between layers..

https://www.youtube.com/watch?v=_t9lsUzMkio

Roxy-3D, 2016-09-21:
@adamfilip  Was it actually changing layer heights each time it did that?   If so, that will be much easier to track down and fix.   The reason I'm asking is the walls of that box you were printing is thicker than one perimeter and it was doing it each time at the same spot.    

If that really is happening each layer change, I know exactly where to look to get the problem fixed!   When I rewrote the mesh_buffer_line() routine to be faster, I moved the Z-Correction up to a higher level.  It used to be done down in the planner.buffer_line() routine.  That was a good place for it because every request funneled together there.   The problem is, at that low of a level a lot of information was lost.    

In the case of the new mesh_buffer_line() it knows when it is doing a horizontal or vertical (Y-Axis) move, and drastically less floating point operations are needed in those cases.  But to do less floating point operations, the Z-Correction has to be done at a higher level where that knowledge is available.  
 
If this funny up and down is happening when it is trying to change layers, it is probably happening with a G1 Z   command and is not associated with any X & Y move.   That will make it much easier to find.

Blue-Marlin, 2016-09-21:
For me it looks like: retract, lift, sweep, go to new position - generated from the slicer.
Our FWRETRACT has no sweep.

adamfilip, 2016-09-21:
using simplify3D to slice.. I dont have any Lift during extraction applied..

adamfilip, 2016-09-21:
I was doing a test print of a single wall box, not spiral

Roxy-3D, 2016-09-21:
> I was doing a test print of a single wall box, not spiral

OK!  That helps to know...  I'll start digging in.   I'm also going to add some verbose debug code to G26 so I can get this other issue figured out:   https://github.com/MarlinFirmware/Marlin/issues/4828#issuecomment-248664190

thinkyhead, 2016-09-21:
Please post the GCode for examination. Chances are it has Z moves explicitly in the code.

adamfilip, 2016-09-21:
Here is the gcode
[TW_Calibration_Cubes.txt](https://github.com/MarlinFirmware/Marlin/files/486249/TW_Calibration_Cubes.txt)

adamfilip, 2016-10-10:
Noticed that the revised code with the switch from MAP M to O isnt working for me
M still works.. O does not

also the notes still reference using M

Roxy-3D, 2016-10-10:
That comment was in regards to G29, right?   The 'O' parameter should work better in G29 now.   I did not find any problems in G26 with its 'M' conversion.

adamfilip, 2016-10-10:
Got it working
https://www.youtube.com/watch?v=y6L8fQW-RkQ

I cant set travel speeds higher then 100mm/s it seems to bind or run into a cpu limit or something else.

Roxy-3D, 2016-10-10:
I really like that printer!   The only problem is I can't see the Z-Axis going up and down as it crosses Mesh Lines in those videos.   And for UBL debugging, that makes things difficult.    

On the travel speed higher than 100mm/s, do you see the Z-Axis moving back and forth very fast?  If your mesh has big corrections in neighboring cells, is it possible your speed is being limited by how fast the Z-Axis can change direction and accelerate the opposite direction?    On my really bad piece of glass, I did notice an issue with that.   But my Z-Axis is painfully slow.  And I had one Mesh Point that was constantly forcing a change in direction of the Z-Axis as the nozzle got near it.

Roxy-3D, 2016-10-10:
> I cant set travel speeds higher then 100mm/s it seems to bind or run into a cpu limit or something else.

I just realized something.   Make sure you think everything is printing correctly before you make this change.   But right now, there is extra debug code.  It is doing the Z Correction the old way, and the new way.   And then making sure the two numbers are very close to the same.   That could be impacting your max speed.    

In Mesh_buffer_line.cpp you will see this code block a number of times.   If you delete each occurance of it, the code will run faster and you should be able to go faster than 100mm/s.     But that code is in there to verify that the new (and faster) mesh corrections are creating the same results as the old code did.

```cpp
//
// debug code to use non-optimized get_z_correction() and to do a sanity check
// that the correct value is being passed to planner.buffer_line()
//
z_optimized = z0;
z0 = blm.get_z_correction( next_mesh_line_x, y);
if ( fabs(z_optimized - z0) > .01 && !isnan(z0) )  {
debug_current_and_destination("!HORIZONTAL z_correction()");
SERIAL_ECHOPAIRPGM("  next_mesh_line_x=", next_mesh_line_x);
SERIAL_ECHOPAIRPGM("  y=", y);
SERIAL_ECHOPAIRPGM("  z0=", z0);
SERIAL_ECHOPAIRPGM("  z_optimized=", z_optimized);
SERIAL_ECHOPAIRPGM("  err=",fabs(z_optimized-z0));
SERIAL_ECHO("\n");
}
```

adamfilip, 2016-10-10:
Thanks I will try that :)

adamfilip, 2016-10-13:
System has been working well recently :)

adamfilip, 2016-10-15:
Before UBL enabled.
[image]

After without any finetuning. pretty good
[image]

closed 2017-10-31 by thinkyhead