MarlinFirmware/Marlin #4810: Cannot print from SD Card using BOARD_K8400 with REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

SwiftNick, 2016-09-15:
PROBLEM:
Prints from SDCard are not possible using BOARD_K8400  (aka 3DualDrag) board 
with REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

The SDCard content can be listed on the panel, but when a file is selected, nothing happens.

CAUSE:
The SD Card detect is not working due to missing pin definitions

SOLUTION:
Add SD detect pin definitions to K8400.h

```
#undef  SD_DETECT_PIN  
#define SD_DETECT_PIN  53
```

thinkyhead, 2016-09-15:
I see that the pin is normally set to 49. Is this a universal exception for 3DRAG with that particular display, or do you think it should apply to all the displays where the pin is being set to 49 (by the `pins_RAMPS.h` file)?

birkett, 2016-09-16:
I'll do some testing later to see whats happening.

Provisionally, I would say that changes will only be needed for the 8400 (not 8200 or generic 3Drag). I'm certain that there would have been bug reports for the earlier models by now.

SwiftNick, 2016-09-16:
@thinkyhead, 
pin 53 is not static in my config. The software detects when I insert and remove the SD card.

In older 1.0.2 (that works just fine), LCD_SDSS was used and its value varied depending on the display type used. I have a REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER so LCD_SDSS was set to 53 under an `#ifdef REPRAP_DISCOUNT_SMART_CONTROLLER` branch in pins.h

The SD_DETECT_PIN definition was introduced in more recent builds and seems to be defined by the type of display you are using in pins_RAMPS.h

@birkett, 
My SD_DETECT_PIN override is too harsh as it doesn't take different displays into account so we need a definition for the variable when using a 3DRag/k8200/k8400 and the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER.

The 3DRag/K8200  & 3DualDrag/K8400 share the same basic architecture with many common pin definitions. I have the schematics if anyone needs them.

With the k8200/k8400 and the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER, the current SD_DETECT_PIN is set to 49 which is wrong.

SwiftNick, 2016-09-16:
I think this should be merged with #4316

closed 2016-09-19 by thinkyhead