MarlinFirmware/Marlin #4817: RC7 Wait on cool down issue

Alex9779, 2016-09-16:
I am using RC7 and after a print I want to wait for the bed to cool down to 30°.
So basically my last line in a print is M190 R30 followed by M140 S0 to turn off the bed completely.
But since RC7 Marlin does not wait until 30° C is reached anymore but the print end somewhere between 55° C and 45° C
Here is a log of the end of a print:
```
Recv:  T:46.9 /0.0 B:46.1 /30.0 T0:46.9 /0.0 T1:21.1 /0.0 @:0 B@:0 @0:0 @1:0 W:?
Recv:  T:46.7 /0.0 B:46.1 /30.0 T0:46.7 /0.0 T1:21.2 /0.0 @:0 B@:0 @0:0 @1:0 W:?
Recv:  T:46.5 /0.0 B:46.1 /30.0 T0:46.5 /0.0 T1:20.9 /0.0 @:0 B@:0 @0:0 @1:0 W:?
Recv:  T:46.6 /0.0 B:46.0 /30.0 T0:46.6 /0.0 T1:21.2 /0.0 @:0 B@:0 @0:0 @1:0 W:?
Recv: ok
Send: N78527 M140 S0*106
Recv: ok
Send: N78528 M117 Print complete*124
Recv: ok
```

There is no error or anything else, it just continues with the next command even though the target temp of 30° C has not been reached yet...

Is there an other timeout or anything taken into account?

brainscan, 2016-09-16:
Have you tried RCBugFix? I noticed the same problem a short time ago but think it's fixed again now as mine is working again. I think some changes were made to try and avoid a wait forever situation but I never tested it down to 30°.

Alex9779, 2016-09-16:
Nope, didn't check RCBugFix yet...
Is there a config option to have an influence on that?

thinkyhead, 2016-09-16:
I know we've been over this code a lot with a fine-toothed comb, but can't remember if the last patch was after RC7. Definitely check `RCBugFix` to see if we finally did get it right.

thinkyhead, 2016-09-16:
The only setting with an additional effect on the behavior is `MIN_COOLING_SLOPE_DEG_BED` which is set in `Marlin_main.cpp` itself. The default is 1.5, meaning the bed has to cool by 1.5 degrees C within `MIN_COOLING_SLOPE_TIME_BED` seconds (1.5°C per minute by default). Probably those settings are fine. There would be at least a 1m delay even if `MIN_COOLING_SLOPE_DEG_BED` were (somehow) too small.

boelle, 2016-11-05:
@Alex9779 have you rechecked with latest RCBugfix?

Alex9779, 2016-11-05:
Nope but I won't be able to do for a while so closing...

closed 2016-11-05 by Alex9779