MarlinFirmware/Marlin #4819: Filament Runout Sensor Not Working 1.1.0 RC7 release

gordo3di, 2016-09-16:
I have spent two days racking my brain on this. First off thank you to everyone for all of the great work on this, it seems many people forget the amount of work that goes into it all. I am trying to learn and use github more so I apologize if my markdown styling is poor.

Ok so here's where I am at:

Basically it seems as though marlin just won't do anything when the filament sensor is triggered. I am hooked up to pin 4 (for servo 3) but I have tried many others already. D5, D6, 19, 2 etc... I am using a basic endstop switch on a gadgets 3d ramps 1.4 setup.

I have been using (and building) 3d printers with marlin for close to 3 years and I can make my way around the firmware however I am not a coding expert. I am using the 1.1.0 RC7 release of the firmware which is why the code is slightly different than the earlier releases (from thinkyhead's changes).

I have been following the threads here:
https://github.com/MarlinFirmware/Marlin/issues/2907
https://github.com/MarlinFirmware/Marlin/issues/4059

Thanks to Roxy-3D I tried the M42 P4 S255 command and that appears to work fine. I can put 5V to the pin as checked with a multimeter. This wells me my wires are good and I have the correct pin asignment. I've also tried editing the main loop in Marlin_main.cpp to remove the SD card and print job timer requirements in case they were causing problems.

      #if ENABLED(FILAMENT_RUNOUT_SENSOR)
    if (!(READ(FIL_RUNOUT_PIN) ^ FIL_RUNOUT_INVERTING)) {
    //if ((IS_SD_PRINTING || print_job_timer.isRunning()) && !(READ(FIL_RUNOUT_PIN) ^ FIL_RUNOUT_INVERTING)) {
      handle_filament_runout();
    }
    #endif

I've tried countless other things which may be difficult to list them all:

Tried commenting out the endstop pullups in Configuration.h
    //#define ENDSTOPPULLUP_FIL_RUNOUT

Removed some of the sanity checks to see if one was causing an issue (like the pin sanity check) but nothing.

The M600 Filament change feature works fine when selected from the menu (although sometimes it runs several times but thats another issue) so i know this isn't an issue.

I had a filament sensor working many years ago but that was done using the Pause Feature outlined here:
http://forums.reprap.org/read.php?1,297350

Basically it's as though the main loop just isn't recognizing the pin input when its triggered, or it recognizes it but nothing happens.

Thanks again for any help.

Roxy-3D, 2016-09-16:
Have you put the GPIO pin you are using into 'Input Mode' ?   If you can see M42 toggle the value on the pin, that means you have the correct pin identified and connected.  However,  M42 does this:
```cpp
    pinMode(pin_number, OUTPUT);
    digitalWrite(pin_number, pin_status);
    analogWrite(pin_number, pin_status);
```
You need the pin in the correct mode to read from it.  Did you do this some where (like init() ) prior to using it? 
```cpp
   pinMode(pin_number, INPUT_PULLUP);
   digitalRead(pin_number, pin_status);
```
If you add this code block to Marlin_main.cpp you will be able to scan a range of pins again and again and see if the connected pin is seeing a value change on it:
```cpp
/**
 * M44: Input Scan    Scan for a pin that is connected to a sensor or switch
 *
 *      This command will scan the non-sensitive pins and read them with Pull Up Resistor
 *      mode enabled.
 *
 * S	Start Pin number.   If not given, will default to 0
 *
 * E	End Pin number.   If not given, will default to 127 
 *
 * W    Wait time (in miliseconds) between scans.  If not given will default to 500
 *
 * R    Repeat range scan this many times
 */

inline void gcode_M44() {
char c;
int p, i, j, s=0, e=127, w=500, repeat_cnt=1; 

  if (code_seen('R')) 
    repeat_cnt = code_value_int();

  if (code_seen('S')) 
    s = code_value_int();

  if (code_seen('E')) 
    e = code_value_int();

  if (code_seen('W')) 
    w = code_value_int();

  for(p=s; p<=e; p++) {
      if ( sensitive_pin(p) )
	  SERIAL_PROTOCOL(".");
      else {
	  j = p % 10;
	  c = '0' + j;
	  SERIAL_PROTOCOL(c);
      }
  }
  SERIAL_PROTOCOL("\n");

  for(i=0; i<repeat_cnt; i++) {
	  for(p=s; p<=e; p++) {
	      pinMode(p, INPUT_PULLUP);
              if ( sensitive_pin(p)) {
		  SERIAL_PROTOCOL(".");
	      } else {
		  j = digitalRead( p );
		  if ( j & 0x01 )
		     SERIAL_PROTOCOL("1");
		  else
		     SERIAL_PROTOCOL("0");
	      }
      }
      SERIAL_ECHO("\n");
      idle();
      delay(w);
  }
} 
```

gordo3di, 2016-09-16:
Thanks @Roxy-3D .

I'm using the same code from the release which has this just below the setup_killpin() code section:

marlin_main.cpp
```c
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
    
   void setup_filrunoutpin() {
    pinMode(FIL_RUNOUT_PIN, INPUT);
    #if ENABLED(ENDSTOPPULLUP_FIL_RUNOUT)
       WRITE(FIL_RUNOUT_PIN, HIGH);
    #endif
    }
#endif
```

The funny thing is the pin will output 5V unless i comment out the ENDSTOPPULLUP conditional kind of how you mentioned (which i guess makes sense to me). I'm confused by the reason for having "WRITE(FIL_RUNOUT_PIN, HIGH);" at all. I don't fully understand using the arduino internal pullup resistors which I think is what the pullup is for (unless im wrong). When I comment out the conditional, the pin no longer outputs 5V but it doesn't seem to input either.

I just tried changing the code below but it didn't work.

```c
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
    
   void setup_filrunoutpin() {
    pinMode(FIL_RUNOUT_PIN, INPUT);
    digitalRead(FIL_RUNOUT_PIN);
    #if ENABLED(ENDSTOPPULLUP_FIL_RUNOUT)
       WRITE(FIL_RUNOUT_PIN, HIGH);
    #endif
    }
#endif
```

I will try the pin scan code and see what I can come up with.

gordo3di, 2016-09-16:
Ugh, finally just figured it out. I understand whats happening now and its totally my fault. I figured it was something simple I was doing wrong. I was under the impression the switch should be between the 5V and D4 pins. When I hook it up to the D4 and Ground pin the switch triggers just fine. Early on I tried wiring the switch this way but it didn't work and I never went back to try again. Thanks for the input. 

For other people with the same issue here's the final code I used in marlin_main.cpp
```c
#if ENABLED(FILAMENT_RUNOUT_SENSOR)

  void setup_filrunoutpin() 
    pinMode(FIL_RUNOUT_PIN, INPUT);
    #if ENABLED(ENDSTOPPULLUP_FIL_RUNOUT)
       WRITE(FIL_RUNOUT_PIN, HIGH);
    #endif

#endif
```

Roxy-3D, 2016-09-16:
Originally...  There was not a:
```cpp
pinMode(pin_number, INPUT_PULLUP);
```
defined.   In order to do an internal pull up, you had to put the pin into INPUT mode and then output a logic 1 which would drive the pull up resistor.  Now...  You can just tell it to go into INPUT_PULLUP mode.

closed 2016-09-17 by thinkyhead