MarlinFirmware/Marlin #4823: After homing distance is not expected

ihouses, 2016-09-16:
Hello,

I have a delta printer, I configure MANUAL_Z_HOME_POS to 300  and

#define X_HOME_BUMP_MM 10
#define Y_HOME_BUMP_MM 10
#define Z_HOME_BUMP_MM 10

So I expect that after G28 the z positions will be 300-10 = 290mm but every time that
I do home the Z position is 272.83mm

Any idea why???

thinkyhead, 2016-09-17:
Possibly due to this…
```cpp
  #if ENABLED(DELTA)
    // move to a height where we can use the full xy-area
    do_blocking_move_to_z(delta_clip_start_height);
  #endif
```
When you do `G1 Z0` is the nozzle at the correct height — right up to the bed?

ihouses, 2016-09-17:
When I execute G1 Z0 the nozzle is about 5mm up from the bed. 

I am just starting to configure my delta with the new marlin. But I remember from older marlin versions that after home the printer position was 300-10 = 290

I will chck the function that you mention. 

Thanks!

ihouses, 2016-09-17:
ok,

Marling calculates automatically a Z position after homing using this function:

  float delta_safe_distance_from_top() {
    float cartesian[3] = {
      LOGICAL_X_POSITION(0),
      LOGICAL_Y_POSITION(0),
      LOGICAL_Z_POSITION(0)
    };
    inverse_kinematics(cartesian);
    float distance = delta[TOWER_3];
    cartesian[Y_AXIS] = LOGICAL_Y_POSITION(DELTA_PRINTABLE_RADIUS);
    inverse_kinematics(cartesian);
    return abs(distance - delta[TOWER_3]);
  }

I was modifiyng this:

#define X_HOME_BUMP_MM 10
#define Y_HOME_BUMP_MM 10
#define Z_HOME_BUMP_MM 10 

But this is another thing not the position after homing.

thinkyhead, 2016-09-17:
Post your `Configuration.h` so I can test it and see what's going on.

thinkyhead, 2016-09-19:
Seems to have been a typo. The issue has been patched in `RCBugFix`.

thinkyhead, 2016-09-19:
I've been patching up `RCBugFix`. Give it a try at your earliest convenience.

ihouses, 2016-09-19:
I have uploaded the last version of RCBugFix and now after home the Z position is 238.79mm
The MANUAL_Z_HOME_POS  is still 300. May be this is the normal behaviour :?

I enclose my configuration files.

[CONFIGURATION_MARLINC_RC7_DELTA.zip](https://github.com/MarlinFirmware/Marlin/files/481173/CONFIGURATION_MARLINC_RC7_DELTA.zip)

thinkyhead, 2016-09-22:
Enable `DEBUG_LEVELING_FEATURE` and re-flash the firmware. Before doing `G28` issue the command `M111 S255`. Then do a `G28`. Paste the log output here so we can see the details of what is going on.

ihouses, 2016-09-22:
This is the log:

21:42:01.479 : start
21:42:01.481 : echo:Marlin 1.1.0-RCBugFix
21:42:01.481 : echo: Last Updated: 2016-07-26 12:00 | Author: (none, default config)
21:42:01.481 : Compiled: Sep 22 2016
21:42:01.484 : echo: Free Memory: 3741  PlannerBufferBytes: 1232
21:42:01.489 : echo:Hardcoded Default Settings Loaded
21:42:01.489 : echo:Steps per unit:
21:42:01.493 : echo:  M92 X160.00 Y160.00 Z160.00 E96.00
21:42:01.493 : echo:Maximum feedrates (mm/s):
21:42:01.493 : echo:  M203 X500.00 Y500.00 Z500.00 E25.00
21:42:01.497 : echo:Maximum Acceleration (mm/s2):
21:42:01.497 : echo:  M201 X6000 Y6000 Z6000 E9000
21:42:01.501 : echo:Accelerations: P=printing, R=retract and T=travel
21:42:01.501 : echo:  M204 P3000.00 R3000.00 T3000.00
21:42:01.509 : echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
21:42:01.510 : echo:  M205 S0.00 T0.00 B20000 X20.00 Z20.00 E5.00
21:42:01.513 : echo:Home offset (mm)
21:42:01.513 : echo:  M206 X0.00 Y0.00 Z0.00
21:42:01.513 : echo:Endstop adjustment (mm):
21:42:01.514 : echo:  M666 X0.00 Y0.00 Z0.00
21:42:01.521 : echo:Delta settings: L=diagonal_rod, R=radius, S=segments_per_second, ABC=diagonal_rod_trim_tower_[123]
21:42:01.522 : echo:  M665 L228.00 R108.20 S160.00 A0.00 B0.00 C0.00
21:42:01.522 : echo:Material heatup parameters:
21:42:01.525 : echo:  M145 S0 H180 B70 F255
21:42:01.525 : echo:  M145 S1 H240 B100 F255
21:42:01.525 : echo:PID settings:
21:42:01.529 : echo:  M301 P22.20 I1.08 D114.00
21:42:01.529 : echo:Filament settings: Disabled
21:42:01.530 : echo:  M200 D3.00
21:42:01.530 : echo:  M200 D0
21:42:05.142 : echo:SD card ok
21:42:05.318 : echo:SD card ok
21:42:05.416 : echo:DEBUG:INFO,ERRORS
21:42:05.420 : echo:Active Extruder: 0
21:42:05.424 : Begin file list
21:42:05.438 : PEP.GCO
21:42:05.438 : GEAR.GCO
21:42:05.438 : PITO.GCO
21:42:05.441 : PENCIL.GCO
21:42:05.442 : End file list
21:42:05.449 : Begin file list
21:42:05.462 : PEP.GCO
21:42:05.462 : GEAR.GCO
21:42:05.462 : PITO.GCO
21:42:05.466 : PENCIL.GCO
21:42:05.466 : End file list
21:42:05.490 : Begin file list
21:42:05.502 : PEP.GCO
21:42:05.503 : GEAR.GCO
21:42:05.503 : PITO.GCO
21:42:05.507 : PENCIL.GCO
21:42:05.507 : End file list
21:42:05.519 : echo:DEBUG:INFO,ERRORS
21:42:05.523 : echo:Active Extruder: 0
21:45:11.863 : echo:DEBUG:ECHO,INFO,ERRORS,DRYRUN,COMMUNICATION,LEVELING
21:45:14.640 : echo:N165 M105*37
21:45:17.700 : echo:N166 M105*38
21:45:20.785 : echo:N167 M105*39
21:45:23.824 : echo:N168 M105*40
21:45:26.884 : echo:N169 M105*41
21:45:28.359 : echo:N170 G28*21
21:45:28.359 : >>> gcode_G28
21:45:28.359 : Machine Type: Delta
21:45:28.359 : Probe: NONE
21:45:28.363 : current_position=(0.00, 0.00, 238.80) : setup_for_endstop_or_probe_move
21:45:28.363 : > endstops.enable(true)
21:45:28.367 : current_position=(0.00, 0.00, 0.00) : sync_plan_position
21:45:28.932 : >>> homeaxis(X)
21:45:28.937 : current_position=(0.00, 0.00, 0.00) : sync_plan_position
21:45:28.937 : > 1st Home 300.00
21:45:28.941 : current_position=(0.00, 0.00, 0.00) : sync_plan_position
21:45:29.051 : current_position=(0.00, 0.00, 0.00) : sync_plan_position
21:45:29.555 : > 2nd Home 20.00
21:45:29.556 : <<< homeaxis(X)
21:45:29.556 : >>> homeaxis(Y)
21:45:29.559 : current_position=(20.00, 0.00, 0.00) : sync_plan_position
21:45:29.563 : > 1st Home 300.00
21:45:29.564 : current_position=(20.00, 0.00, 0.00) : sync_plan_position
21:45:29.674 : current_position=(20.00, 0.00, 0.00) : sync_plan_position
21:45:30.178 : > 2nd Home 20.00
21:45:30.178 : <<< homeaxis(Y)
21:45:30.182 : >>> homeaxis(Z)
21:45:30.182 : current_position=(20.00, 20.00, 0.00) : sync_plan_position
21:45:30.187 : > 1st Home 450.00
21:45:30.187 : current_position=(20.00, 20.00, 0.00) : sync_plan_position
21:45:30.301 : current_position=(20.00, 20.00, 0.00) : sync_plan_position
21:45:30.805 : > 2nd Home 20.00
21:45:30.805 : <<< homeaxis(Z)
21:45:30.805 : >>> set_axis_is_at_home(X)
21:45:30.805 : For X axis:
21:45:30.805 : home_offset = 0.00
21:45:30.806 : position_shift = 0.00
21:45:30.809 : soft_endstop_min = -100.00
21:45:30.809 : soft_endstop_max = 100.00
21:45:30.809 : > home_offset[X] = 0.00
21:45:30.813 : current_position=(0.00, 20.00, 20.00) :
21:45:30.813 : <<< set_axis_is_at_home(X)
21:45:30.814 : >>> set_axis_is_at_home(Y)
21:45:30.814 : For Y axis:
21:45:30.817 : home_offset = 0.00
21:45:30.817 : position_shift = 0.00
21:45:30.817 : soft_endstop_min = -100.00
21:45:30.818 : soft_endstop_max = 100.00
21:45:30.822 : > home_offset[Y] = 0.00
21:45:30.822 : current_position=(0.00, 0.00, 20.00) :
21:45:30.822 : <<< set_axis_is_at_home(Y)
21:45:30.826 : >>> set_axis_is_at_home(Z)
21:45:30.826 : For Z axis:
21:45:30.826 : home_offset = 0.00
21:45:30.826 : position_shift = 0.00
21:45:30.827 : soft_endstop_min = 0.00
21:45:30.830 : soft_endstop_max = 300.00
21:45:30.830 : > home_offset[Z] = 0.00
21:45:30.834 : current_position=(0.00, 0.00, 300.00) :
21:45:30.834 : <<< set_axis_is_at_home(Z)
21:45:30.835 : current_position=(0.00, 0.00, 300.00) : sync_plan_position_kinematic
21:45:30.838 : current_position=(0.00, 0.00, 300.00) : (DELTA)
21:45:30.838 : >>> do_blocking_move_to(0.00, 0.00, 238.80)
21:45:30.842 : destination=(0.00, 0.00, 300.00) : set_destination_to_current
21:45:30.846 : destination=(0.00, 0.00, 238.80) : prepare_move_to_destination_raw
21:45:30.850 : current_position=(0.00, 0.00, 238.80) : zone border move
21:45:30.851 : current_position=(0.00, 0.00, 238.80) : xy move
21:45:30.853 : <<< do_blocking_move_to
21:45:32.456 : current_position=(0.00, 0.00, 238.80) : clean_up_after_endstop_or_probe_move
21:45:32.457 : <<< gcode_G28
21:45:32.464 : echo:N171 M105*32

ihouses, 2016-09-22:
M503 command result is:

21:47:45.267 : echo:Steps per unit:
21:47:45.267 : echo:  M92 X160.00 Y160.00 Z160.00 E96.00
21:47:45.271 : echo:Maximum feedrates (mm/s):
21:47:45.271 : echo:  M203 X500.00 Y500.00 Z500.00 E25.00
21:47:45.271 : echo:Maximum Acceleration (mm/s2):
21:47:45.275 : echo:  M201 X6000 Y6000 Z6000 E9000
21:47:45.275 : echo:Accelerations: P=printing, R=retract and T=travel
21:47:45.279 : echo:  M204 P3000.00 R3000.00 T3000.00
21:47:45.288 : echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
21:47:45.288 : echo:  M205 S0.00 T0.00 B20000 X20.00 Z20.00 E5.00
21:47:45.288 : echo:Home offset (mm)
21:47:45.291 : echo:  M206 X0.00 Y0.00 Z0.00
21:47:45.292 : echo:Endstop adjustment (mm):
21:47:45.292 : echo:  M666 X0.00 Y0.00 Z0.00
21:47:45.296 : echo:Delta settings: L=diagonal_rod, R=radius, S=segments_per_second, ABC=diagonal_rod_trim_tower_[123]
21:47:45.300 : echo:  M665 L228.00 R108.20 S160.00 A0.00 B0.00 C0.00
21:47:45.300 : echo:Material heatup parameters:
21:47:45.304 : echo:  M145 S0 H180 B70 F255
21:47:45.304 : echo:  M145 S1 H240 B100 F255
21:47:45.304 : echo:PID settings:
21:47:45.304 : echo:  M301 P22.20 I1.08 D114.00
21:47:45.308 : echo:Filament settings: Disabled
21:47:45.308 : echo:  M200 D3.00
21:47:45.308 : echo:  M200 D0
21:47:47.762 : echo:N220 M105*39

thinkyhead, 2016-09-23:
Sure enough. It's just lowering the carriages after homing to the height (Z=238.80) where full XY movement is possible. Currently that is the standard behavior. At the point where homing is completed, before this move, the Z position is set to 300.0, as configured.

ihouses, 2016-09-24:
Thanks so much for the clarification.