MarlinFirmware/Marlin #4824: Problem with MicroStepping in RC7

jmceara, 2016-09-17:
Hi.
I'm building first printer and I'm having problem with microstepping and I think it's related to Marlin.
In the X and Y axis I'm using GT2 belt with 20 Tooth pulley on stepper (i've checked this 4 times, counting tooths on pulley and checking label on belt). I've tried with DRV8825 and A4988....both I was using all 3 jumpers inserted, which means 1/16 in A4988 and 1/32 in DRV8825. After inserting drivers and power up board, I've checked M0, M1 and M2 on driver pins and all thre are high (5v), which is correct.
In Marlin configuration, I'm using this (for DRV8825):
```
#define DEFAULT_AXIS_STEPS_PER_UNIT   {160, 160, 1600, 95}  // default steps per unit for Ultimaker
#define DEFAULT_MAX_FEEDRATE          {500, 500, 2, 25}    // (mm/sec)
#define DEFAULT_MAX_ACCELERATION      {3000,3000,15,2000}    // X, Y, Z, E maximum start speed for accelerated moves. E default values are good for Skeinforge 40+, for older versions raise them a lot.

#define DEFAULT_ACCELERATION          1200    // X, Y, Z and E acceleration in mm/s^2 for printing moves
#define DEFAULT_RETRACT_ACCELERATION  1200    // E acceleration in mm/s^2 for retracts
#define DEFAULT_TRAVEL_ACCELERATION   1200    // X, Y, Z acceleration in mm/s^2 for travel (non printing) moves

// The speed change that does not require acceleration (i.e. the software might assume it can be done instantaneously)
#define DEFAULT_XYJERK                20.0    // (mm/sec)
#define DEFAULT_ZJERK                 0.4     // (mm/sec)
#define DEFAULT_EJERK                 5.0    // (mm/sec)

```

The problem is: when I connect my repetir Host and move X or Y my 10mm, they move 20mm. If I change those values to hald, they move correctly. I already checked eeprom values, making and factory reset (If I'm correct, M502) to get configuration.h values, still same problem.
After changing drivers to A4988 and changin values of steps/mm to 80, same problem.....when try to move 10mm, they move 20mm. Chaning values to 40, they move correctly.

Also, my stepper is 1.8º (200 per full rotation). I'm using Marlin RC7.

moxadesign, 2016-09-17:
Hi jmceara, i use the same machine hardware configuration and it work fine, but into the software i have set 160.03 steps for X and Y axis and 1600.00 for Z.  Can be possible to have to insert a decimal point for make compiler directive for setting floating constant.
Try and send consideration.
Bye.

jmceara, 2016-09-17:
Still same problem :(
Any other idea or test?

jmceara, 2016-09-17:
One curious thing: on extruder stepper, I have MK8 extruder with 11mm drive gear and DRV8825 as driver, using jumper only on M3 (Ramps 1.4), which means 1/16.
(200 * 16) / (11 * pi) = 92,60 and I don't remember where, but recomends use 95 for MK8 extruder. I'm using 95 and it's working fine....I've measured 10mm of fillament and it's perfect with 95, which means that this 'axis' is working fine with 1/16 configured in driver AND in marlin. If I had the same problem of all other axis, I would use 46.300 to get the it working correctly.

Also, I tried to swap/exchange drivers from extruder and axis to test, ot's not driver related.

moxadesign, 2016-09-17:
Question: into the slicing program do you have a section where you insert the step/mm parameters??
 
Moxà Design
info@moxa-design.it

thinkyhead, 2016-09-18:
For DRV8825 the correct jumper setting for 32x is 101, not 111.
```
 001  2x  ( 400 spr)
 010  4x  ( 800 spr)
 011  8x  (1600 spr)
 100 16x  (3200 spr)
 101 32x  (6400 spr)
```

jmceara, 2016-09-18:
Same problem :(
I've re-checked stepper angle (1.8), belt (GT2) and pulley (20T). I'll try to make some code to spin motor exactly one revolution to test pulse count.

thinkyhead, 2016-09-18:
Don't forget `M500` after `M502` if you intend to save the defaults back to EEPROM.

str8up2, 2016-09-26:
I'm having the same type of problem. I have a Delta with Marlin 1.1.0 RC7 loaded as of last night 09/25/2016 . I moved all the parameters from a working version to RC7. Mine was moving twice the distance and twice as fast. I know it is a micro-stepping issue. I also noticed that if I disable EEPROM support the micro-steps return to the right value. I am using the DRV8825 set to 32. My motors are 0.9 deg. When I enable EEPROM support I get double the distance. 
Also I have noticed that all 3 towers step approx 40mm away from home after homing - where can I change this?

thinkyhead, 2016-09-27:
@str8up2 You cleared the EEPROM (`M502` followed by `M500`) and still see the issue? What does the output of `M503` look like when EEPROM is off versus when EEPROM is on?

str8up2, 2016-09-27:
@thinkyhead Thanks for your help. The M502 followed by M500 cleared the micro-steps issue.  As far as the 40mm away from home after homing issue is concerned - for me you could leave that code.

closed 2016-09-27 by thinkyhead