MarlinFirmware/Marlin #4825: Feature Request - GCode setable Variables

RogerRepRap, 2016-09-17:
I am playing with the concept of a portable cable driven machine using kinematics. The kinematics are not a problem. but the math will need to be adjusted to compensate for the exact placement of the capstans. This is not a problem if some correction factors could be sent to the kinematics and re-compiling firmware is not a good option as correction will be made through repeated approximations and observation.

Would it be possible to define some number (<=10?) of non-persistent user variables that could be set by GCode? Maybe something in the format of Mxxxx S value. where S is the variable number? This may have more general application than my special case.

Roger

thinkyhead, 2016-09-18:
`M92` exists to set the steps-per-mm for each axis. What other factors do you need to alter?

RogerRepRap, 2016-09-20:
Consider a machine that consists of a wheeled platform that is pulled around by three cable capstans. The capstans are at the corners of an equilateral (60 degrees apart) triangle. The kinematics are pretty simple. To move the platform the capstans will either pull or relax. As the rig is portable the capstans may not be exactly positioned at 60 degrees in the field and the sin 60 in the kinematics will need to be replaced to reflect the actual setup. I envisioned determining the actual positioning by trial and error and updating the variables in the header of the g code. Re-compiling is not an option.

I was thinking that this may have other applications but if not it will not be hard for me to add in a separate branch.

Roger

thinkyhead, 2016-09-21:
I look forward to seeing your work. Please keep us posted as your development progresses!

Roxy-3D, 2016-09-21:
> The capstans are at the corners of an equilateral (60 degrees apart) triangle. The kinematics are pretty simple. To move the platform the capstans will either pull or relax. As the rig is portable [...]

This would be useful for a 'Street Painter' that sets up to do a 20' x 20' Parking Lot or Side Walk painting.

thinkyhead, 2017-10-31:
Do `M665` or `M666` in the latest code fulfill some of this request?

Grover49, 2018-09-17:
I have been looking for a variables option in marlin for years, every time I find reference to it there is someone saying it isn't needed, and someone saying it can't be done. I have limited knowledge of C++
but it seems to me variables are everywhere. The options in Marlin I would like to see is setting variables for coordinate systems so I could allow algorithms to move parts around in real time based on part or  machine variables. For example for batch runs I print a small part 20 times on a platter with a set distance between parts, if I want to change the number of parts or spacing I have to re post or manually edit the code. Not to mention the size of a program as the part count increases. If I could use variables I could post the program for a single part, loop it on a count variable and offset it's coordinate system between  loops for spacing. Making things instantly ready for customization and with a program size 10 to 100 times smaller.

AnHardt, 2018-09-17:
We also can't do loops. Ask your cad, slicer or host to do that.

thinkyhead, 2018-09-17:
@Grover49 — Marlin now supports workspaces and arc / bezier orientations. It will soon support tool offsets, and will continue to be extended with more CNC-like features as those submissions are received.

boelle, 2019-02-18:
@RogerRepRap 

Do `M665` or `M666` in the latest code fulfill some of this request?

boelle, 2019-06-22:
missing label

boelle, 2019-07-20:
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-20 by boelle