MarlinFirmware/Marlin #4826: dual Extruder : under extrusion

Jinxycob, 2016-09-17:
the Issue: appears E1 wilfully  under extruding!  for a shared value.
 setup: using two titans and  a chimera,  steppers aint the same but the E steps are close for both

    so far I swapped  hardware over between E1 and E2 "driver, nozzle heaters the lot,  and every time E1 produces fine cubes!!". yet E2 fails every time.
using the latest marlin I altered the config.h  Extruders  to 2 and that's about all I changed for a dual  extruder, am I missing something ?.

paltoni, 2016-09-18:
the extruder gears all is same tipe? If not you should change it, or wait for the software update where you can change the "step per unit" value on the configuration.h for the different extruders.

thinkyhead, 2016-09-18:
If the E-steps differs for your second extruder, there is a workaround currently. You can insert `M221 T1 S120` (for example) into your GCode to get 120% flow-rate on the second extruder. This affects E movement both forwards and backwards.

Jinxycob, 2016-09-18:
the two setups are identical, so Esteps value is good for both extruders, yet E2 is under extrudering  which makes no sense if they both sharing the same Esteps,,, "that making sense"

thinkyhead, 2016-09-18:
Is this what you find when you measure the amount of filament being pulled into the extruder? For example, if you make a mark on the filament and move E by 100mm does it fail to move 100mm?

Jinxycob, 2016-09-18:
it does moves to a 100,,  more puzzled  now, I'll make a video in the morning and switch to cura , been using slic3r

Jinxycob, 2016-09-19:
recall the bombers, seems it a slicer problem , get a fine first layers with cura 

think we can close this one

closed 2016-09-19 by Jinxycob