MarlinFirmware/Marlin #4827: Incorrect axis movement when using REVERSE_ENCODER_DIRECTION

mperdue, 2016-09-17:
I downloaded the current rc-fix today and built it with REVERSE_ENCODER_DIRECTION enabled. This corrected both the direction used to increase a value and the direction of the menu selection on my printer. Well, almost. It also reversed the direction the move axis items on the menu so now moving the axis in a positive direction is accomplished by turning the knob counter clockwise.

Blue-Marlin, 2016-09-17:
`REVERSE_MENU_DIRECTION` ?!

mperdue, 2016-09-17:
That reverses the menu so it's wrong again. I didn't check to see if that would reverse the axis movement. I'll give it a shot after my current print project is complete.

thinkyhead, 2016-09-18:
I see. So let me see if I have this summary correct:
- `REVERSE_ENCODER_DIRECTION` by itself: menu direction is good, value editing is good, axis movement is backward.
- `REVERSE_MENU_DIRECTION` by itself: menu direction is good, value editing is backward, axis movement is good.
- Using both together: menu direction is backward, value editing is good, axis movement is backward.

mperdue, 2016-09-18:
Yes, that is exactly the situation for the combinations given.

thinkyhead, 2016-09-18:
I forgot one case.
- When neither option is enabled:
  - menu direction is backward
  - value editing is backward
  - axis movement is good
…?

thinkyhead, 2016-09-18:
Color me stumped. According to my reading of `ultralcd.cpp` value editing and axis movement must always correspond. If values go up with clockwise movement, axis positions also go up with clockwise movement.

Do you happen to have a printer where the bed is rotated so that the XY home position is in the back-right corner?

mperdue, 2016-09-18:
Well, the concept of front and back is somewhat arbitrary but...

If I view my printer such that the X axis moves left to right then:
  * negative X is to the left and positive to the right
  * negative Y is away from me and positive moves towards me
  * negative Z moves the XY gantry down and positive up

so, X0Y0 puts the X all the way to the left and the Y all the way to the back which, if I have correctly interpreted your question is the front left corner.

mperdue, 2016-09-18:
I currently have a print running but when it's done I'll verify everything and post again. As I think about it, I don't know that I checked all axes with the control knob. I know the Z axis numbers increase when I turn the knob clockwise but maybe that's a point of view thing. If I had a printer where bed moved up and down in the Z axis and was fixed in the XY plane the bed would move down in the positive direction which would seem more natural to the user when moving the knob counterclockwise.

This is a long print so I probably won't get to check it till tomorrow unless the print fails before it's done.

thinkyhead, 2016-09-18:
Note: When there's a moving bed we don't reckon Y in terms of the bed, but in terms of the nozzle's position relative to the bed. So "negative Y" means the nozzle is over the front of the bed in a typical setup, while "positive Y" places the nozzle over the back of the bed.

thinkyhead, 2016-09-18:
> the bed would move down in the positive direction which would seem more natural to the user when moving the knob counterclockwise.

But if you were only looking at the screen, clockwise would increase the Z value.

mperdue, 2016-09-18:
Yes, that is what I would expect, the numbers increase as the knob is turned clockwise. I don't want a new feature unless others are having this same issue. If no one else is then it's probably a configuration error on my part.

mperdue, 2016-09-18:
If it makes a difference, I'm using a Viki 2 on an Azteeg X3 Pro.

thinkyhead, 2016-09-19:
I still cannot account for the discrepancy. The encoder direction applies exactly the same way to value editing as it does to axis movement. They both use the same method to modify the numeric value. I'll have to add some logging to see what's going on in your case. I'll let you know when there's a branch you can test.

thinkyhead, 2016-09-19:
Download this branch and test the encoder while connected to a host.

https://github.com/thinkyhead/Marlin/tree/screwy_encoder

It will print out the encoder values as you do either axis moves or change editable values on-screen. Do a bunch of clockwise rotations to start, then do a bunch of counter-clockwise rotations. Report the results and/or paste the output here.

mperdue, 2016-09-19:
When on the information screen turning the encoder wheel results in no output.
When on the menu turning the encoder wheel results in no output.

In control motion menu setting Z Offset Encoder Wheel Clockwise

Encoder Value:1886 Edit:-01.14 Min:-2000 Scale:100
Encoder Value:1886 Edit:-01.14 Min:-2000 Scale:100
Encoder Value:1886 Edit:-01.14 Min:-2000 Scale:100
Encoder Value:1886 Edit:-01.14 Min:-2000 Scale:100
Encoder Value:1886 Edit:-01.14 Min:-2000 Scale:100
Encoder Value:1886 Edit:-01.14 Min:-2000 Scale:100
Encoder Value:1885 Edit:-01.15 Min:-2000 Scale:100
Encoder Value:1885 Edit:-01.15 Min:-2000 Scale:100
Encoder Value:1885 Edit:-01.15 Min:-2000 Scale:100
Encoder Value:1885 Edit:-01.15 Min:-2000 Scale:100
Encoder Value:1885 Edit:-01.15 Min:-2000 Scale:100
Encoder Value:1885 Edit:-01.15 Min:-2000 Scale:100
Encoder Value:1885 Edit:-01.15 Min:-2000 Scale:100
Encoder Value:1885 Edit:-01.15 Min:-2000 Scale:100

In control motion menu setting Z offset Encoder Wheel Counterclockwise

Encoder Value:1917 Edit:-00.83 Min:-2000 Scale:100
Encoder Value:1917 Edit:-00.83 Min:-2000 Scale:100
Encoder Value:1917 Edit:-00.83 Min:-2000 Scale:100
Encoder Value:1917 Edit:-00.83 Min:-2000 Scale:100
Encoder Value:1917 Edit:-00.83 Min:-2000 Scale:100
Encoder Value:1917 Edit:-00.83 Min:-2000 Scale:100
Encoder Value:1918 Edit:-00.82 Min:-2000 Scale:100
Encoder Value:1918 Edit:-00.82 Min:-2000 Scale:100
Encoder Value:1918 Edit:-00.82 Min:-2000 Scale:100
Encoder Value:1918 Edit:-00.82 Min:-2000 Scale:100
Encoder Value:1918 Edit:-00.82 Min:-2000 Scale:100
Encoder Value:1918 Edit:-00.82 Min:-2000 Scale:100
Encoder Value:1918 Edit:-00.82 Min:-2000 Scale:100
Encoder Value:1918 Edit:-00.82 Min:-2000 Scale:100

In prepare, move axis, X Axis, Encoder Wheel Clockwise

Encoder Value:4294967295 Axis:0.00
Encoder Value:4294967295 Axis:0.00
Encoder Value:4294967295 Axis:0.00
Encoder Value:4294967295 Axis:0.00
Encoder Value:4294967295 Axis:0.00
Encoder Value:4294967295 Axis:0.00
Encoder Value:4294967295 Axis:0.00
Encoder Value:4294967295 Axis:0.00
Encoder Value:4294967295 Axis:0.00
Encoder Value:4294967295 Axis:0.00
Encoder Value:4294967295 Axis:0.00
Encoder Value:4294967295 Axis:0.00
Encoder Value:4294967295 Axis:0.00
Encoder Value:4294967295 Axis:0.00

In prepare, move axis,  X Axis,  Encoder Wheel Counterclockwise

Encoder Value:1 Axis:17.00
Encoder Value:1 Axis:18.00
Encoder Value:2 Axis:20.00
Encoder Value:1 Axis:21.00
Encoder Value:2 Axis:23.00
Encoder Value:2 Axis:25.00
Encoder Value:1 Axis:26.00
Encoder Value:1 Axis:27.00
Encoder Value:1 Axis:28.00
Encoder Value:2 Axis:30.00
Encoder Value:1 Axis:31.00
Encoder Value:1 Axis:32.00
Encoder Value:1 Axis:33.00
Encoder Value:1 Axis:34.00

thinkyhead, 2016-09-20:
> When on the information screen turning the encoder wheel results in no output.
When on the menu turning the encoder wheel results in no output.

Right.

### "…as you do either axis moves or change editable values…"

====

What exactly were your settings for `REVERSE_MENU_DIRECTION` and `REVERSE_ENCODER_DIRECTION` when you did this testing? Did you test with any other combinations besides the one?

thinkyhead, 2016-09-20:
So it looks like you've just found **a bug in value editing**! If value editing wasn't broken, you would be able to enable `REVERSE_MENU_DIRECTION` and everything would be perfect. So let me see if I can figure out how to fix the bug and we'll get you on your way.

thinkyhead, 2016-09-20:
At first I thought it looke wrong. But on a closer look, actually value editing is correct. If you simply enable `REVERSE_MENU_DIRECTION` and disable `REVERSE_MENU_DIRECTION` then all will be perfect.

To sum up the log output:
- Value Edit (CW) : Count goes down while value decreases (1885 < 1886 and -1.15 < -1.14)
- Value Edit (CCW) : Count goes up while value increases (1918 > 1917 and -0.82 > -0.83)
- Move Axis (CW): Count is "negative" while axis decreases
- Move Axis (CCW): Count is positive while axis increases

Thus only the encoder count is reversed. CW is decreasing the count in both cases. The count is applying correctly when value-editing, and also with axis-movement.

mperdue, 2016-09-20:
That seems to do the same thing with respect to what I've been seeing but now I have more information. It's really only one value that I find working backwards. In control, motion the Z Offset value decreases when you turn the encoder clockwise. This has the logical effect of moving the extruder nozzle up relative to the probe, but the numbers decrease. It seems backwards to me but I can live with it. Ideally, all the numbers should increase on screen with a clockwise motion.

closed 2016-09-20 by thinkyhead