MarlinFirmware/Marlin #4829: Dual extruder with different gear

paltoni, 2016-09-18:
Hi!
I have a printer with dual extruder but different gear and it not possible to change the step per unit separate for the E0 and the E1 in the firmware.Can you help solve this problem?

Jinxycob, 2016-09-18:
ello,
  I've started messing around dual extruders, what  I discovered so far  is that, there only one Estep value for all extruders, so a possible couple solutions is to a)have identical extruders and b) use the flow modifier in your slicer to increase/decease flow values for the second extruder,,,
now am no expert so hoping am wrong  about the E steps!!!

paltoni, 2016-09-18:
i found a firmware what can handle it, the name of this : Marlin kimbra

thinkyhead, 2016-09-18:
Yes, we're well aware of MagoKimbra's extra steps-per-unit entries. We haven't taken that road, but may in the future.

thinkyhead, 2016-09-18:
Yes, we know about MagoKimbra's extra steps-per-unit entries.
We haven't taken that road, but may in the future. It would be good to have this added.

The only solution in Marlin currently is to set a different flow multiplier for the second extruder. For example, if the steps per unit for E0 is 160 and the E1 needs 200, then you would use `M221 T1 S125` to set the flow rate to 125% for the second extruder.

thinkyhead, 2016-09-23:
#4849 coming soon…

boelle, 2016-11-02:
since this is a dublicate why not close?

thinkyhead, 2016-11-03:
Once #4849 is ready and gets merged it will be nice to have @paltoni give it a test run, so let's leave this open for now. It will be at least a few days before I have a chance to finish #4849 (which is incomplete), and with recent options for special types of extruders it's a little more complicated than it used to be, and will require being more meticulous.

thinkyhead, 2016-12-08:
#5371 has been merged.

closed 2016-12-08 by thinkyhead