MarlinFirmware/Marlin #4830: AUTO BED LEVELING FEATURE crashes to the bed with latest RCBugFix

renzaaa, 2016-09-18:
After G29 start print and nozzle crashes to the bed.

thinkyhead, 2016-09-18:
Do the coordinates appear to be wrong at the end of G29, before starting the print?

Could you do a G29 with `DEBUG_LEVELING_FEATURE` enabled (after `M111 S255` or at least `M111 S32`), and post the log output?

Also, I have a set of changes pending, which includes some potential bug fixes, so you might want to test with that branch rather than just `RCBugFix`. I'd do more testing myself, but I don't have a 3D printer handy at this time.

https://github.com/thinkyhead/Marlin/tree/rc_marlin_reforms

Jartza, 2016-09-18:
I think I've ran into same issue (RCBugFix, last commit "d41eeb62af4bfff0094a6c293761dfbfc3bf0f75"). 

I'm using BLTouch with safe leveling, no other endstops.

I started by clearing EEPROM (M502) after firmware update.

After I home with G28, level bed with G29, I set nozzle over bed with G1 X100 Y100 Z0

I then measure nozzle offset from bed to be 1.2mm.

Then I set the offset with M851 Z-1.2 and store it with M500

After that, G28 and G29 again. Everything looks good... except after G1 X100 Y100 Z0 the nozzle crashes to bed.

If I set the offset to -0.6 (half of the value) I get approximately correct offset. Could it be that the offset is applied twice?

mgroene, 2016-09-18:
> Could it be that the offset is applied twice?

Just testing the RCbugfix, and came to that thought, too. After G28 everything nozzle just touches the bed as expected, after the following g29, nozzle would crash into bed.

thinkyhead, 2016-09-18:
Anyone feel like providing a log to help debug the issue?

thinkyhead, 2016-09-18:
> I started by clearing EEPROM (M502) after firmware update.

M502 doesn't clear the EEPROM.

Jartza, 2016-09-18:
> M502 doesn't clear the EEPROM.

True. And usually I start with M502 + M500 which should take values from firmware and store them to EEPROM, unless I'm mistaken? Bad wording.

I'll try to get the log out in few minutes.

Jartza, 2016-09-18:
Here's the Debuglog

`Attempting connection at /dev/cu.usbserial-AH03EKZU...
READ: start
  Connected to machine!
READ: echo:Marlin 1.1.0-RCBugFix
Marlin 1.1.0-RCBugFix
READ: 
READ: echo: Last Updated: 2016-07-26 12:00 | Author: Jartza
Last Updated: 2016-07-26 12:00 | Author: Jartza
READ: Compiled: Sep 18 2016
READ: echo: Free Memory: 3222  PlannerBufferBytes: 1232
Free Memory: 3222  PlannerBufferBytes: 1232
READ: echo:Hardcoded Default Settings Loaded
Hardcoded Default Settings Loaded
READ: echo:Steps per unit:
Steps per unit:
READ: echo:  M92 X80.00 Y80.00 Z400.00 E97.00
M92 X80.00 Y80.00 Z400.00 E97.00
READ: echo:Maximum feedrates (mm/s):
Maximum feedrates (mm/s):
READ: echo:  M203 X300.00 Y300.00 Z12.00 E50.00
M203 X300.00 Y300.00 Z12.00 E50.00
READ: echo:Maximum Acceleration (mm/s2):
Maximum Acceleration (mm/s2):
READ: echo:  M201 X1500 Y1500 Z100 E10000
M201 X1500 Y1500 Z100 E10000
READ: echo:Accelerations: P=printing, R=retract and T=travel
Accelerations: P=printing, R=retract and T=travel
READ: echo:  M204 P900.00 R3000.00 T900.00
M204 P900.00 R3000.00 T900.00
READ: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
READ: echo:  M205 S0.00 T0.00 B20000 X14.00 Z0.40 E5.00
M205 S0.00 T0.00 B20000 X14.00 Z0.40 E5.00
READ: echo:Home offset (mm)
Home offset (mm)
READ: echo:  M206 X0.00 Y0.00 Z0.00
M206 X0.00 Y0.00 Z0.00
READ: echo:Material heatup parameters:
Material heatup parameters:
READ: echo:  M145 S0 H195 B55 F0
M145 S0 H195 B55 F0
READ: echo:  M145 S1 H240 B75 F0
M145 S1 H240 B75 F0
READ: echo:PID settings:
PID settings:
READ: echo:  M301 P16.10 I0.76 D85.73
M301 P16.10 I0.76 D85.73
READ: echo:  M304 P450.06 I63.88 D792.67
M304 P450.06 I63.88 D792.67
READ: echo:Filament settings: Disabled
READ: echo:  M200 D3.00
READ: echo:  M200 D0
READ: echo:Z-Probe Offset (mm):
READ: echo:  M851 Z0.00
SENT: T0
SENT: M105
SENT: M105
READ: echo:SD init fail
READ: echo:Active Extruder: 0
READ: ok
READ: ok T:21.5 /0.0 B:21.7 /0.0 @:0 B@:0
READ: ok T:21.5 /0.0 B:21.7 /0.0 @:0 B@:0
SENT: M111 S255
READ: echo:DEBUG:ECHO,INFO,ERRORS,DRYRUN,COMMUNICATION,LEVELING
DEBUG:ECHO,INFO,ERRORS,DRYRUN,COMMUNICATION,LEVELING
READ: ok
SENT: G28
READ: echo:G28
[... 298 lines cut ...]
READ: do_probe_raise(10.00)
READ: >>> do_blocking_move_to(122.00, 190.00, 10.00)
READ: <<< probe_pt
READ: current_position=(122.00, 190.00, 10.00) : set_probe_deployed
READ: deploy: 0
READ: do_probe_raise(15.00)
READ: >>> do_blocking_move_to(122.00, 190.00, 15.00)
READ: >>> do_blocking_move_to(122.00, 190.00, 15.00)
READ: current_position=(122.00, 190.00, 15.00) : clean_up_after_endstop_or_probe_move
READ: current_position=(122.00, 190.00, 15.00) : > probing complete
READ: Eqn coefficients: a: -0.00199306 b: 0.00020312 d: 0.14427092
READ: 
READ: 
READ: Bed Level Correction Matrix:
READ: +0.999998 +0.000000 -0.001993 
READ: +0.000000 +1.000000 +0.000203 
READ: +0.001993 -0.000203 +0.999998 
READ: current_position=(122.00, 190.00, 15.00) : G29 uncorrected XYZ
READ: Z from Probe:15.00  Matrix:15.03  Discrepancy:-0.03
READ: current_position=(122.00, 190.00, 15.00) : sync_plan_position
READ: current_position=(122.00, 190.00, 15.00) : G29 corrected XYZ
READ: <<< gcode_G29
READ: X:122.00 Y:190.00 Z:15.00 E:0.00 Count X: 9762 Y:15200 Z:5989
READ: ok
`

Jartza, 2016-09-18:
The debug-log above had M851 set to 0.00, but if needed, I can get another log where Z-offset has been set to something else, if needed?

thinkyhead, 2016-09-18:
If `M851` is not set to "the distance from the nozzle to the bed when the probe triggers" then the results may not be very useful.

Jartza, 2016-09-18:
Okay. I'll set it to -1.20mm which was the original measured distance and get new log in a minute.

thinkyhead, 2016-09-18:
It will help if you do `G29 V4` to print out the bed matrix.

renzaaa, 2016-09-18:
@thinkyhead Coordinates is correct at the end of G29, but something wrong with offset. 
Before the latest rcbugfix was everything fine.

thinkyhead, 2016-09-18:
The probe offset isn't actually used by `G29`. It's only used when doing `G28` homing with a probe to determine the current Z.

renzaaa, 2016-09-18:
I don`t know whats wrong with this fix, but before latest fix is all good.

thinkyhead, 2016-09-18:
I don't know either.

Jartza, 2016-09-18:
start
echo:Marlin 1.1.0-RCBugFix

echo: Last Updated: 2016-07-26 12:00 | Author: Jartza
Compiled: Sep 18 2016
echo: Free Memory: 3222  PlannerBufferBytes: 1232
echo:V25 stored settings retrieved (394 bytes)
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z400.00 E97.00
echo:Maximum feedrates (mm/s):
echo:  M203 X300.00 Y300.00 Z12.00 E50.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X1500 Y1500 Z100 E10000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P900.00 R3000.00 T900.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X14.00 Z0.40 E5.00
echo:Home offset (mm)
echo:  M206 X0.00 Y0.00 Z0.00
echo:Material heatup parameters:
echo:  M145 S0 H195 B55 F0
echo:  M145 S1 H240 B75 F0
echo:PID settings:
echo:  M301 P16.10 I0.76 D85.73
echo:  M304 P450.06 I63.88 D792.67
echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:Z-Probe Offset (mm):
echo:  M851 Z-1.20
echo:SD init fail
echo:DEBUG:ECHO,INFO,ERRORS,DRYRUN,COMMUNICATION,LEVELING
ok
echo:G28
>>> gcode_G28
Machine Type: Cartesian
Probe: SERVO PROBE
Probe Offset X:38 Y:-10 Z:-1.20 (Right-Front & Below Nozzle)
reset_bed_level
current_position=(0.00, 0.00, 0.00) : setup_for_endstop_or_probe_move
> endstops.enable(true)
Raise Z (before homing) to 10.00
>>> do_blocking_move_to(0.00, 0.00, 10.00)
>>> homeaxis(X)
current_position=(0.00, 0.00, 10.00) : sync_plan_position
> 1st Home -300.00
current_position=(0.00, 0.00, 10.00) : sync_plan_position
current_position=(0.00, 0.00, 10.00) : sync_plan_position
> 2nd Home -10.00
>>> set_axis_is_at_home(X)
For X axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = 0.00
 soft_endstop_max = 200.00
> home_offset[X] = 0.00
current_position=(1.00, 0.00, 10.00) : 
<<< set_axis_is_at_home(X)
current_position=(1.00, 0.00, 10.00) : sync_plan_position
current_position=(1.00, 0.00, 10.00) : > AFTER set_axis_is_at_home
<<< homeaxis(X)
current_position=(1.00, 0.00, 10.00) : > homeX
>>> homeaxis(Y)
current_position=(1.00, 0.00, 10.00) : sync_plan_position
> 1st Home -315.00
current_position=(1.00, 0.00, 10.00) : sync_plan_position
current_position=(1.00, 0.00, 10.00) : sync_plan_position
> 2nd Home -10.00
>>> set_axis_is_at_home(Y)
For Y axis:
[... 289 lines cut ...]
 T |           | H
   |    (-)    | T
   |           |
   O-- FRONT --+
 (0,0)
 +0.05361 +0.00611 -0.21389
 +0.21861 +0.09611 -0.06389
 +0.01111 +0.09361 -0.20139

Corrected Bed Height vs. Bed Topology:
 +0.06167 +0.14125 +0.04833
 +0.21708 +0.22167 +0.18875
 +0.00000 +0.20958 +0.04167

Bed Level Correction Matrix:
+0.999998 +0.000000 -0.002118 
-0.000000 +1.000000 -0.000120 
+0.002118 +0.000120 +0.999998 
current_position=(122.00, 190.00, 15.00) : G29 uncorrected XYZ
Z from Probe:16.20  Matrix:15.06  Discrepancy:1.14
current_position=(122.00, 190.00, 16.20) : sync_plan_position
current_position=(122.00, 190.00, 16.20) : G29 corrected XYZ
<<< gcode_G29
X:122.00 Y:190.00 Z:16.20 E:0.00 Count X: 9763 Y:15200 Z:6457
ok

thinkyhead, 2016-09-18:
@Jartza Your probing seems very erratic, with offsets all over the place. Please run an `M48` to see how reliable your probe is. Note that planar bed leveling will not produce good results if the probe is inaccurate or the bed is uneven. It only works well with a flat bed and an accurate probe.

Jartza, 2016-09-18:
Well, I had to remove the glass because there was buildtak attached to it, which now is pretty much scratched because I was too slow to hit the panic reset, and the nozzle started scraping the buildtak off 😢 

8 of 8   z: 15.000000 mean: 1.225312   sigma: 0.023599
Mean: 1.225312
Standard Deviation: 0.023599

But the same happens when I put clean glass on, still the nozzle hits bed.

Jartza, 2016-09-18:
Same log but now with 4mm glass on bed:

start
echo:Marlin 1.1.0-RCBugFix

echo: Last Updated: 2016-07-26 12:00 | Author: Jartza
Compiled: Sep 18 2016
echo: Free Memory: 3222  PlannerBufferBytes: 1232
echo:V25 stored settings retrieved (394 bytes)
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z400.00 E97.00
echo:Maximum feedrates (mm/s):
echo:  M203 X300.00 Y300.00 Z12.00 E50.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X1500 Y1500 Z100 E10000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P900.00 R3000.00 T900.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X14.00 Z0.40 E5.00
echo:Home offset (mm)
echo:  M206 X0.00 Y0.00 Z0.00
echo:Material heatup parameters:
echo:  M145 S0 H195 B55 F0
echo:  M145 S1 H240 B75 F0
echo:PID settings:
echo:  M301 P16.10 I0.76 D85.73
echo:  M304 P450.06 I63.88 D792.67
echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:Z-Probe Offset (mm):
echo:  M851 Z-1.20
echo:SD init fail
echo:DEBUG:ECHO,INFO,ERRORS,DRYRUN,COMMUNICATION,LEVELING
ok
echo:G28
>>> gcode_G28
Machine Type: Cartesian
Probe: SERVO PROBE
Probe Offset X:38 Y:-10 Z:-1.20 (Right-Front & Below Nozzle)
reset_bed_level
current_position=(0.00, 0.00, 0.00) : setup_for_endstop_or_probe_move
> endstops.enable(true)
Raise Z (before homing) to 10.00
>>> do_blocking_move_to(0.00, 0.00, 10.00)
>>> homeaxis(X)
current_position=(0.00, 0.00, 10.00) : sync_plan_position
> 1st Home -300.00
current_position=(0.00, 0.00, 10.00) : sync_plan_position
current_position=(0.00, 0.00, 10.00) : sync_plan_position
> 2nd Home -10.00
>>> set_axis_is_at_home(X)
For X axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = 0.00
 soft_endstop_max = 200.00
> home_offset[X] = 0.00
current_position=(1.00, 0.00, 10.00) : 
<<< set_axis_is_at_home(X)
current_position=(1.00, 0.00, 10.00) : sync_plan_position
current_position=(1.00, 0.00, 10.00) : > AFTER set_axis_is_at_home
<<< homeaxis(X)
current_position=(1.00, 0.00, 10.00) : > homeX
>>> homeaxis(Y)
current_position=(1.00, 0.00, 10.00) : sync_plan_position
> 1st Home -315.00
current_position=(1.00, 0.00, 10.00) : sync_plan_position
current_position=(1.00, 0.00, 10.00) : sync_plan_position
> 2nd Home -10.00
[... 291 lines cut ...]
 T |           | H
   |    (-)    | T
   |           |
   O-- FRONT --+
 (0,0)
 +0.16833 +0.04083 -0.15417
 +0.10583 +0.03833 -0.10667
 +0.01083 -0.02917 -0.07417

Corrected Bed Height vs. Bed Topology:
 +0.11583 +0.09167 +0.00000
 +0.07792 +0.11375 +0.07208
 +0.00750 +0.07083 +0.12917

Bed Level Correction Matrix:
+0.999999 +0.000000 -0.001722 
+0.000001 +1.000000 +0.000307 
+0.001722 -0.000307 +0.999998 
current_position=(122.00, 190.00, 15.00) : G29 uncorrected XYZ
Z from Probe:16.20  Matrix:15.01  Discrepancy:1.19
current_position=(122.00, 190.00, 16.20) : sync_plan_position
current_position=(122.00, 190.00, 16.20) : G29 corrected XYZ
<<< gcode_G29
X:122.00 Y:190.00 Z:16.20 E:0.00 Count X: 9762 Y:15200 Z:6475
ok

thinkyhead, 2016-09-18:
Hmm, well the probe seems accurate enough. I'm re-examining the way I'm doing the final Z correction, because this seems quite wrong…
```
Z from Probe:16.20  Matrix:15.06  Discrepancy:1.14
```
The "Z from Probe" value is supposed to determine where Z is based on the final probe point. It should be close to 15, but clearly here the probe offset is being included. Possibly a mistake in the code on my part. Checking into it now…

renzaaa, 2016-09-18:
Tomorrow I will post the log output also, if the problem is not resolved.

renzaaa, 2016-09-18:
*if the problem will not resolved :)

thinkyhead, 2016-09-18:
Aha. I found it. Typo.

thinkyhead, 2016-09-18:
Here's the fix:

```diff
-         float measured_z = probe_pt(xProbe, yProbe, stow_probe_after_each, verbose_level);
+         measured_z = probe_pt(xProbe, yProbe, stow_probe_after_each, verbose_level);
```

thinkyhead, 2016-09-18:
Okay, I patched this in `RCBugFix`. Time for more coffee!

Jartza, 2016-09-18:
I will test.

renzaaa, 2016-09-18:
float measured_z = probe_pt(xProbe, yProbe, stow_probe_after_each, verbose_level);
This code is on before latest rcbugfix and work fine.

thinkyhead, 2016-09-18:
@renzaaa With a recent patch the code declares `measured_z` earlier and needs to retain the value until probing is complete. That line conceals the original declaration and creates a temporary variable that is thrown away at the end of the block. Not good.

renzaaa, 2016-09-18:
In the morning I will test.

Jartza, 2016-09-18:
Confirmed, works now!

thinkyhead, 2016-09-18:
Excellent. Thanks @Jartza for helping to track it down. This new Z correction at the end of probing will actually fix the Z position even in cases where the `M851` offset is wrong.

closed 2016-09-18 by thinkyhead

gcormier, 2016-09-19:
Is this in RCBugFix now?

renzaaa, 2016-09-19:
Yes

renzaaa, 2016-09-19:
Latest rcbugfix is working fine.

PabloSica, 2018-03-03:
Hello. I have a similar problem. This must have already been resolved.
When I command G29 the sensor levels the first point and the second point, but in the third point, the nozzle presses the table without stopping. I have to press e-stop every time.
I'm using AUTO_BED_LEVELING_BILINEAR and CONFIGURATION_H_VERSION 010107

thinkyhead, 2018-03-04:
@PabloSica — Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If you still see the bad behavior we should investigate further.

PabloSica, 2018-03-04:
I tested this version last night and the problem persists. I'm also using the Octoprint.
How can I generate a log to send you?
I have not found bugfix-2.0.x in your links.

thinkyhead, 2018-03-04:
> I tested this version last night and the problem persists.

Please post a new issue. This issue is closed.