MarlinFirmware/Marlin #4831: Changed homing behavior/endstop requirements - Issue when using not endstops at all

Ianmcmill, 2016-09-18:
Hello,
I am currently trying to use **1.1.0RCBugfix (from RC7)** and I'm having issues using an inductive probe with auto-bed leveling but without XY endstops. I can't even get to the actual probing steps.

My prusa i3 has no endstops for XYZ. In RC5 I manually moved the head to 0,0 and always parked the head at Z 10. After powering RAMPS I entered G92 X0Y0Z10 and started printing. This worked for three years without problems.

I recently got an inductive probe and installed it and upgraded from RC5 to RC7 because I thought newer is always better.
I took the old configuration.h from RC5 and applied the proven values over to RC7 considering depreciated commands.
Compiles fine but without hardware endstops and a homing cycle (G28) I cannot use G29.
Software endstops and the probe trigger correctly when testing with M119. The inductive probe triggers correctly when testing with a piece of aluminum.

### Problem 1:
After powering Ramps and connecting via Repetier host I enter this to test movement:
G1Z10
It moves the head to Z10 but after that no G1 command works.
If I send G0 Z0 it moves to Z0 but after that no G0 commend works.
M119 says all _min and _probe endstops are open.
Only if I send G92 X0Y0Z0, the G1/0 commands are unlocked and work but then only work one single time like described above and I have to send G92X0Y0Z0 again.

Is this intended? If so, why?
I don't get it...

### 2nd issue:
```
// If you want to enable the Z probe pin, but disable its use, uncomment the line below.
// This only affects a Z probe endstop if you have separate Z min endstop as well and have
// activated Z_MIN_PROBE_ENDSTOP below. If you are using the Z Min endstop on your Z probe,
// this has no effect.
#define DISABLE_Z_MIN_PROBE_ENDSTOP
```

I don't get this either? What does this mean? Why should I want to enable z probe pin but disable its use?

This is the endstop and probe configuration:

```
//===========================================================================
//============================== Endstop Settings ===========================
//===========================================================================

// @section homing

// Specify here all the endstop connectors that are connected to any endstop or probe.
// Almost all printers will be using one per axis. Probes will use one or more of the
// extra connectors. Leave undefined any used for non-endstop and non-probe purposes.
#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG
//#define USE_XMAX_PLUG
//#define USE_YMAX_PLUG
//#define USE_ZMAX_PLUG

// coarse Endstop Settings
//#define ENDSTOPPULLUPS // Comment this out (using // at the start of the line) to disable the endstop pullup resistors

#if DISABLED(ENDSTOPPULLUPS)
  // fine endstop settings: Individual pullups. will be ignored if ENDSTOPPULLUPS is defined
  //#define ENDSTOPPULLUP_XMAX
  //#define ENDSTOPPULLUP_YMAX
  //#define ENDSTOPPULLUP_ZMAX
  //#define ENDSTOPPULLUP_XMIN
  //#define ENDSTOPPULLUP_YMIN
  //#define ENDSTOPPULLUP_ZMIN
  //#define ENDSTOPPULLUP_ZMIN_PROBE
#endif

// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
#define X_MIN_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define X_MAX_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
[... 167 lines cut ...]

// @section homing

//#define Z_HOMING_HEIGHT 4  // (in mm) Minimal z height before homing (G28) for Z clearance above the bed, clamps, ...
                             // Be sure you have this distance over your Z_MAX_POS in case.

// ENDSTOP SETTINGS:
// Sets direction of endstops when homing; 1=MAX, -1=MIN
// :[-1,1]
#define X_HOME_DIR -1
#define Y_HOME_DIR -1
#define Z_HOME_DIR -1

#define min_software_endstops false // If true, axis won't move to coordinates less than HOME_POS.
#define max_software_endstops false  // If true, axis won't move to coordinates greater than the defined lengths below.

// @section machine

// Travel limits after homing (units are in mm)
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 205
#define Y_MAX_POS 205
```

Ianmcmill, 2016-09-18:
Sorry. After some restarts the G0/1 commands seem to work now. Strange. 

Anyways. How can I perfom a G29 without a G28 before? Do I need to work with G30 instead?

thinkyhead, 2016-09-18:
You can't. `G29` does movements to specific points on the bed, and therefore needs to know where the carriage is located. And that's what `G28` establishes.

Ianmcmill, 2016-09-18:
So Z probe without XY endstop is not possible?

thinkyhead, 2016-09-18:
How do you establish XY on your machine without endstops?

Ianmcmill, 2016-09-18:
Manually positioning at (roughly) 0,0. I don't need to have the exact 0,0. Reseting or G92 X0 Y0 the ramps. Worked for three years now.

thinkyhead, 2016-09-18:
> `DISABLE_Z_MIN_PROBE_ENDSTOP`… I don't get this either? What does this mean? Why should I want to enable z probe pin but disable its use?

This setting is confusing to me too. It's probably going to go away.

thinkyhead, 2016-09-18:
> Worked for three years now.

Marlin now tracks whether it has been homed with `G28` as a safety feature. It also requires re-homing if the steppers go to sleep. If you need support for a machine with no endstops, it will need to be added as an optional feature. For the time-being, just add the following to `setup()` in your personal copy:
```cpp
LOOP_XYZ(i) axis_known_position[i] = axis_homed[i] = true;
```

Roxy-3D, 2016-09-18:
>> DISABLE_Z_MIN_PROBE_ENDSTOP… I don't get this either? What does this mean? Why should I want to enable z probe pin but disable its use?
> This setting is confusing to me too. It's probably going to go away.

Can we make it go away prior to establishing an RC-8 base line?  The reason I ask is if this is going to go away, it might be good to have it gone for a little while before we promote RC-8 to be 'Stable'.

Ianmcmill, 2016-09-18:
Thank you thank you thank you thinkyhead. This removes the G28 problem and deals with it just the way I like it.

However I had to change the configuration.h posted in my inital post to the following. I do not understand the difference from these two settings but my gut told me change these:

from:
#define Z_MIN_PROBE_ENDSTOP
to:
//#define Z_MIN_PROBE_ENDSTOP

from:
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
to:
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

I also did change:
#define Z_PROBE_DEPLOY_HEIGHT 2 // Z position for the probe to deploy/stow
#define Z_PROBE_TRAVEL_HEIGHT  5 // Z position for travel between points
 
to 2 and 5 from 15 and 5 for speed.

thinkyhead, 2016-09-19:
> Can we make it go away prior to establishing an RC-8 base line?

I think so. Its intent was to un-define the `Z_MIN_PROBE_PIN` when you also have `Z_MIN_PROBE_ENDSTOP` enabled. But when I enable `DISABLE_Z_MIN_PROBE_ENDSTOP` with `Z_MIN_PROBE_ENDSTOP` there's a sanity-check telling me that `Z_MIN_PROBE_ENDSTOP` requires `Z_MIN_PROBE_PIN`. It seems like old circular weirdness to me. But I will go back through the issues to see how it was explained to me before. Maybe there's some scenario I'm totally missing.

thinkyhead, 2016-09-19:
@Roxy-3D #4842

thinkyhead, 2016-09-20:
@Ianmcmill Are you sure the `M851` setting is also zero? And… Surely you don't want a probe Z offset of zero…?

Also, I'm not sure how good the 3-point leveling is, in general. Or, right at the moment.

Are you also using `Z_SAFE_HOMING`? (It's meant for when you use the probe to home Z in `G28`, but it can also be used with a regular Z min endstop if that's preferred.)

thinkyhead, 2016-09-20:
The easiest way to see what's happening is to enable `DEBUG_LEVELING_FEATURE` and issue `M111 S247` before you do `G28` and `G29`. From the detailed log output it should be a simple matter to uncover the issue.

Ianmcmill, 2017-05-30:
Grave digging here... sorry.
does this still work in 1.1.1?:
`LOOP_XYZ(i) axis_known_position[i] = axis_homed[i] = true;`

closed 2018-02-06 by thinkyhead