MarlinFirmware/Marlin #4838: Improve bltouch configuration?

gcormier, 2016-09-18:
Bltouch servo number is not visible in configuration.h. Not sure if it makes sense to add that right below the define since it's not that far a stretch to imagine not having it on port 0. 

I can try my hand at a pull if this is something useful?

thinkyhead, 2016-09-18:
@gcormier Thanks for the suggestion. Rather than adding a new option we'll just throw an error if the servo isn't set, and remove the override that always sets it to 0.

thinkyhead, 2016-09-19:
Merged! I also added support to handle the BLTouch error condition. Probing will abort if the probe errors while being deployed (though not once probing starts, so maybe that needs to be added). And a menu item appears to reset the probe if it's in an error state.

closed 2016-09-19 by thinkyhead