MarlinFirmware/Marlin #4843: compile failure, 'abl2' was not declared in this scope

VanessaE, 2016-09-19:
```
Arduino: 1.6.11 (Linux), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

/home/vanessa/RepRap/arduino-1.6.11/arduino-builder -dump-prefs -logger=machine -hardware /home/vanessa/RepRap/arduino-1.6.11/hardware -tools /home/vanessa/RepRap/arduino-1.6.11/tools-builder -tools /home/vanessa/RepRap/arduino-1.6.11/hardware/tools/avr -built-in-libraries 

[...redundant compiler spew deleted...]

"/home/vanessa/RepRap/arduino-1.6.11/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/home/vanessa/RepRap/arduino-1.6.11/hardware/arduino/avr/cores/arduino" "-I/home/vanessa/RepRap/arduino-1.6.11/hardware/arduino/avr/variants/mega" "/tmp/builda2a2970193a4404e5d60b1ae4bfc203f.tmp/sketch/MarlinSerial.cpp" -o "/tmp/builda2a2970193a4404e5d60b1ae4bfc203f.tmp/sketch/MarlinSerial.cpp.o"
"/home/vanessa/RepRap/arduino-1.6.11/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/home/vanessa/RepRap/arduino-1.6.11/hardware/arduino/avr/cores/arduino" "-I/home/vanessa/RepRap/arduino-1.6.11/hardware/arduino/avr/variants/mega" "/tmp/builda2a2970193a4404e5d60b1ae4bfc203f.tmp/sketch/Marlin_main.cpp" -o "/tmp/builda2a2970193a4404e5d60b1ae4bfc203f.tmp/sketch/Marlin_main.cpp.o"
/tmp/builda2a2970193a4404e5d60b1ae4bfc203f.tmp/sketch/Marlin_main.cpp: In function 'void gcode_G29()':
Marlin_main.cpp:3585: error: 'abl2' was not declared in this scope
       qr_solve(plane_equation_coefficients, abl2, 3, eqnAMatrix, eqnBVector);
                                             ^
Marlin_main.cpp:3585: error: 'eqnAMatrix' was not declared in this scope
       qr_solve(plane_equation_coefficients, abl2, 3, eqnAMatrix, eqnBVector);
                                                      ^
Marlin_main.cpp:3585: error: 'eqnBVector' was not declared in this scope
       qr_solve(plane_equation_coefficients, abl2, 3, eqnAMatrix, eqnBVector);
                                                                  ^
Marlin_main.cpp:3585: error: 'qr_solve' was not declared in this scope
       qr_solve(plane_equation_coefficients, abl2, 3, eqnAMatrix, eqnBVector);
                                                                            ^
Marlin_main.cpp:3587: error: 'mean' was not declared in this scope
       mean /= abl2;
       ^
Marlin_main.cpp:3612: error: 'do_topography_map' was not declared in this scope
       if (do_topography_map) {
           ^
Marlin_main.cpp:3628: error: 'abl_grid_points_y' was not declared in this scope
         for (int8_t yy = abl_grid_points_y - 1; yy >= 0; yy--) {
                          ^
Marlin_main.cpp:3629: error: 'abl_grid_points_x' was not declared in this scope
           for (uint8_t xx = 0; xx < abl_grid_points_x; xx++) {
                                     ^
Marlin_main.cpp:3630: error: 'indexIntoAB' was not declared in this scope
             int ind = indexIntoAB[xx][yy];
                       ^
Marlin_main.cpp:3636: error: 'x_tmp' was not declared in this scope
             apply_rotation_xyz(planner.bed_level_matrix, x_tmp, y_tmp, z_tmp);
                                                          ^
Marlin_main.cpp:3636: error: 'y_tmp' was not declared in this scope
             apply_rotation_xyz(planner.bed_level_matrix, x_tmp, y_tmp, z_tmp);
                                                                 ^
Marlin_main.cpp:3636: error: 'z_tmp' was not declared in this scope
             apply_rotation_xyz(planner.bed_level_matrix, x_tmp, y_tmp, z_tmp);
                                                                        ^
Marlin_main.cpp:3653: error: 'abl_grid_points_y' was not declared in this scope
           for (int8_t yy = abl_grid_points_y - 1; yy >= 0; yy--) {
                            ^
Marlin_main.cpp:3654: error: 'abl_grid_points_x' was not declared in this scope
             for (uint8_t xx = 0; xx < abl_grid_points_x; xx++) {
                                       ^
Marlin_main.cpp:3655: error: 'indexIntoAB' was not declared in this scope
               int ind = indexIntoAB[xx][yy];
                         ^
Marlin_main.cpp:3660: error: 'y_tmp' was not declared in this scope
               apply_rotation_xyz(planner.bed_level_matrix, x_tmp, y_tmp, z_tmp);
                                                                   ^
Marlin_main.cpp:3660: error: 'z_tmp' was not declared in this scope
               apply_rotation_xyz(planner.bed_level_matrix, x_tmp, y_tmp, z_tmp);
                                                                          ^
exit status 1
'abl2' was not declared in this scope
```

```c++
/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
[... 2187 lines cut ...]
 * M155 A99  ; Target slave address
 * M155 B77  ; M
 * M155 B97  ; a
 * M155 B114 ; r
 * M155 B108 ; l
 * M155 B105 ; i
 * M155 B110 ; n
 * M155 S1   ; Send the current buffer
 *
 * ; Example #2
 * ; Request 6 bytes from slave device with address 0x63 (99)
 * M156 A99 B5
 *
 * ; Example #3
 * ; Example serial output of a M156 request
 * echo:i2c-reply: from:99 bytes:5 data:hello
 */

// @section i2cbus

//#define EXPERIMENTAL_I2CBUS
#define I2C_SLAVE_ADDRESS  0 // Set a value from 8 to 127 to act as a slave

#endif // CONFIGURATION_ADV_H
```

thinkyhead, 2016-09-20:
I believe this is now fixed. With your configurations I get the following:
```
In file included from /var/folders/z5/dqn_cqn97jx89t_xvc4pgxrc0000gn/T/build2be42bf204d0a3e60cf2b85dc6e5d099.tmp/sketch/MarlinConfig.h:39:0,
                 from /Volumes/Projects/MAKER/Firmware/Thinkyhead Forks/Marlin/Marlin/Marlin.ino:37:
SanityCheck.h:137: error: #error "Z_PROBE_DEPLOY_HEIGHT and Z_PROBE_TRAVEL_HEIGHT are now Z_CLEARANCE_DEPLOY_PROBE and Z_CLEARANCE_BETWEEN_PROBES. Please update your configuration."
   #error "Z_PROBE_DEPLOY_HEIGHT and Z_PROBE_TRAVEL_HEIGHT are now Z_CLEARANCE_DEPLOY_PROBE and Z_CLEARANCE_BETWEEN_PROBES. Please update your configuration."
    ^
SanityCheck.h:445: error: #error "You must define Z_CLEARANCE_DEPLOY_PROBE in your configuration."
     #error "You must define Z_CLEARANCE_DEPLOY_PROBE in your configuration."
      ^
exit status 1
#error "Z_PROBE_DEPLOY_HEIGHT and Z_PROBE_TRAVEL_HEIGHT are now Z_CLEARANCE_DEPLOY_PROBE and Z_CLEARANCE_BETWEEN_PROBES. Please update your configuration."
```

thinkyhead, 2016-09-20:
Thanks for the report! There's a patch pending which will be merged shortly.

VanessaE, 2016-09-20:
> "You must define Z_CLEARANCE_DEPLOY_PROBE in your configuration."

I'm not sure that applies to me, as I am using a fixed, inductive probe.  Is not the above option meant for a servo-actuated switche?

thinkyhead, 2016-09-20:
It applies to all probes, including inductive. Especially inductive. If there is no raise, the probe might be right up against the bed, and would get triggered right away.

closed 2016-09-20 by thinkyhead

VanessaE, 2016-09-20:
Ok, I think I understand.  My confusion stems from "deploy" meaning, in my mind, to swing a switch down with a servo.

thinkyhead, 2016-09-21:
Hopefully the updated language for these options will make it clearer. This will be getting merged pretty soon:
```cpp
/**
 * Z probes require clearance when deploying, stowing, and moving between
 * probe points to avoid hitting the bed and other hardware.
 * Servo-mounted probes require extra space for the arm to rotate.
 * Inductive probes need space to keep from triggering early.
 *
 * Use these settings to specify the distance (mm) to raise the probe (or
 * lower the bed). The values set here apply over and above any (negative)
 * probe Z Offset set with Z_PROBE_OFFSET_FROM_EXTRUDER, M851, or the LCD.
 * Only integer values >= 1 are valid here.
 *
 * Example: `M851 Z-5` with a RAISE of 4  =>  9mm from bed to nozzle.
 *     But: `M851 Z+1` with a RAISE of 2  =>  2mm from bed to nozzle.
 */
```

VanessaE, 2016-09-21:
Yes, that indeed makes sense now!