MarlinFirmware/Marlin #4846: [DELTA] Default Endstop Adjustments

serrainoalessio, 2016-09-19:
I added some extra macros which allows to save a default endstop adjustment within the firmware. This setting will be loaded on startup and on M502 (Reset configuration, loads the one written in the program memory). At the time M502 sets to 0 endstop adjustment.

Here is the diff:

    diff --git a/configuration_store.cpp b/configuration_store.cpp
    index db10bba..5897296 100644
    --- a/configuration_store.cpp
    +++ b/configuration_store.cpp
    @@ -607,7 +607,9 @@ void Config_ResetDefault() {
       #endif
     
       #if ENABLED(DELTA)
    -    endstop_adj[X_AXIS] = endstop_adj[Y_AXIS] = endstop_adj[Z_AXIS] = 0;
    +    endstop_adj[X_AXIS] = ENDSTOP_ADJUST_DEFAULT_X;
    +    endstop_adj[Y_AXIS] = ENDSTOP_ADJUST_DEFAULT_Y;
    +    endstop_adj[Z_AXIS] = ENDSTOP_ADJUST_DEFAULT_Z;
         delta_radius =  DELTA_RADIUS;
         delta_diagonal_rod =  DELTA_DIAGONAL_ROD;
         delta_segments_per_second =  DELTA_SEGMENTS_PER_SECOND;
    
    diff --git a/Configuration.h b/Configuration.h
    index 1091e93..e4468a8 100644
    --- a/Configuration.h
    +++ b/Configuration.h
    @@ -809,9 +809,15 @@
    //#define MANUAL_Y_HOME_POS 0
    #define MANUAL_Z_HOME_POS 250 // Distance between the nozzle to printbed after homing
    +
    +// Endstop Default Adjustment, in millimeters (floats). Can be overwritten with M666
    +// Positive Adjustment will move towars the endstop, Negative will move away from the endstops
    +#define ENDSTOP_ADJUST_DEFAULT_X -0.20
    +#define ENDSTOP_ADJUST_DEFAULT_Y -1.70
    +#define ENDSTOP_ADJUST_DEFAULT_Z +0.00
     
     // Use "Z Safe Homing" to avoid homing with a Z probe outside the bed area.
     //
    
    diff --git a/Marlin_main.cpp b/Marlin_main.cpp
    index d1c4876..7367e22 100644
    --- a/Marlin_main.cpp
    +++ b/Marlin_main.cpp
    @@ -462,8 +462,11 @@ static uint8_t target_extruder;
       float cartesian_position[3] = { 0 };
       #define SIN_60 0.8660254037844386
       #define COS_60 0.5
    -  float endstop_adj[3] = { 0 };
    +
       // these are the default values, can be overriden with M665
    +  float endstop_adj[3] = { [X_AXIS] = ENDSTOP_ADJUST_DEFAULT_X,
    +                           [Y_AXIS] = ENDSTOP_ADJUST_DEFAULT_Y,
    +                           [Z_AXIS] = ENDSTOP_ADJUST_DEFAULT_Z };
       float delta_radius = DELTA_RADIUS;
       float delta_tower1_x = -SIN_60 * (delta_radius + DELTA_RADIUS_TRIM_TOWER_1); // front left tower
       float delta_tower1_y = -COS_60 * (delta_radius + DELTA_RADIUS_TRIM_TOWER_1);

(Maybe the extra macro might be added in Configuration_adv.h)
I know it is not a good thing to set endstop adjustment, it is much better to physically move the endstops, but I glued them to the frame, to ensure they would not move when hit by the mechanical parts.
If enabled EEPROM this might be quite unuseful because on each startup settings in the EEPROM are loaded.

thinkyhead, 2016-09-20:
```diff
+  float endstop_adj[3] = { [X_AXIS] = ENDSTOP_ADJUST_DEFAULT_X,
+                           [Y_AXIS] = ENDSTOP_ADJUST_DEFAULT_Y,
+                           [Z_AXIS] = ENDSTOP_ADJUST_DEFAULT_Z };
```
I think you meant…
```diff
+  float endstop_adj[3] = { ENDSTOP_ADJUST_DEFAULT_X,
+                           ENDSTOP_ADJUST_DEFAULT_Y,
+                           ENDSTOP_ADJUST_DEFAULT_Z };
```

serrainoalessio, 2016-09-22:
At pag. 130, paragraph 33
http://www.open-std.org/jtc1/sc22/wg14/www/docs/n1256.pdf

Actually I didn't know the order in which endstop_adj values should be stored, if XYZ or ZYX, or something else. I noticed in other parts of the code they were accessed by using X, Y and Z _AXIS labels instead of an explicit numerical index, so I initialized the array that way.

thinkyhead, 2016-09-22:
Just pointing out that's not valid C++ code.

closed 2017-10-31 by thinkyhead