MarlinFirmware/Marlin #4847: Issue with nozzle height

gcormier, 2016-09-19:
Current running RCBugFix with the recent fix for G29.

I ran a repeat-ability test on my BLTouch probe and got very nice results.

1) Adjust Z height using G28 and M851 with a 0.15mm feeler until acceptable.
2) Print something (successfully)
3) Try to do another print.
4) Fail. Watch as printer tries to extrude in "mid air". Observer G28/G29 putting Z0 about 1mm above the bed surface.
5) Go to step 1. Repeat.

Value seems to fluctuate back and forth.

> 16:57:33.344 : echo:V24 stored settings retrieved (398 bytes)
> 16:57:33.344 : echo:Steps per unit:
> 16:57:33.344 : echo:  M92 X80.00 Y80.00 Z400.00 E605.00
> 16:57:33.348 : echo:Maximum feedrates (mm/s):
> 16:57:33.348 : echo:  M203 X300.00 Y300.00 Z5.00 E25.00
> 16:57:33.348 : echo:Maximum Acceleration (mm/s2):
> 16:57:33.352 : echo:  M201 X2500 Y2500 Z100 E10000
> 16:57:33.352 : echo:Accelerations: P=printing, R=retract and T=travel
> 16:57:33.356 : echo:  M204 P2000.00 R2000.00 T2000.00
> 16:57:33.364 : echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
> 16:57:33.364 : echo:  M205 S0.00 T0.00 B20000 X15.00 Z0.40 E5.00
> 16:57:33.364 : echo:Home offset (mm)
> 16:57:33.368 : echo:  M206 X0.00 Y0.00 Z0.00
> 16:57:33.368 : echo:Material heatup parameters:
> 16:57:33.368 : echo:  M145 S0 H180 B70 F0
> 16:57:33.368 : echo:  M145 S1 H240 B110 F0
> 16:57:33.372 : echo:PID settings:
> 16:57:33.372 : echo:  M301 P23.72 I1.56 D90.34
> 16:57:33.376 : echo:  M304 P150.52 I12.25 D462.40
> 16:57:33.377 : echo:Filament settings: Disabled
> 16:57:33.377 : echo:  M200 D3.00
> 16:57:33.377 : echo:  M200 T1 D3.00
> 16:57:33.377 : echo:  M200 D0
> 16:57:33.380 : echo:Z-Probe Offset (mm):
> 16:57:33.380 : echo:  M851 Z-0.30

I'm stumped.

Roxy-3D, 2016-09-19:
One good work around is to just use G28 to home the machine.   And then I always have a G29 in my startup code in the Slicer.   I just tweak my Z_PROBE_OFFSET_FROM_EXTRUDER until things are exactly right and it lays down the first layer the way I want it.    If you also enable Z_BABYSTEPPING you will get a good first layer every time.

gcormier, 2016-09-19:
Is there a reason I have to keep readjusting m851 between prints?

FHeilmann, 2016-09-20:
Do you store your z-offset using M500?

gcormier, 2016-09-20:
Yup. M851, then M500, then G28,G29 to confirm.. repeat that until happy, finish with M500, print.

Then the next print, nozzle height is completely off.

gcormier, 2016-09-20:
I just did a fresh RCBugFix build and I can't repro this now.... I'll run some prints and update this.

gcormier, 2016-09-20:
Confirmed. Just did another print, nozzle is now touch the bed at Z=0.3.

thinkyhead, 2016-09-20:
It's unclear from your description so far whether you're doing another `G29` or just jumping straight to the next print. Is there anything special about the "start gcode" in the files you're printing?

gcormier, 2016-09-21:
Nothing special. I G28/G29 before each print.

I'm going to run some tests now on heated versus non. Perhaps it's an issue of thermal expansion. Going to cook those nozzles to see if I can exaggerate the condition (or not).

thinkyhead, 2016-09-21:
A Z probe offset of only 0.3mm is astoundingly small, and the `BLTOUCH` itself wasn't very well supported so far. But the `BLTOUCH` code has been undergoing some changes today that should help improve it. No guarantee it will help with your setup. And maybe there is something about a warmer bed that could change the result.

gcormier, 2016-09-21:
Embarrassing. I think this is heat related. I homed, observed Z0 being off. Left it there and then set my nozzle temp, and left it for about 10 minutes. Came back and it was then at a nice height.

Sorry about that guys!

thinkyhead, 2016-09-21:
Thanks for the update! We'll have to make a note about the potential for heat to warp the bed in the documentation on homing and leveling.

gcormier, 2016-09-21:
That would be great! Also note this issue didn't present itself when the nozzle reached the temp - it took a bit longer for the heat soak to actually make things expand/warp.

closed 2016-09-22 by thinkyhead