MarlinFirmware/Marlin #4848: PID Autotune cancels M104/M109 on other extruder/bed

GreatGrizzly, 2016-09-19:
When I issue a M190 to heat up my bed then run PID Autotune on the Hot End, the bed temperature changes back to zero during the process, forcing me to use an additional set of commands to bring the temperature back to how it was before. Here is my start code:

;If printer is waiting for heatup (M109) or bed headup (M190), break out of the wait.
M108

;Save settings just incase I forget.
M500

;Wait for bed tempurature to get to target tempurature.
M190 R[first_layer_bed_temperature]

;Home all Axis
G28

;Move print head to back left to get it out of the way.
G0 X0 Y245 Z5 F2000

;PID AutoTune the HotEnd to desired non-first layer tempurature
M303 E0 S[temperature] C4 U1

;Save settings just incase
M500

;hot end reaches target tempurature, continue on.
M104 R[first_layer_temperature]

;Wait for bed tempurature to get to target tempurature, again...
M190 R[first_layer_bed_temperature]

FHeilmann, 2016-09-20:
Why do you run a PID autotune before every print?

Autotune is meant to be run once when you change Voltage/Heating hardware and the results committed to EEPROM. It is not necessary to re-run the autotune for each print.

GreatGrizzly, 2016-09-20:
From my understanding: PID values change depending on the target
temperature of the hot end/bed. If I have different types a filaments at
different temperatures, wouldn't it make sense to PID tune the hot end/bed
for each one? I want to introduce as much automation as possible.

My thinking is that most of the wait time before a print is for the warm up
of the hot end/bed, so doing a PID Autotune cycle is very small increase in
time for improved convenience.

Roxy-3D, 2016-09-20:
Well...  What you say is true.    But the PID algorithms are not ridiculously sensitive.   They are pretty stable.   You you load up an extruder with a Black PLA filament and run the autotune at 205 C. , you will get a number.   If you switch the nozzle to a White ABS (and purge it) and autoturn at 230 C. you will get another number.

You can usually take the middle ground and load that into your Configuration.h and be very happy with the results regardless of what you are printing with.

GreatGrizzly, 2016-09-20:
Another thing about turning on each print is over time (assuming you 
have the same filament and temperature), the PID settings will get more 
accurate on each run, as it is using the old results to get the new 
results which will probably be very similar. I have found that 
flimiments like petg are extremely sensitive to 1 degree variations, so 
taking the middle ground might not be a viable option. Especially when I 
switch between PLA and PETG. 195c vs 250c.

Roxy-3D, 2016-09-20:
> I have found that filaments like PETG are extremely sensitive to 1 degree variations, so
taking the middle ground might not be a viable option. 

For the 'very sensitive' filaments, it might be reasonable to actually extrude some filament  during the tuning process.   It could be argued how much of that needs to be done, but for sure that is going to affect what the 'optimum parameters' are.

GreatGrizzly, 2016-09-20:
That is a very good idea. Usually when I autotune, I have the filament in
the chamber.

brainscan, 2016-09-20:
I've only needed to use auto tune once and my nozzle temp is always solid regardless of which filament. What hotend do you use maybe if you're having problems it's hardware related as I've never known a filament to have a 1° operating window.

-

thinkyhead, 2016-09-20:
You'll note also that when you run filament faster through the extruder, it cools down more quickly. Same thing if your spool of filament is a few degrees cooler on a given day. The autotune is meant to provide stability for the bulk of the thermal mass in the hot-end.

You should try using the `AUTOTEMP` feature. This causes the heater to adjust itself adjust based on the volume of material passing being pushed into the chamber. It allows you to set a factor, which you can base on the kind of material you're using and the ambient temperature.

closed 2017-10-31 by thinkyhead