MarlinFirmware/Marlin #4850: Travis CI failing on u8glib download

thinkyhead, 2016-09-20:
```
Verifying archive integrity...
java.lang.Exception: CRC doesn't match. File is corrupted.
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:77)
	at cc.arduino.contributions.libraries.LibraryInstaller.install(LibraryInstaller.java:97)
	at processing.app.Base.<init>(Base.java:414)
	at processing.app.Base.guardedMain(Base.java:222)
	at processing.app.Base.main(Base.java:135)

The command "arduino --install-library "U8glib"" failed and exited with 255 during .
```

esenapaj, 2016-09-20:
I remember this...
Broken SHA hash on MarlinDev-MarlinPlatform1.6.7b.zip · Issue #3846 · MarlinFirmware/Marlin
https://github.com/MarlinFirmware/Marlin/issues/3846

thinkyhead, 2016-09-20:
Unfortunately we're using the built-in:
```
  - arduino --install-library "U8glib"
```
So possibly the issue is that the version of Arduino is just on the way to being too old, and perhaps has an outdated library URL. The lib itself downloads, but the hash –wherever it's hosted– perhaps isn't being found, or has been corrupted.

I'll see if a newer Arduino can be used…

thinkyhead, 2016-09-20:
#4857 fixes the issue, and updates to Arduino 1.6.10.

closed 2016-09-20 by thinkyhead