MarlinFirmware/Marlin #4853: BLTouch reset logic broken

Jartza, 2016-09-20:
Git commit f38a33a5d8de4c58c83d1511c4b048043ba4865c has logic for BLTouch reset, which is broken.

BLTouch is controlled like servo motor, but it never reports anything back to servo control pins (I don't know if it would even be possible due the nature how servos are controlled?).

The only output from BLTouch is the endstop. It will report triggered whenever the probe is in error situation, and "moving servo" to position 160 will clear the error and stow the pin up, if possible. This will also clear the endstop.

BLTouch error state cannot then be read from servo. But also, setting servo to 10 or 90 will not work while BLTouch is in error state. If endstop status will not get cleared after setting servo to 160, then some kind of error / warning could be issued.

Jartza, 2016-09-20:
...also, this line:

`MENU_ITEM(gcode, MSG_RESET_BLTOUCH, "M280 S90 P" STRINGIFY(Z_ENDSTOP_SERVO_NR));`

Doesn't seem to work, even if I change it to S160 (reset BLTouch) and remove the if... Looks to me like the MENU_ITEM with gcode always needs PSTR().

Jartza, 2016-09-20:
> Looks to me like the MENU_ITEM with gcode always needs PSTR().

Works when PSTR() added, and the if() removed (as the if-logic tries to read servo value, which is wrong)

thinkyhead, 2016-09-20:
Thank you for the correction. The documentation given to me was in error about the meaning of the servo values. 

I think some kind of `if()` would be helpful. Perhaps we can check to see if it is triggered and present the menu item only then. I'll take a look at the code in M119 and the code that updates probe state to see.

Jartza, 2016-09-20:
> Thank you for the correction. The documentation given to me was in error about the meaning of the servo values.

Yep, the servo values are only sent to BLTouch and BLTouch "emulates" servo in a sense, but it's write only.

Servo "positions":

10 = Push pin out and ready to measure
90 = Retract pin (in print mode keeps the pin away from print)
120 = Self test
160 = Reset error situation and retract pin

The self-test is not very useful in my opinion, it'll just push pin out, retract, push pin out, retract... ad infinity

But whenever BLTouch hits something before it's supposed to, it goes into error mode, where position 160 is useful.

>I think some kind of if() would be helpful. Perhaps we can check to see if it is triggered and present the menu item only then. I'll take a look at the code in M119 and the code that updates probe state to see.

Yes, endstop shows always constantly triggered while BLTouch is in error mode (BLTouch led blinks), and right after the error is cleared (M280 P0 S160) , the endstop status switches back to open (unless error is persistent).

Roxy-3D, 2016-09-20:
> But whenever BLTouch hits something before it's supposed to, it goes into error mode, where position 160 is useful.

Since the error status can not be read, would it make sense to always go to 160 degrees just for an instant before 'deploying' the probe at 10 degrees?  That would clear the error state if it was in that state.

Jartza, 2016-09-20:
> Since the error status can not be read, would it make sense to always go to 160 degrees just for an instant before 'deploying' the probe at 10 degrees? That would clear the error state if it was in that state.

I would maybe reset the BLTouch every time before starting the homing process. As X and Y need to be homed for safe homing with BLTouch, maybe even reset it before homing X and Y to make sure BLTouch has time to reset itself?

thinkyhead, 2016-09-20:
All should be better with #4856

Jartza, 2016-09-20:
RCBugFix doesn't compile.

`      MENU_ITEM(gcode, MSG_BLTOUCH_TEST, PSTR("M280 P" STRINGIFY(Z_ENDSTOP_SERVO_NR) " S" STRINGIFY(BLTOUCH_SELFTEST)));`

should it be MSG_BLTOUCH_SELFTEST ?

Jartza, 2016-09-20:
Also BLTouch reset doesn't really work, as BLTOUCH_RESET is not defined anywhere... should be BLTOUCH_RELEASE:

`MENU_ITEM(gcode, MSG_BLTOUCH_SELFTEST, PSTR("M280 P" STRINGIFY(Z_ENDSTOP_SERVO_NR) " S" STRINGIFY(BLTOUCH_SELFTEST)));`

and

`MENU_ITEM(gcode, MSG_BLTOUCH_RESET, PSTR("M280 P" STRINGIFY(Z_ENDSTOP_SERVO_NR) " S" STRINGIFY(BLTOUCH_RELEASE)));`

Jartza, 2016-09-20:
[bltouch.txt](https://github.com/MarlinFirmware/Marlin/files/483738/bltouch.txt)

Jartza, 2016-09-20:
Another issue... homing is now painfully slow and has strange pauses as BLTouch gets pushed out & stowed back, even when homing X and Y. Also the standard "bump" while homing has worked pretty well, not sure if there's reason to add those extra stowing functionalities?

Jartza, 2016-09-20:
Ahh. I guess the extra delay comes from #define SERVO_DELAY which was 300 in my case. Anyways there's no use pushing / pulling BLTouch pin while homing X and Y?

Jartza, 2016-09-20:
[bltouch2.txt](https://github.com/MarlinFirmware/Marlin/files/483758/bltouch2.txt)

Jartza, 2016-09-20:
For some reason, lowering the SERVO_DELAY under 80 gives erratic results. I thought the servo stayed active after move(), if I don't have DEACTIVATE_SERVOS_AFTER_MOVE defined?

thinkyhead, 2016-09-21:
Patching….

I guess 1/3 of a second is pretty long as a default delay for `BLTOUCH`. Just another thing that needs to be tweaked. The `SERVO_DELAY` always applies, and is no longer tied to just deactivation, but to every servo move, to give the servo time to move, I'm guessing.

Jartza, 2016-09-21:
> What amount of SERVO_DELAY do you find is useful and not erratic?

Seems 100 works pretty much always.

Jartza, 2016-09-21:
> Seems 100 works pretty much always. 80 seems to work 99% of time but I think 100 should be safe.

Actually, I think I need to retest this later today, looks like I messed up with my tests and I'm not sure if I actually saved the Configuration.h before testing :)

Jartza, 2016-09-21:
I tested the latest RCBugFix and the BLTouch default 50 seems to work just fine (tested bed leveling 10 times with 9 points). The only thing I noticed now, when homing (G28) , after the "bump" the BLTouch is not retracted, like when leveling with G29?

Jartza, 2016-09-21:
start
echo:Marlin 1.1.0-RCBugFix

echo: Last Updated: 2016-07-26 12:00 | Author: Jartza
Compiled: Sep 21 2016
echo: Free Memory: 3231  PlannerBufferBytes: 1232
echo:V25 stored settings retrieved (394 bytes)
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z400.00 E97.00
echo:Maximum feedrates (mm/s):
echo:  M203 X300.00 Y300.00 Z12.00 E50.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X1500 Y1500 Z100 E10000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P900.00 R3000.00 T900.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X14.00 Z0.40 E5.00
echo:Home offset (mm)
echo:  M206 X0.00 Y0.00 Z0.00
echo:Material heatup parameters:
echo:  M145 S0 H195 B55 F0
echo:  M145 S1 H240 B75 F0
echo:PID settings:
echo:  M301 P16.10 I0.76 D85.73
echo:  M304 P450.06 I63.88 D792.67
echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:Z-Probe Offset (mm):
echo:  M851 Z-1.10
echo:SD init fail
ok
ok
echo:DEBUG:ECHO,INFO,ERRORS,DRYRUN,COMMUNICATION,LEVELING
ok
echo:G28
>>> gcode_G28
Machine Type: Cartesian
Probe: SERVO PROBE
Probe Offset X:38 Y:-10 Z:-1.10 (Right-Front & Below Nozzle)
reset_bed_level
current_position=(0.00, 0.00, 0.00) : setup_for_endstop_or_probe_move
> endstops.enable(true)
Raise Z (before homing) to 10.00
>>> do_blocking_move_to(0.00, 0.00, 10.00)
>>> homeaxis(X)
current_position=(0.00, 0.00, 10.00) : sync_plan_position
> 1st Home -300.00
current_position=(0.00, 0.00, 10.00) : sync_plan_position
current_position=(0.00, 0.00, 10.00) : sync_plan_position
> 2nd Home -10.00
>>> set_axis_is_at_home(X)
For X axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = 0.00
 soft_endstop_max = 200.00
> home_offset[X] = 0.00
current_position=(1.00, 0.00, 10.00) : 
<<< set_axis_is_at_home(X)
current_position=(1.00, 0.00, 10.00) : sync_plan_position
current_position=(1.00, 0.00, 10.00) : > AFTER set_axis_is_at_home
<<< homeaxis(X)
current_position=(1.00, 0.00, 10.00) : > homeX
>>> homeaxis(Y)
current_position=(1.00, 0.00, 10.00) : sync_plan_position
> 1st Home -315.00
current_position=(1.00, 0.00, 10.00) : sync_plan_position
current_position=(1.00, 0.00, 10.00) : sync_plan_position
> 2nd Home -10.00
>>> set_axis_is_at_home(Y)
For Y axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = 0.00
 soft_endstop_max = 210.00
> home_offset[Y] = 0.00
current_position=(1.00, -19.00, 10.00) : 
<<< set_axis_is_at_home(Y)
current_position=(1.00, -19.00, 10.00) : sync_plan_position
current_position=(1.00, -19.00, 10.00) : > AFTER set_axis_is_at_home
<<< homeaxis(Y)
current_position=(1.00, -19.00, 10.00) : > homeY
Z_SAFE_HOMING >>>
current_position=(1.00, -19.00, 10.00) : sync_plan_position
destination=(62.00, 115.00, 10.00) : Z_SAFE_HOMING
>>> do_blocking_move_to(62.00, 115.00, 10.00)
>>> homeaxis(Z)
current_position=(62.00, 115.00, 10.00) : set_probe_deployed
deploy: 1
do_probe_raise(15.00)
>>> do_blocking_move_to(62.00, 115.00, 15.00)
>>> do_blocking_move_to(62.00, 115.00, 15.00)
current_position=(62.00, 115.00, 0.00) : sync_plan_position
> 1st Home -315.00
current_position=(62.00, 115.00, 0.00) : sync_plan_position
current_position=(62.00, 115.00, 0.00) : sync_plan_position
> 2nd Home -6.00
>>> set_axis_is_at_home(Z)
For Z axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = 0.00
 soft_endstop_max = 210.00
*** Z HOMED WITH PROBE (Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN) ***
> zprobe_zoffset = -1.10
> home_offset[Z] = 0.00
current_position=(62.00, 115.00, 1.10) : 
<<< set_axis_is_at_home(Z)
current_position=(62.00, 115.00, 1.10) : sync_plan_position
current_position=(62.00, 115.00, 1.10) : > AFTER set_axis_is_at_home
current_position=(62.00, 115.00, 1.10) : set_probe_deployed
deploy: 0
do_probe_raise(15.00)
>>> do_blocking_move_to(62.00, 115.00, 15.00)
>>> do_blocking_move_to(62.00, 115.00, 15.00)
<<< homeaxis(Z)
<<< Z_SAFE_HOMING
current_position=(62.00, 115.00, 15.00) : > (home_all_axis || homeZ) > final
current_position=(62.00, 115.00, 15.00) : sync_plan_position
current_position=(62.00, 115.00, 15.00) : clean_up_after_endstop_or_probe_move
<<< gcode_G28
X:62.00 Y:115.00 Z:15.00 E:0.00 Count X: 4960 Y:9200 Z:6000
ok

closed 2016-09-23 by thinkyhead