MarlinFirmware/Marlin #4854: compile error question board ULUTIMAIN_2

rooiejoris, 2016-09-20:
Hi,

As a product designer I am not an expert in firmware, but am able to change some things to my purpose. For an experiment I want to use the M163 mixing Gcode on an Ultimaker2.
To transfer the complete changes to the 'old state' of the UM2 marlin is a lot of work, but my workaround idea was to install the Marlinfirmware on a UM2 and connect an Ultimaker original Ulticontroller so you don't have to include the new UM2 Oled screen code. Sort of an UM original plus...

Now the real question : )
When I only change the line in configuration.h
  #define MOTHERBOARD BOARD_ULTIMAKER

to 
  #define MOTHERBOARD BOARD_ULTIMAIN_2

I get the following error:

In file included from /var/folders/xb/znvywn3934ld_x727jcksynw0000gn/T/build4046661928469401744.tmp/cardreader.cpp:26:0:
/var/folders/xb/znvywn3934ld_x727jcksynw0000gn/T/build4046661928469401744.tmp/stepper.h:138:14: error: 'constexpr' does not name a type
       static constexpr int motor_current_setting[3] = PWM_MOTOR_CURRENT;
              ^
/var/folders/xb/znvywn3934ld_x727jcksynw0000gn/T/build4046661928469401744.tmp/stepper.h:138:14: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
Error compiling.

The compiling works when I switch back to
  #define MOTHERBOARD BOARD_ULTIMAKER

any ideas...?!

cheers / joris

thinkyhead, 2016-09-20:
That `constexpr` should just be `const`. I've just patched it in `RCBugFix`. Thanks for the report!

closed 2016-09-20 by thinkyhead

rooiejoris, 2016-09-20:
thanks for the quick response, but...
did try it immediately and still get the/a error:

In file included from /var/folders/xb/znvywn3934ld_x727jcksynw0000gn/T/build4046661928469401744.tmp/pins.h:99:0,
                 from /var/folders/xb/znvywn3934ld_x727jcksynw0000gn/T/build4046661928469401744.tmp/MarlinConfig.h:33,
                 from /var/folders/xb/znvywn3934ld_x727jcksynw0000gn/T/build4046661928469401744.tmp/cardreader.h:26,
                 from /var/folders/xb/znvywn3934ld_x727jcksynw0000gn/T/build4046661928469401744.tmp/cardreader.cpp:23:
/var/folders/xb/znvywn3934ld_x727jcksynw0000gn/T/build4046661928469401744.tmp/pins_ULTIMAIN_2.h:70:36: error: a brace-enclosed initializer is not allowed here before '{' token
 #define DEFAULT_PWM_MOTOR_CURRENT  {1300, 1300, 1250}
                                    ^
/var/folders/xb/znvywn3934ld_x727jcksynw0000gn/T/build4046661928469401744.tmp/stepper.h:141:35: note: in expansion of macro 'DEFAULT_PWM_MOTOR_CURRENT'
         #define PWM_MOTOR_CURRENT DEFAULT_PWM_MOTOR_CURRENT
                                   ^
/var/folders/xb/znvywn3934ld_x727jcksynw0000gn/T/build4046661928469401744.tmp/stepper.h:143:51: note: in expansion of macro 'PWM_MOTOR_CURRENT'
       static const int motor_current_setting[3] = PWM_MOTOR_CURRENT;
                                                   ^
/var/folders/xb/znvywn3934ld_x727jcksynw0000gn/T/build4046661928469401744.tmp/pins_ULTIMAIN_2.h:70:53: error: invalid in-class initialization of static data member of non-integral type 'const int [3]'
 #define DEFAULT_PWM_MOTOR_CURRENT  {1300, 1300, 1250}
                                                     ^
/var/folders/xb/znvywn3934ld_x727jcksynw0000gn/T/build4046661928469401744.tmp/stepper.h:141:35: note: in expansion of macro 'DEFAULT_PWM_MOTOR_CURRENT'
         #define PWM_MOTOR_CURRENT DEFAULT_PWM_MOTOR_CURRENT
                                   ^
/var/folders/xb/znvywn3934ld_x727jcksynw0000gn/T/build4046661928469401744.tmp/stepper.h:143:51: note: in expansion of macro 'PWM_MOTOR_CURRENT'
       static const int motor_current_setting[3] = PWM_MOTOR_CURRENT;
                                                   ^
Error compiling.

thinkyhead, 2016-09-21:
In fact, it compiles even if I leave the `constexpr` in place of `const`.

What version of Arduino IDE are you compiling with?

rooiejoris, 2016-09-21:
arduino 1.7.8
which version are you using...?!

Blue-Marlin, 2016-09-21:
Mopst of us do use the IDEs provided by Arduino.cc (https://www.arduino.cc/en/Main/Software) currently 1.6.11. Marlin is pretty untested with the IDEs from Arduino.org (http://www.arduino.org/downloads) (1.7.10).

rooiejoris, 2016-09-21:
Thanks a lot...!!
It works. (so far, now the adjustments... : )

thinkyhead, 2016-09-21:
I would even go so far as to say that Marlin is _unsupported_ for the arduino.org toolchain. They seem to be using an altogether different version of the gcc compiler.