MarlinFirmware/Marlin #4862: Auto Level Problem CoreXY Printer

genzo76, 2016-09-21:
I opened two tickets for the auto level problem of my printer. I tried different branch but with all I have always the same problem.
The last code on my printer is Marlin-RCBugFix of 21/09/2016.
On my printer corexy the matrix which calculates the height of the point Z, it seems to be reversed both for the X axis and Y.
At position 0.0 in the pot should go up, while pulling down instead nell'angollo 200,200 should fall, but instead the plate rises on.

To give you an idea I shot a video I posted on my youtube account:

[https://www.youtube.com/watch?v=PpLZca50yDU](url)

The printer with your firmware is perfect but this defect forces me to manually modifcare the flat level before each print after running the self-leveling.

Please if you can help me.

thinkyhead, 2016-09-22:
There's a procedure that you can follow to help us collect data. First enable `DEBUG_LEVELING_FEATURE` and re-flash with that enabled. Before you use `G28` and `G29` execute `M111 S255` to turn on all logging. When you do `G29` do it as `G29 V4` to turn on the most verbose output. Then post the log output here.

I will take a look at the video in a moment…

genzo76, 2016-09-22:
I have follow your procedure and this is the result:

```
SENT: M111
READ: echo:DEBUG:off
DEBUG:off
READ: ok
SENT: M105
READ: ok T:24.8 /0.0 B:24.5 /0.0 @:0 B@:0
SENT: M105
READ: ok T:24.8 /0.0 B:24.5 /0.0 @:0 B@:0
SENT: M105
READ: ok T:24.7 /0.0 B:24.4 /0.0 @:0 B@:0
SENT: M105
READ: ok T:24.8 /0.0 B:24.5 /0.0 @:0 B@:0
SENT: M105
READ: ok T:24.7 /0.0 B:24.4 /0.0 @:0 B@:0
SENT: S255
READ: echo:Unknown command: "S255"
Unknown command: "S255"
READ: ok
SENT: M105
READ: ok T:24.7 /0.0 B:24.4 /0.0 @:0 B@:0
SENT: M105
READ: ok T:24.7 /0.0 B:24.5 /0.0 @:0 B@:0
SENT: M105
READ: ok T:24.8 /0.0 B:24.4 /0.0 @:0 B@:0
SENT: M105
READ: ok T:24.9 /0.0 B:24.4 /0.0 @:0 B@:0
SENT: M105
READ: ok T:24.8 /0.0 B:24.5 /0.0 @:0 B@:0
SENT: M105
READ: ok T:24.8 /0.0 B:24.5 /0.0 @:0 B@:0
SENT: S255
READ: echo:Unknown command: "S255"
Unknown command: "S255"
READ: ok
SENT: M105
READ: ok T:24.8 /0.0 B:24.5 /0.0 @:0 B@:0
SENT: M105
READ: ok T:24.8 /0.0 B:24.5 /0.0 @:0 B@:0
SENT: M105
READ: ok T:24.8 /0.0 B:24.5 /0.0 @:0 B@:0
SENT: M105
READ: ok T:24.8 /0.0 B:24.5 /0.0 @:0 B@:0
SENT: M105
READ: ok T:24.8 /0.0 B:24.5 /0.0 @:0 B@:0
SENT: G29 V4
READ: G29 Auto Bed Leveling
SENT: M105
READ: echo:busy: processing
busy: processing
SENT: M105
READ: echo:busy: processing
busy: processing
SENT: M105
READ: echo:busy: processing
busy: processing
SENT: M105
READ: Bed X: 0.000 Y: 34.000 Z: 9.509
READ: echo:busy: processing
busy: processing
SENT: M105
READ: echo:busy: processing
busy: processing
SENT: M105
READ: echo:busy: processing
busy: processing
SENT: M105
READ: Bed X: 92.000 Y: 34.000 Z: 8.211
READ: echo:busy: processing
busy: processing
SENT: M105
READ: echo:busy: processing
busy: processing
SENT: M105
READ: echo:busy: processing
busy: processing
SENT: M105
READ: Bed X: 184.000 Y: 34.000 Z: 7.290
READ: echo:busy: processing
busy: processing
SENT: M105
READ: echo:busy: processing
busy: processing
READ: echo:busy: processing
busy: processing
READ: Bed X: 184.000 Y: 107.000 Z: 7.696
READ: echo:busy: processing
busy: processing
READ: echo:busy: processing
busy: processing
READ: echo:busy: processing
busy: processing
READ: Bed X: 92.000 Y: 107.000 Z: 8.553
READ: echo:busy: processing
busy: processing
READ: echo:busy: processing
busy: processing
READ: Bed X: 0.000 Y: 107.000 Z: 9.722
READ: echo:busy: processing
busy: processing
READ: echo:busy: processing
busy: processing
READ: echo:busy: processing
busy: processing
READ: Bed X: 0.000 Y: 180.000 Z: 9.664
READ: echo:busy: processing
busy: processing
READ: echo:busy: processing
busy: processing
READ: echo:busy: processing
busy: processing
READ: Bed X: 92.000 Y: 180.000 Z: 8.810
READ: echo:busy: processing
busy: processing
READ: echo:busy: processing
busy: processing
READ: Bed X: 184.000 Y: 180.000 Z: 8.178
READ: echo:busy: processing
busy: processing
READ: Eqn coefficients: a: -0.01038428 b: 0.00374707 d: 9.18027591
READ: Mean of sampled points: 8.62585830
READ: 
READ: Bed Height Topography:
READ:    +--- BACK --+
READ:    |           |
READ:  L |    (+)    | R
READ:  E |           | I
READ:  F | (-) N (+) | G
READ:  T |           | H
READ:    |    (-)    | T
READ:    |           |
READ:    O-- FRONT --+
READ:  (0,0)
READ:  +1.03838 +0.18384 -0.44828
READ:  +1.09657 -0.07313 -0.93010
READ:  +0.88323 -0.41495 -1.33556
READ: 
READ: 
READ: Corrected Bed Height vs. Bed Topology:
READ:  +0.00000 +0.10076 +0.42394
READ:  +0.33169 +0.11729 +0.21562
READ:  +0.39186 +0.04898 +0.08367
READ: 
READ: 
READ: 
READ: Bed Level Correction Matrix:
READ: +0.999946 +0.000000 -0.010384 
READ: +0.000039 +0.999993 +0.003747 
READ: +0.010384 -0.003747 +0.999939 
READ: X:183.99 Y:130.01 Z:21.27 E:0.00 Count A: 25130 B:4342 Z:32762
READ: ok

```

genzo76, 2016-09-22:
I have find the bug.

If I use "#define INVERT_Z_DIR false" have the problem, but if change in "#define INVERT_Z_DIR true" the auto level work.

The problem is the direction of Z axis.

FHeilmann, 2016-09-23:
You need to send` M111 S255` as one command, not two separate commands

thinkyhead, 2016-09-24:
Also get the latest `RCBugFix` code. Several changes have been applied, some affecting bed leveling.

genzo76, 2016-10-01:
Have try todey with the RCBugFix version, but have a new type of problem, the autolevel on travel not raise the senso from the plate and move the extrudere on random coordinate of X/Y.
I have loaded old firmware this not work.

This is my configuration:

`/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#ifndef CONFIGURATION_H
#define CONFIGURATION_H

/**
 *
 *  ***********************************
 *  **  ATTENTION TO ALL DEVELOPERS  **
 *  ***********************************
 *
 * You must increment this version number for every significant change such as,
 * but not limited to: ADD, DELETE RENAME OR REPURPOSE any directive/option.
 *
 * Note: Update also Version.h !
 */
#define CONFIGURATION_H_VERSION 010100

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
[... 1315 lines cut ...]
 * 34 - RAMPS1.4 - uses Analog input 5 on the AUX2 connector
 * 81 - Printrboard - Uses Analog input 2 on the Exp1 connector (version B,C,D,E)
 * 301 - Rambo  - uses Analog input 3
 * Note may require analog pins to be defined for different motherboards
 **********************************************************************/
// Uncomment below to enable
//#define FILAMENT_WIDTH_SENSOR

#define DEFAULT_NOMINAL_FILAMENT_DIA 3.00  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation

#if ENABLED(FILAMENT_WIDTH_SENSOR)
  #define FILAMENT_SENSOR_EXTRUDER_NUM 0   //The number of the extruder that has the filament sensor (0,1,2)
  #define MEASUREMENT_DELAY_CM        14   //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

  #define MEASURED_UPPER_LIMIT         3.30  //upper limit factor used for sensor reading validation in mm
  #define MEASURED_LOWER_LIMIT         1.90  //lower limit factor for sensor reading validation in mm
  #define MAX_MEASUREMENT_DELAY       20     //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

  #define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially

  //When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
  //#define FILAMENT_LCD_DISPLAY
#endif

#endif // CONFIGURATION_H`

thinkyhead, 2016-10-02:
@genzo76 I see you're using a normal servo probe. That type of probe should do the normal raise between probes. Please post a log. First, enable `DEBUG_LEVELING_FEATURE` and re-flash. Then send the command `M111 S247` before doing `G28` and `G29 V4` to give full details. Post the log here and we should be able to see what's going on with the missing raise.

genzo76, 2016-10-02:
excuse me not is in my home, attach the log in the next week.

I have see the log and in the log is write rise probe 20, but not is execute, I think becouse not have the time for the operation, after move at next point and not raise the probe.

thinkyhead, 2016-10-02:
I think you are correct. Something is causing Z raise to fail, and there are now various issues about it. A typo or logical error sneaked in during recent code cleanup. I will track it down soon.

boelle, 2016-11-02:
@thinkyhead is this still to be tracked?

thinkyhead, 2016-11-04:
I believe this was fixed when we patched an endstop issue. @genzo76 can you confirm?

boelle, 2016-11-12:
@genzo76 can you confirm the problem is fixed?

closed 2016-11-16 by thinkyhead