MarlinFirmware/Marlin #4869: Whaddya know? We have static_assert!

thinkyhead, 2016-09-22:
It turns out we can, in fact, add sanity checking for configuration options that are `float`, and also any `const` values, including items inside of `const` (or perhaps just `constexpr` arrays. For example:
```cpp
static_assert(Z_PROBE_TRAVEL_HEIGHT > -Z_PROBE_OFFSET_FROM_EXTRUDER,
  "Z_PROBE_TRAVEL_HEIGHT must have a greater magnitude than Z_PROBE_OFFSET_FROM_EXTRUDER");
```

Roxy-3D, 2016-09-22:
I presume the example could have used floating point numbers in it!
```cpp
static_assert(Z_PROBE_TRAVEL_HEIGHT >= 1.23 -
  (Z_PROBE_OFFSET_FROM_EXTRUDER < 0.0 ? Z_PROBE_OFFSET_FROM_EXTRUDER : 0.0),
  "Z_PROBE_TRAVEL_HEIGHT is too small for the Z_PROBE_OFFSET_FROM_EXTRUDER."
);
```

thinkyhead, 2016-09-23:
Yes. But there's no need to add decimal points in the assert. If the define has a decimal point, as it almost always does, the code will still do the right thing.

Roxy-3D, 2016-09-24:
> I find that sort of thing overly pedantic.

Wow!    I never thought of myself as 'Overly Pedantic' before.   But I guess I am with floating point constants!   :)

thinkyhead, 2016-09-24:
You might just be a float-Nazi, Roxanne!

I figure, since the compiler knows what I mean by `if (myfloat > 0)` and I know what I mean, then there's no real need. But I realize there are some who even insist that we stick an `f` on the end of every `float` too, so it won't get cast as `double`. We'll probably start hearing more about that as we get into the 32-bit version of Marlin.

Lukelectro, 2016-10-29:
Cool.

I set Z_max_feedrate to -10.2 today by accident (Mixed up M203 &M206), causing strange sound effects and a unusable printer. So I'm all for extra sanity checks. (Maybe even at runtime, as I made the change at runtime. I don't know if a negative max feedrate makes any sense. If it doesn't, consider it fuzy testing...)

thinkyhead, 2016-10-29:
@Lukelectro I agree it would be a good thing to do more runtime checks for bad parameters. We can certainly afford the extra code. On most boards, Marlin takes up less than half of the available program memory.

Lukelectro, 2016-10-29:
(I have a Melzi with Atmega644, but still agree.)

thinkyhead, 2016-10-29:
Marlin should still fit ok on 64K / 4K boards — not with every feature, of course. The main bottleneck is the SRAM. Global vars are using around 3K which is getting pretty tight for the smaller boards. @fmalpartida and others have been very keen that we should keep the SRAM footprint as small as possible. So we need as much feedback as we can get from users of boards like Melzi V1.

fmalpartida, 2016-10-31:
More than me, there are a good range of boards out there with a small memory footprint. 

While all boards based on the AT90USB family have already been excluded from Marlin, I don't think it should also exclude boards with small footprints.

thinkyhead, 2016-10-31:
> already been excluded from Marlin

How seriously excluded are they? I'd prefer that Marlin accommodate all the boards that it has historically included. There's still a lot of room to improve. We've been in the "elaboration" phase and haven't hit the "revision" and "optimization" phase too heavily yet.

Blue-Marlin, 2016-10-31:
> AT90USB family have already been excluded from Marlin

Really? Or is it that not all features, like the emergency parser, can be used.

fmalpartida, 2016-10-31:
Since moving on the arduino version on RC5 or RC6 you simply can't compile for them.

fmalpartida, 2016-10-31:
That was a move that should have been done after releasing version 1.1 and then move on to 2.0 as a way to indicate that there is a big IDE and interface change.

thinkyhead, 2016-10-31:
> after releasing version 1.1 and then move on to 2.0

Requiring Arduino 1.5 was done to bypass buggy versions of Arduino originally. Now we've adopted some C++11 features, so it's a little extra work to restore the ability to compile with older toolchains. However, it probably wouldn't take more than an hour to strip out C++11 features and make it work. Rather than revert the whole codebase, perhaps just a branch with the necessary changes applied will suffice.

Blue-Marlin, 2016-10-31:
Tested right now to compile for a Teensy++ 2.0.
All what the compiler says,  is some warnings regarding redefinition of  USBCON introduced with #4820.
No clue if it works, but it compiles.

So it's not a general AT90USB problem.

thinkyhead, 2016-10-31:
Good to know newer Arduino IDEs are still supporting the smaller boards. Has Arduino ever "end-of-lifed" or abandoned any of their older boards/chipsets?

Roxy-3D, 2016-10-31:
The PrintRboard has an AT90USB128 on it.   And I can't bring up the current Marlin because I don't have the support files for the newer compilers.   I do have a new AT-Mega256 and RAMPS board to put on it to get around that problem.   I just haven't done it yet.

My problem is because the PrintRboard people won't update their support files and those files only work with Arduino v.22     I actually have it working with Arduino V1.05 but still...

Blue-Marlin, 2016-10-31:
@thinkyhead 
Wrong question.
Arduino is maintaining __their__ stuff.
Additional, not Arduino, boards like you can find in https://github.com/MarlinFirmware/Marlin/tree/1.0.x/ArduinoAddons are not maintained by Arduino but by their communities. Ether the communities care about updating or integrating into Arduino IDE (like Teensy) or not (like OMC, ...)
For me this is neither a Arduino nor a Marlin problem. It's a too small, or not interested enough, community (if still existent) of owners/supporters of the boards. They have to keep up with the progress of the Arduino IDE, not we.
If it would matter we'd have much more issues here. (Sorry @boelle).

@Roxy-3D 
Obviously there are people who can work with printerboards. Else we wont get updates for the pin -files regularly.

thinkyhead, 2016-10-31:
> They have to keep up with the progress of the Arduino IDE

I see! It makes sense. So in general Arduino provides one possible toolchain, and meanwhile UECIDE, platformio, and AVR Studio are some of the other options. For any of these to support a given board or chipset it's up to the community to compose the appropriate support files, and then they'll all work as they should.

fmalpartida, 2016-10-31:
Most AT90USB leverage the teensy porting but given how interrupt management has changed in the ide, it doesn't work with Marlin. For example my board you can compile it using teensy's configuration re doing the pin assignment while it will compile it will not work. As soon as you start streaming gcode to it will not work.

Blue-Marlin, 2016-10-31:
> they'll all work as they should.

As always it's a lot more complicated.
Arduino is supporting a bunch of boards on very different platforms like the 8 bit AVRs like on the UNO or Mega and the Atmel ARMs on the DUE. They do their best to make the Arduino __API__ work on all their platforms. As always such a API is a compromise. Some features are simple and fast on the one platform but slow or impossible on others. Arduino API tries to be a kind of HAL to all the platforms.
The problem with porting Marlin to other platforms is - "We don't use the API!" - we bypass the API about everywhere. It begins with FastIO, a special SD-library, and ends in dedicated registers for timers. For example we could use analogRead() for the thermistors if we'd accept the busy wait until the measurement is ready. But we prefer to invent our own wheel and make the ADC readout much faster but very specialised. We are very specialised on a very small number of 8bit AVR processors to make them work as fast as possible. In the process of speeding things up, we are in danger to narrow down the bandwidth of the possible targets until we end up with the Mega as single possible target. To maintain a bandwidth, of processors, we need a define a lot of different code pathes inside Marlin instead of letting the API do that for us.

At the current DUE ports i see about the same. They are specialised to the DUE . Only @esenapaj 's fork tries to maintain AVRs and the DUE. For the 'official' 32bit port i hope the processors are fast enough to make one step back and use a higher level API like mbed or at least CMSIS. It would be great if we could invest the increased power of the processors only partially in increasing the steprate (maybe 2 or 4 times) and the rest into a higher abstraction.

thinkyhead, 2016-10-31:
Is that fixable from the code side in some way? Perhaps by choosing different interrupts? Or is it due to the bootloader? As far as I understand it we have total control over how interrupts are initialized.

thinkyhead, 2016-10-31:
> But we prefer to invent our own wheel

Well, to be fair, Arduino documentation does offer guidance and suggestions to do things more directly, if one chooses. When it comes down to it, how many smaller boards or chipsets are we talking about supporting? Just a few, isn't it?

fmalpartida, 2016-10-31:
Spot on @Blue-Marlin. It is more than just the bootloader where you can work around. 
That is the reason I suggested a HAL a long time ago.

fmalpartida, 2016-10-31:
@Scott is more about processor platforms. The question is are we just going to support Megas?

fiveangle, 2017-05-03:
@Blue-Marlin : 

> @printerboardowners
> How do you do it?

Install Arduinio IDE 1.8.2, Teensy 1.3.2 extensions, config, compile, upload hex with Atmel FLiP.  And before that is was 1.6.11... Teensy 1.3.1... config, compile, wash rinse repeat. Granted, I don't have any LCD, but otherwise working fine until just recently (#6550)

closed 2017-11-01 by thinkyhead