MarlinFirmware/Marlin #4871: doesn't go home at the end

delid, 2016-09-22:
i am generating gcodes with rhino plugin grasshopper and wanted to make it starts with home position and ends with home position, but it just stops without going home.

it's not really bother me since next code will start with home anyway, but couldn't figure it out.

here is an short example of gcode;

`G4 P0
G28
G1 X14.67 Y5.88 Z98.00 F4000
G4 P0
M106 S255
G1 X14.80 Y5.40 Z98.00 F2000
G1 X14.93 Y4.91 Z98.00 F2000
G1 X15.06 Y4.43 Z98.00 F2000
G1 X15.19 Y3.95 Z98.00 F2000
G1 X15.32 Y3.47 Z98.00 F2000
G1 X15.45 Y2.98 Z98.00 F2000
G1 X15.46 Y2.93 Z98.00 F2000
G4 P0
M106 S0
G1 X-1.45 Y-0.59 Z98.00 F4000
G4 P0
M106 S255
G1 X-1.22 Y-0.15 Z98.00 F2000
G1 X-0.98 Y0.29 Z98.00 F2000
G1 X-0.74 Y0.73 Z98.00 F2000
G1 X-0.51 Y1.17 Z98.00 F2000
G1 X-0.27 Y1.61 Z98.00 F2000
G1 X-0.04 Y2.05 Z98.00 F2000
G1 X0.20 Y2.49 Z98.00 F2000
G1 X0.25 Y2.59 Z98.00 F2000
G4 P0
M106 S0
G4 P0
G28`

FHeilmann, 2016-09-23:
What are you using to control the printer? Maybe one of your tools (i.e. Octoprint) is injecting G-Code at the end of your print that turns off your motors.

thinkyhead, 2016-09-24:
What does `G4 P0` do??

thinkyhead, 2016-09-24:
Are you using 32X microstepping on your machine, by any chance?

delid, 2016-09-29:
thanks for answers, 

i am just plug the sd with gcodes in it and run it, there is no extra control.

i am not using my machine as printer now, i need to control when to power d9 and without `G4 P0` it's powering before i need. (problem is something like G and M codes are not working synchronized and it makes M code wait till all G ones end before it runs)

and it's delta btw. not sure about microstepping and home speed atm, but home speed should be default and for microstepping, i grounded all 3 pins on ramps 1.4, didn't changed any extra setting.

anyway, it's fixed now. i was messing with grasshopper and somehow it fixed, not sure what i change to make it work, cant see any difference on exported gcodes.

thinkyhead, 2016-10-01:
You're probably using `G4 P0` where you should be using `M400`. You can use `M400` before you turn the power on / off (or change the power level) (with commands other than `M106`) to make sure the change occurs just before the move that follows.

GCode moves go into a queue right away, but are processed later.

Meanwhile, commands to enable / disable a power plug or change a pin state take effect right away, not in synch with the G0 or G1 move that follows.

(However, the fan speed set by `M106` / `M107` should go into the command buffer, and take effect when the move that follows it is current.)

closed 2016-10-01 by thinkyhead