MarlinFirmware/Marlin #4872: Easy way to free up 750 bytes of RAM

Roxy-3D, 2016-09-22:
@thinkyhead   Scott, I shrank the RAM foot print of the UBL branch by well over half of a KB of memory by defining two macros.   I then searched for ECHO(" and converted all of those to ECHOPGM(" and got another 100 bytes of RAM freed up.   

Its a really easy way to free up a chunk of RAM in RC-7 if you are so inclined.

```cpp
#define SERIAL_PROTOCOLPAIRPGM(name, value) SERIAL_ECHOPAIRPGM(name, value)
#define SERIAL_ECHOPAIRPGM(name,value) do{ SERIAL_ECHOPGM(name); SERIAL_ECHO(value);} while(0);
```

thinkyhead, 2016-09-23:
Hmm. I think that `SERIAL_ECHOPAIR` already treats the first argument as a PGM string. And sure enough, it is…
```cpp
#define SERIAL_ECHOPAIR(name,value)    SERIAL_PROTOCOLPAIR(name, value)
#define SERIAL_PROTOCOLPAIR(name, value)    (serial_echopair_P(PSTR(name),(value)))
```
The `serial_echopair_P` function is overloaded to always print the PGM string, then the second argument by type. For example…
```cpp
void serial_echopair_P(const char* s_P, const char *v) { serialprintPGM(s_P); SERIAL_ECHO(v); }
```
Your change is basically doing the same thing, so it's interesting/weird that splitting up the single function call into two function calls is saving SRAM. But I won't complain. It's a useful quirk.

I'll have to experiment and see why it's behaving this way, and whether it's producing not only shorter, but more efficient code.

thinkyhead, 2016-09-23:
Actually, it occurs to me that simply redefining the `SERIAL_ECHOPAIR` macro ought accomplish exactly the same thing:
```cpp
#define SERIAL_ECHOPAIR(name,value) do{ SERIAL_ECHOPGM(name); SERIAL_ECHO(value); }while(0);
```

thinkyhead, 2016-09-23:
Before applying that change:
```
Sketch uses 50,610 bytes (19%) of program storage space.
Global variables use 2,726 bytes (33%) of dynamic memory
```
After:
```
Sketch uses 51,906 bytes (20%) of program storage space.
Global variables use 2,726 bytes (33%) of dynamic memory
```
Hmm. That's actually 1296 bytes larger!

Then, with `AUTO_BED_LEVELING` and `DEBUG_LEVELING_FEATURE`
Before the change:
```
Sketch uses 74,206 bytes (29%) of program storage space.
Global variables use 2,810 bytes (34%) of dynamic memory
```
After:
```
Sketch uses 76,130 bytes (29%) of program storage space.
Global variables use 2,810 bytes of dynamic memory
```
And that's 1924 bytes larger.

So far the results aren't looking too good.

thinkyhead, 2016-09-23:
Just did a code sweep, and sure enough there are no spots where the PGM macro needs to be applied. Looks like I got them all pretty well. Notice the size of the code above. How does it compare with the code from RC7, I wonder…?

jasoroony, 2016-09-23:
There's Arduino tools that dump RAM usage of a sketch, I wrote a quick C program to use that output to calculate the size of each variable (not sure if it's 100% accurate, since the last item in RAM can't be calculated) and here's the output: (one of the few times I've used the C qsort!)

```cpp
                                      _ZN7Planner12block_bufferE     1232
                                                            card      653
                                     _ZN8SdVolume12cacheBuffer_E      512
                                 _ZL13command_queue.lto_priv.172      384
                                  _ZTV24U8GLIB_ST7920_128X64_RRD      202
                      _ZZ16lcd_sdcard_menu2vE17longFilenameCache      135
                                                       rx_buffer      130
                                   u8g_dev_st7920_128x64_rrd_buf      128
                                             _ZL14sensitive_pins       98
                  _ZZ19get_serial_commandsvE18serial_line_buffer       96
                                              lcd_status_message       67
                          _ZZ16lcd_sdcard_menu2vE13filenameCache       65
                                       _ZL20manual_feedrate_mm_m       64
                                                             u8g       62
                                                  screen_history       60
                                                          buzzer       30
                                    _ZN7Stepper14count_positionE       16
                                   _ZL11destination.lto_priv.177       16
                        _ZN7Planner26max_acceleration_mm_per_s2E       16
                               _ZN11Temperature14raw_temp_valueE       16
                                 _ZN7Planner17max_feedrate_mm_sE       16
                                                current_position       16
                                 _ZN7Planner17axis_steps_per_mmE       16
                                    _ZN7Planner14previous_speedE       16
                          _ZL20homing_feedrate_mm_s.lto_priv.186       16
                                           _ZN7Planner8positionE       16
                     _ZN7Planner29max_acceleration_steps_per_s2E       16
                                       _ZN7Planner11steps_to_mmE       16
                                                 print_job_timer       14
                                                  position_shift       12
                                _ZN7Stepper18endstops_trigstepsE       12
                                                soft_endstop_max       12
                                                     home_offset       12
                                                soft_endstop_min       12
                                                            conv        8
                                                     _ZTV6U8GLIB        8
                                                      _ZTV5Print        8
                                                     _ZTV6SdFile        8
_ZZN11Temperature26thermal_runaway_protectionEPNS_7TRStateEPmffiiiE21tr_target_temperature        8
                                    u8g_dev_st7920_128x64_rrd_pb        8
                                                     _ZTV6Buzzer        6
                                u8g_dev_st7920_128x64_rrd_sw_spi        6
                                 _ZL13feedrate_mm_s.lto_priv.183        4
                                      _ZN7Planner12accelerationE        4
                                                   filament_size        4
                                       _ZN7Planner11max_xy_jerkE        4
                                                   timer0_millis        4
                                           timer0_overflow_count        4
                            _ZN7Planner22previous_nominal_speedE        4
                                        _ZN7Planner10max_z_jerkE        4
                          _ZN11Temperature19current_temperatureE        4
                                        _ZN7Planner10max_e_jerkE        4
                                 _ZN7Planner17min_feedrate_mm_sE        4
                         _ZN11Temperature20watch_heater_next_msE        4
                                   _ZN7Stepper15count_directionE        4
                          _ZN7Planner24min_travel_feedrate_mm_sE        4
                              _ZN7Planner20retract_accelerationE        4
                                            _ZN11Temperature2KpE        4
                                _ZN8SdVolume17cacheMirrorBlock_E        4
                                _ZN8SdVolume17cacheBlockNumber_E        4
                               _ZN7Planner19travel_accelerationE        4
                                          manual_move_start_time        4
                                            _ZN11Temperature2KdE        4
                                   _ZN7Planner15autotemp_factorE        4
                                      _ZN7Planner12autotemp_maxE        4
                                       lastEncoderMovementMillis        4
                           _ZN11Temperature18raw_temp_bed_valueE        4
[... 147 lines cut ...]
                                                 wait_for_heatup        1
                             _ZL17commands_in_queue.lto_priv.159        1
[... cut ...]
                             _ZL17cmd_queue_index_w.lto_priv.189        1
                      _ZZ10lcd_updatevE23lcd_status_update_delay        1
                                                manual_move_axis        1
                                 _ZN7Planner17block_buffer_headE        1
                                 _ZN7Planner17block_buffer_tailE        1
                                              _ZL13home_all_axis        1
                                                           errno        0
```

jasoroony, 2016-09-23:
Someone wrote to me the other day asking "you likey?" :

Batch file for dropping the .elf file in your temp directory onto...

```cpp
C:\jason\arduino\arduino-nightly\hardware\tools\avr\bin\avr-objdump.exe -S -j .data %1 > vars.txt
C:\jason\arduino\arduino-nightly\hardware\tools\avr\bin\avr-objdump.exe -S -j .bss %1 >> vars.txt
C:\jason\scanvars\Debug\scanvars vars.txt
```

scanvars.cpp :

```cpp
#include <stdlib.h>
#include <string.h>
#include <stdio.h>

int    sc = 0;
char   sym [5000][128];
int    siz [5000];
int    srt [5000];

int comp (const void * elm1, const void * elm2) {
  int f = *((int *)elm1);
  int s = *((int *)elm2);

  if (siz [f] > siz [s]) return 1;
  if (siz [f] < siz [s]) return -1;
  return (0);
}

void main ( int argc, char *argv[ ], char *envp[ ] ) {
  FILE * f;
  char * c;
  char   s [261];
  int    offset = 0;
  int    next;

  if (argc != 2)
    return;

  f = fopen (argv [1], "r");
  if (f) {
    while (fgets (s, 260, f)) {
      if (s [0] == '0') {
        strcpy (sym [sc], &(s [10]));
        c = strstr (sym [sc], ">");
        if (c) c [0] = 0;
        srt [sc] = sc;

        next = strtol (s, NULL, 16);

        if (offset != 0)
          siz [sc - 1] = next - offset;

        offset = next;
        sc++;
      }
    }
    fclose (f);

    f = fopen (argv [1], "a");
    if (f) {

      qsort (srt, sc, sizeof(int), comp);

      next = sc - 1;
      while (next >= 0) {
        fprintf (f, "%64s %8d\n", sym [srt [next]], siz [srt [next]]);
        next--;
      }
      fclose (f);
    }
  }
}
```

Roxy-3D, 2016-09-23:
> I need to warn you that you can't just convert all of them. The ones that I've left unchanged are printing strings from buffers and so forth, which cannot be PROGMEM strings. The code may compile just fine, but when it goes to print the strings, if they're not really in PROGMEM then it will just print garbage.

Right.  I understand.   What I did was search for things like ECHO("  with a double quote mark.  It was only those that I converted.

>Maybe some of the valid PROGMEM strings were still using SERIAL_ECHO by mistake in the codebase you started from for UBL, but I know that today every string that can possibly use the PGM version of the serial echoing macros is using them. I periodically sweep the code looking for any that can be updated. Also, every one that prints a single character is now using SERIAL_CHAR.

OK.   I must have branched before you had converted all those.     I did find a fair number of SERIAL_ECHO(" occurrences.

Anyway...   I got 750 bytes back which I needed because those graphics LCD Displays are memory pigs.

kmbecker13, 2016-09-23:
On this subject, has anyone ever tried using the RAM expansion boards for the Arduino Mega to get some  additional storage?  I believe the LCD may be using the SPI interface, but i believe Arduino can handle more than 1 SPI device assuming the pins are handled somehow.

Roxy-3D, 2016-09-23:
> On this subject, has anyone ever tried using the RAM expansion boards for the Arduino Mega to get some additional storage? I believe the LCD may be using the SPI interface, but i believe Arduino can handle more than 1 SPI device assuming the pins are handled somehow.

I'm not aware of anybody expanding the memory on the AVR boards.   And really...   I don't think we are in need of doing that yet.   I'm probably the most abusive user of the RAM and EEPROM space, and I don't feel like we need to do weird things to get more memory.

In the case of the UBL System, I'm usually running a 10 x 10 Mesh which is 400 bytes of floating point numbers.   At 15 x 15, that is still 'only' 900 bytes.   Without a graphics display, that works.    And we can cut those numbers in half by storing them as fixed point numbers in integer types.   (For speed reasons, I'm not doing that.  But in the EEPROM, where speed is not important, we may want to consider doing that.  However, even there, right now we have room for 8 or 9  10x10 Mesh.  So it isn't like we need to compress things just yet.)

Do you know what bigger memory boards are available for the Arduino Mega ?   Can I just buy them on eBay?   If I could get more memory for a reasonable cost and it just connected up to my Mega board with a few wires, it might be fun to play with one.   My guess is it would be a lot slower than the built in memory.  Do you know anything about the speed of the external memory?

thinkyhead, 2016-09-23:
Thing is, we allocate all the memory we need through declaring globals. There is no dynamic allocation apart from what some libraries might be using. And of course, the stack (but that's not really "dynamic allocation" in the same sense). And they are also supposed to allocate through globals. My sense is that whatever the compiler reports as the free memory (e.g., "leaving 3,761 bytes for local variables") is the amount of free space we have, and that when Marlin is running, almost all of that memory is available to the stack.

Roxy-3D, 2016-09-23:
> My sense is that whatever the compiler reports as the free memory (e.g., "leaving 3,761 bytes for local variables") is the amount of free space we have, and that when Marlin is running, almost all of that memory is available to the stack.

Yes, except I think the 'objects' get created by the run time C++ start up code.   My guess would be their size gets subtracted from local variable size.

thinkyhead, 2016-09-23:
> 400 bytes of floating point numbers

If we want to store the probed points in EEPROM for later usage we can save on half of the storage easily, and maybe even 3/4 of that storage. Here's how:
```cpp
uint16_t start_x, start_y; // start X and Y for probe points - int is close enough
uint16_t space_x, space_y; // space between points, in microns
uint8_t num_x, num_y; // number of points in each dimension
uint_8_t z_avg_height; // integer for the mm average probe height
uint8_t z_heights[num_x*num_y]; // convert Z height offsets into 256ths of a mm
```

thinkyhead, 2016-09-23:
> I think the 'objects' get created by the run time C++ start up code

Actually, they are declared as global object instances. There is no use of `new`.

Roxy-3D, 2016-09-23:
> Of course, these days mesh leveling is much preferred. I suggest we store that in its own section, at the end of the EEPROM, so it won't be overwritten by M500. Same with the M303 results and any other parameters that have their own individual save flags in the future.

Yes...   That is what I'm doing.   The Mesh State information is at the very end of the EEPROM.  And then what ever space is still free is used to provide 'slots' for your various Mesh Patterns.

```cpp

void bed_leveling::store_state()    {
int k;
	k = E2END - sizeof( blm.state );
	eeprom_write_block( (void *) &blm.state , (void *) k, sizeof(blm.state) );
	return;
}

void bed_leveling:: store_mesh(int m) {
int j, k;

	k = E2END - sizeof( state );
	j = (k - Unified_Bed_Leveling_EEPROM_start) / sizeof( z_values );

	if ( m<0 || m>=j || Unified_Bed_Leveling_EEPROM_start<=0) {
		SERIAL_PROTOCOLLNPGM("?EEPROM storage not available to load mesh.\n");
SERIAL_PROTOCOL(m);
SERIAL_PROTOCOL(" mesh slots available.\n"  );
SERIAL_PROTOCOLPAIR("\nE2END     : ", E2END  );
SERIAL_PROTOCOLPAIR("\nk         : ", k  );
SERIAL_PROTOCOLPAIR("\nj         : ", j  );
SERIAL_PROTOCOLPAIR("\nm         : ", m  );
SERIAL_PROTOCOL("\n");
		return;
	}

	j = k-(m+1)*sizeof( z_values);	
	eeprom_write_block( (const void *) &z_values , (void *) j, sizeof(z_values) );

	SERIAL_PROTOCOLPGM("Mesh saved in slot ");
	SERIAL_PROTOCOL( m );
	SERIAL_PROTOCOLPGM("  at offset 0x");
	prt_hex_word(j);
	SERIAL_PROTOCOLPGM("\n");
}
```

thinkyhead, 2016-09-24:
I highly recommend implementing the method I described to save on EEPROM space, if needed.

Roxy-3D, 2016-09-24:
> I highly recommend implementing the method I described to save on EEPROM space, if needed. It will come in handy in the future if space should get tight.

I'm a little hesitant to assume 1/64 of a mm is as high of precision as we will ever need.    But with that said, a very natural way to cut the storage cost would be to just store an int instead of a float.   If each Mesh Point float was multiplied by 1000.0 that would give us 3 decimal points of precision and still allow the Mesh Points to be +/- 32mm in size.    That would cut the storage cost in half.    

And if we were willing to pick up a floating point multiply at run time, we could cut the RAM foot print in half also.  We would multiply each Mesh Point int by .001 to get it back into the world of floating point.    But I think I'm going to wait until the Delta's are up and running well to do that.  The Delta's with a Graphical Display will be the place to test that.

thinkyhead, 2016-09-24:
Given that the smallest practical FDM layer heights are 50µm, I don't think +- 1/64th of a millimeter (15µm) is bad at all. At most, a mesh junction might be off by 15µm, and most of the time they will be within 8µm of true. That's well within the error margin of most probes.

A 2-byte integer is also fine, as it certainly cuts down the storage needs. But obviously 1µm of accuracy is overkill. If we enforced a maximum deviation of 2mm (rather than 4mm), then the accuracy goes up to +- 1/128th of a mm, or ~8µm, still within a byte, which again is more than enough. It would also be possible to combine both matrix and mesh, using the mesh to correct for any deviations in the matrix.

Another option is to compact the data. Use the high bit of each byte to mean "take the next byte too." Any deviations from 0-127 128ths of a mm would then take a single byte, while only larger deviations would need 2.

In the long run we may go for fixed point throughout. And in the even longer run it may be prudent to express more values at print time in terms of stepper steps.

Roxy-3D, 2016-09-25:
> A 2-byte integer is also fine, as it certainly cuts down the storage needs. But obviously 1µm of accuracy (or, as 2:14 fixed-point, 16384ths of a mm, 0.06µm) is overkill. If we enforced a maximum deviation of 2mm (rather than 4mm), then the accuracy with a single byte goes up to +- 1/128th of a mm, or ~8µm, which again is more than enough for FDM.

Mostly, I'm resisting the 1/64mm idea because I want Marlin to still be used by those Graffiti people that are using spray cans instead of extruders.  And really...  It wouldn't be that hard to have an option that controlled whether we compress to two bytes or 1 byte.   It would only be in a few places in the code.   (I fully understand the need not to burn up memory needlessly!)

> A 2-byte integer is also fine, as it certainly cuts down the storage needs. But obviously 1µm of accuracy (or, as 2:14 fixed-point, 16384ths of a mm, 0.06µm) is overkill. If we enforced a maximum deviation of 2mm (rather than 4mm), then the accuracy with a single byte goes up to +- 1/128th of a mm, or ~8µm, which again is more than enough for FDM.

One question:   Isn't what I suggested 5:10 with 1 bit used for the sign?   I was thinking if I had 3 decimal places of precision (.001) that could handle any hardware that is going to show up in the next couple of years.   And if I did that, that would imply I could go as big as +/- 32.000  which would be 5 bits of significand.   In floating point numbers there is an implied 1 bit that they don't include in the storage.  But this is kind of a fixed point numbers so I don't think I get the benefit of that 'implied 1 bit'.

> Mainly I'm thinking of chopping this down because I need to manage a 16x16 mesh (256 points) in my current project.

Wow!   I certainly understand where you are coming from!   But one problem is the bigger the bed, the more the height is going to change across the bed.   If you can really have only 2mm of height difference across a big bed, we can drop it to one byte.   I'm just a little worried that is going to be too restrictive.

thinkyhead, 2016-09-26:
I guess the EEPROM on most boards is 4K. So actually I'm not too worried about filling it up. We're using less than 500 bytes at this point.

thinkyhead, 2016-09-26:
> Isn't what I suggested 5:10 with 1 bit used for the sign?

Could go that way. Of course if you include the lowest Z point, which may be negative, then all the offsets from that point will be positive.

thinkyhead, 2016-09-26:
> that could handle any hardware that is going to show up 

I agree it's good to think ahead. They are only going to get better, and we should start thinking about 32-bit support going forward.

thinkyhead, 2016-09-26:
> I'm just a little worried that is going to be too restrictive.

I tend to agree. This particular bed is circular with a radius of 400mm from the center. And my client in Portland has this 1.5m square bed. People are getting ambitious with these bed sizes.

closed 2017-10-31 by thinkyhead