MarlinFirmware/Marlin #4876: Problem with Dual Z Steppers

Phife54, 2016-09-22:
Im trying to setup dual Z steppers on my MiniRambo board with the latest marlin firmware. Its a little different as the MiniRambo only has 1 extruder driver, I also have a unique situation as this is for a SLA Printer, so I do not have any extruders and only 2 steppers for Z axis.

As I understand, the Dual_Z_Steppers uses the unused extruder driver which is usually E1, I have my two steppers plugged in at Z and E0. Ive enabled Dual_Z_Steppers and Dual_Z_Homing and have changed the MiniRambo_Pins to define the pins for E1 as they were not there, I have just copied the pins from E0. So far so good, I can move both steppers with Z move commands

The problem I am facing is that while both steppers can move in one direction correctly (Min), when I try to move in the opposite direction only the Z stepper reverses and moves in the opposite direction, the stepper on E0 will only turn one direction. I have swaped steppers and the problem moves to the other stepper so its not the steppers or wiring.

I think its got to be a problem with the direction signal and the way I have "tricked" Marlin into using E0 as Z2 instead of E1. 

Can someone please help me out in figuring this out? Ive spent a couple days on it now.

Thanks

jasoroony, 2016-09-23:
Have you try setting the original E0 pins to some pin numbers you are not using?  It's probably setting the direction on both E0 and E1 so, since they the same in your config, the last one set is the value used.

Phife54, 2016-09-23:
That was it! Thanks jasoroony!

Now I have an issue with Z Dual Endstops.

I press Home Z and the steppers start to move to Z Min but they only turn about 2 revolutions and stop. Rambo reports homed and shows axis locations at 0,0,0. If I trigger either endstop while the Z is homing nothing happens. It continues until it does 2 revolutions and stops, each time, no matter what the state of the endstop is.

Homing works correctly when disabling Z Dual Endstops, but I would like to use two endstops for leveling the Z axis. I am using Hall-o Endstops and they are working correctly and plugged into Z min and X Max as per the instructions for Dual Z Homing. I can see the endstop states change with the M119 command.

Any Ideas?

jasoroony, 2016-09-23:
If the 2nd endstop's wires are anywhere near the stepper motor wires you might be picking up cross-talk?  That's certainly an issue I've had, but it didn't turn my leadscrew twice, just a bit down and a lot up...  Quite annoying since it ends up being in a worse position than when it started.  Simply moving the wires fixed that.

But it sounds like something else is wrong here, maybe the config needs to be checked for inverting their signal?

Phife54, 2016-09-23:
Ive given up at this point with the dual endstop homing, I saw some someone else with a similar issue here: https://github.com/MarlinFirmware/Marlin/issues/3814

But I dont think the solution is in there. Its not an issue with noise or crosstalk as it works fine when not using Dual Endstop homing. There must be something about it in the firmware thats causing this problem. I will revisit this again in the future, for now, I will live with a single endstop for the dual Z axis.

closed 2016-09-23 by Phife54

thinkyhead, 2016-09-24:
If there's a bug in dual Z endstop homing, perhaps someone with that kind of setup will come along to help us debug it. Fingers crossed!