MarlinFirmware/Marlin #4877: Repeat last print function

gordo3di, 2016-09-23:
Working on a "Repeat last print" option to come up after a print completes (and only then). We print a lot of the same parts and this would be great.

I'm looking at the section where you select a print from the sd card and initiate the print for more code ideas. Then the menu function would trigger as soon as a print is finished. Im not s coder but I will brute force this if needed.

Just looking for any help/ideas on where to start diving into this.

Roxy-3D, 2016-09-23:
There are a lot of issues to consider.   For starters, how does the 1st print get removed from the bed?  (Some of my prints stick so hard to the glass I have to put the glass into the freezer before I can remove them.)  Does anything need to be done to prepare the bed for the next print?  (If I don't spray hair spray on my glass, it is not uncommon for the first layer not to stick.)    Does the 2nd print just continue and assume its first layer got put down with good adhesion?  (Even with an accurately tuned auto bed leveling system, the 1st layer can fail.  Does the printer just continue and make a rats nest of filament?)

Good software design requires careful thought about the use cases.   Let's see if we can get a good discussion going on this topic such that a developer can feel confident the work to  code this will be successful.

gordo3di, 2016-09-23:
Thank you for the considerations.  We want to accommodate many printer types so I can only speak for ourselves.
I imagine this feature being "off" be default but can be allowed via the LCD menu so you can turn it on later. Also it would default to "no" after the print is finished and you can rotate the knob to "yes" to repeat the print or "no"to just ignore it...

For us we print on 1/4" acrylic. For our prints you pretty much can easily pop off every print up to about the 10th print (if its the same exact print in the same spot using the same material).. After that you need a simple light sanding then its ready again so you would just sand then click "yes" or whatever. We might be lucky on that, but it's our normal routine. If you print different shapes in different orientations you can do 20-30 prints before needing a light sanding hence why you can see this feature is huge for us. I have no problem programming it but it may be sloppy. I have already programmed a Purge PLA/ABS which is very useful but its not the cleanest method. It uses some weird gcode which cant exceed 4 commands or it fails but it works.

Not to go on a tangent but we have been very interested in connecting with you (and everyone) but have reached out to someone already with no response. Our goal is to add features and help the community in the best way possible without being a drag (since we aren't the most githubsavy). We are also very interested in providing donations, but we do not want favors or anything, if that makes sense. We just want to give back to the best community and think you should be paid for your work. Anyway this if for another forum but it's hard to get a response sometimes.

Roxy-3D, 2016-09-23:
> I have already programmed a Purge PLA/ABS which is very useful but its not the cleanest method. It uses some weird gcode which cant exceed 4 commands or it fails but it works.

This is a much needed and important feature.   You may want to consider making improvements to the 'P' option on the G26 Mesh Validation Pattern command in the https://github.com/MarlinFirmware/Marlin/tree/devel-ubl branch.     You can examine the code and see how it does its work.  You may need to bring up the UBL System on a machine so you can see why the Mesh Validation Pattern is important and how the G26 command fits into the system.    But the point is, there is a good, debugged 'Prime' function already coded if you want to leverage that code.

gordo3di, 2016-09-23:
Thank you @Roxy-3D

bgort, 2016-09-23:
> We are also very interested in providing donations, but we do not want favors or anything, if that makes sense. We just want to give back to the best community and think you should be paid for your work. 

I feel the same way.  Is there a way to donate to the project?  You guys are doing a fantastic job and I'd really like to contribute financially (and otherwise, time permitting).

emartinez167, 2016-09-23:
Scott has a Patreon account, if that is what you need to know. Not sure about Roxy or the others.

Regards,

Ernesto

gordo3di, 2016-09-23:
Thanks. Backed Scott. If we were to offer hardware to whom or where would it go?

Roxy-3D, 2016-09-23:
> Thanks. Backed Scott. If we were to offer hardware to whom or where would it go?

What kind of hardware are you thinking?   I would be interested in joining a discussion to find good ideas and a home for any hardware!

gordo3di, 2016-09-23:
I am the co-owner of gCreate. We build the gMax line of printers. Maybe we could work something out :) http://www.gcreate.com/

Roxy-3D, 2016-09-23:
> I am the co-owner of gCreate. We build the gMax line of printers. Maybe we could work something out :) http://www.gcreate.com/

@gordo3di 

Wow!   Those are very nice looking printers.   And with a 16" x 16" bed, they are going to really benefit from the UBL Bed Leveling System.    If you have an older prototype printer laying around,  I'll bring up the UBL System on it and get everything working perfect on it.   And then, switch to using that as the development platform for the code base!

gordo3di, 2016-09-23:
That's music to our ears :) While we try to make our beds as flat as possible the UBL system would be the icing on the cake.  Do you have an email address we can talk further?

Roxy-3D, 2016-09-23:
Unfortunately, we can't send private messages on GitHub.  

But there are a couple of ways to exchange email addresses.    I tried sending a message to gordo3di@gcreate.com    but it bounced.   If you have easy access to your 'Contact Support' at www.gcreate.com I can generate a 'Support Ticket' and use that to send you my email address.

Alternatively, if you are willing to create an account at www.3DPrintBoard.com you can send a private message to Roxy there with your email address.

gordo3di, 2016-09-23:
gordo3di is my internet pseudonym :) if you email info@gcreate.com I'll catch that

Roxy-3D, 2016-09-23:
Done!

gordo3di, 2016-09-23:
That would be a great way to do it. Then when we walk over, pop off the print and just hold the knob we're ready to go. Its funny, it's a small feature but for production use it really does make things easier. And yes we print from SD  so if the filename is still stored it should hopefully be pretty simple.

thinkyhead, 2016-09-24:
Thank you for the backing! Every bit helps, and I sure do spend a crazy amount of time on this (very worthwhile) project.

There's actually a facility for repeating a print, sort of. It's the `M32` command. You can make a GCode file and that looks like this…

```gcode
M32 P !lantern.gco#
M0 Click to print again
M32 P !lantern.gco#
M0 Click to print again
M32 P !lantern.gco#
M0 Click to print again
M32 P !lantern.gco#
```
…and then "print" that file. It will then print multiple copies of some other file.

Would that work for you?

gordo3di, 2016-09-24:
Ohh thats a pretty useful way to do things. I will play with it. Thanks!

gordo3di, 2016-09-24:
Weird. Doesn't seem to work. Tried making the gcode file etc.. then tried running the command from host software. Keeps saying open failed

> SENDING:M32 P !TEST_SQUARE.GCODE#
> echo:Now fresh file: TEST_SQUARE.GCODE#
> open failed, File: TEST_SQUARE.GCODE#.

Tried these
M32 P !TEST_SQUARE.GCODE#
M32 P !TEST_SQUARE.GCODE
M32 P TEST_SQUARE.GCODE 
M32 TEST_SQUARE.GCODE
M32 !TEST_SQUARE.GCODE#
M32 P !./TEST_SQUARE.GCODE#
M32 P !../TEST_SQUARE.GCODE#
M32 S100 !TEST_SQUARE.GCODE#

Also tried lower case but pronterface will only send caps and I tried all those same commands with .gco instead of .gcode  Still nothing. Also tried shortening the filename. Weird.

File prints fine if I select it directly

Saw this too:
https://github.com/MarlinFirmware/Marlin/wiki/M32

Any ideas?

Blue-Marlin, 2016-09-24:
Try 8.3-FILENAMES in capital letters.

gordo3di, 2016-09-24:
Ok got it working. I can finally get a file to finish printing then wait for the next one if I keep the total character count under 8 (including the extension). I haven't tried the SFN convention but I assume this would work too. Thanks @Blue-Marlin I learned a lot today about the 8.3 filename stuff

https://en.wikipedia.org/wiki/8.3_filename#How_to_convert_a_long_filename_to_a_short_filename

Now my code is simply:

```gcode
M32 P !test.gco# ; # at the end stops buffer
M0 Click for another ; Wait for user
M32 P !test.gco# ; # at the end stops buffer
M0 Click for another ; Wait for user
```

Roxy-3D, 2016-09-24:
> Ok got it working. I can finally get a file to finish printing then wait for the next one if I keep the total character count under 8 (including the extension).

I'm not sure that this would be a good thing to do.   But I would try taking the test.gco file and putting these lines as the last lines of the file:
```cpp
M32 P !test.gco# ; # at the end stops buffer
M0 Click for another ; Wait for user
```
I'm not sure what happens when you keep nesting deeper and deeper into the same file.  But if that worked,  it simplifies things.   What ever file you print, just keeps restarting itself at the end.

gordo3di, 2016-09-24:
Ah, so simple. Then at the end if you don't want another just restart the printer. Are there gcode commands which ask for input conditional? Like Yes/no? Anyway I have goodies for you @Roxy-3D . Been working on them today.

gordo3di, 2016-09-24:
Confirmed works fine! No need for a new feature. Thanks everyone

thinkyhead, 2016-09-24:
There's no command that asks for continue/cancel or other user input, but that has been in the back of my mind as an extension to `M0`/`M1`. Other commands are free to ask for more detail or present a longer menu, but for `M0`/`M1` I think continue/cancel would be the only viable general option. `M0`/`M1` already includes a parameter for how long to wait for the user before continuing on its own. But it just proceeds with the next GCode in the queue. In an SD printing situation it could offer the option to cancel instead of continuing when the timer expires, and it could present an on-screen menu to continue/cancel, while displaying the countdown, if any.

gordo3di, 2016-09-24:
Having a simple Yes/no would be literally amazing. It would extend the usefulness of gcode so much. I made 4 files this week for printer tests and they all leverage the M0 command basically as a pause. Being able to have a simple input would do a lot. Now a wishlist would be to have an input that then continue to a specified line of code based on the input ie. if "yes" go to 118 and in "no" go to 150...

thinkyhead, 2016-09-24:
>  if "yes" go to 118 and in "no" go to 150

It's a nice idea, but we'll never do that. Line numbers, such as they are, are generated by the host software, and are not a standard feature within GCode files. And, really, GCode should not be extended into a "programming language" with conditionals, line labels, goto, etc. Such features would only be appropriate from the host side. Marlin aims only to be a faithful runner of standard GCode. I'm afraid that the best that can be offered is continue/cancel within the paradigm.

gordo3di, 2016-09-24:
Got it... I said wishlist lol :) But really marlin is seriously an amazing feat as it is.

gordo3di, 2016-09-26:
New problem using M32 at the end of a file...

Used this code:

```gcode
M0 Click for another copy ; Wait for user
M32 P !32.gco# ; Print again
```

After the 2nd print the printer said:

> KILLED
> Please reset printer

Tried this a few times with the same result. Trying the original idea of using one gcode file to run others and will report on results.

thinkyhead, 2016-09-27:
Does removing the comments (after the semicolons) make any difference?

thinkyhead, 2017-10-31:
@Roxy-3D recently implemented this feature (`SD_REPRINT_LAST_SELECTED_FILE`) and it's pretty awesome. It will be included in Marlin 1.1.7 coming as soon as a few lingering issues are scrubbed…

closed 2017-10-31 by thinkyhead