MarlinFirmware/Marlin #4880: FILAMENT_CHANGE_FEATURE  wihout LCD Display

sikousikou, 2016-09-23:
Hallo,
I have a printer without LCD displays I control it with the PC with repetier host
it is possible to use the FILAMENT_CHANGE_FEATURE wihout LCD display  for example with send an M command (m6xx) for the confirmation that Filament was exchanged 
sorry for my bad english

petrzjunior, 2016-09-23:
No, Filament change is currently tied to LCD. However there was some discussion at #4099 about using buttons and endstops, but seems nobody did it.

MagoKimbra, 2016-09-23:
Use pause of Repetier Host. You can set the gcode when click pause in repetier host.
You can put @pause in your gcode to enable this option..

thinkyhead, 2016-09-24:
Unfortunately, although @pause would allow invoking a command like `M600`, the `M600` command still cannot be enabled without a controller because all its interaction is through the controller. So you would need to assign @pause to a whole list of commands, if that's possible, which would eject the filament, wait for insertion, print messages, wait for user, etc.

Using endstops in place of buttons is a nifty idea, but it needs to be applied in a lot of places and all that code would have to have special handling for when there's no LCD, so it hasn't been implemented yet. And it would need to be careful not to invoke a print abort when `ABORT_ON_ENDSTOP_HIT_FEATURE_ENABLED` is on. It's not as simple as just making a custom `LCD_CLICKED` macro that checks endstops.

Note that in `RCBugFix` currently, with `EMERGENCY_PARSER` enabled the `M108` command can be used to break out of an `M0` or `M1` pause when there's no LCD attached. That was done as a preliminary test of the idea. And even that simple command has to have special handling.

The `M108` trick could be applied to `M600` as well, obviously. But for a proper implementation `M600` needs to print messages to the serial output (like those that would be printed on the LCD) to give user direction. It shouldn't just be "blank" and I don't think we should go in the direction of recent `G29` implementations (adding numbered steps) either.

Roxy-3D, 2018-03-10:
> No, Filament change is currently tied to LCD. However there was some discussion at #4099 about using buttons and endstops, but seems nobody did it.

I think we should allow end stop switches to be used during printing to signal this and that.   Of course there are issues with this.   The trick is to identify all the issues.   If there are not a crazy number of issues associated with doing this, it can be done.    

But it doesn't make sense to do this hack (and it *IS* a hack) without carefully thinking through the issues it will cause.

It might be worth while for people interested in this topic to read the thread where this was posted:
https://github.com/MarlinFirmware/Marlin/pull/4099#issuecomment-227310240

boelle, 2019-02-18:
given that an lcd cost next to nothing depending on what level of quality you want this does not make any sense

i would vote it down or reject it

InsanityAutomation, 2019-02-18:
Actually this becomes important with external touchscreens and extensible UI. Host prompt support was the first stage of this. It is now feasible to remove the limitations requiring the lcd. Its just a matter of shifting a few things so it still compiles as expected now and publicizing M876 as the handler.

InsanityAutomation, 2019-10-15:
@boelle @shitcreek aside from M702, this is done. Can you close this?

boelle, 2019-10-15:
yep, the axe is flying again :-D

closed 2019-10-15 by boelle